Zendure/Zendure-HA #99: manual ac charge mode did not work

FrColin, 2025-04-16:
Hi,
Trying to set manual mode to make ac charge did not produce any effect.
Zendure app show ac mode disabled due to smart ct mode on.
Should disabled ct mode when switch to manual

Here is some log output 

`[32m2025-04-16 08:15:50.360 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000 22 = 2 * (11 - 0)[0m
[32m2025-04-16 08:15:50.360 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 200 stateManagerState.DISCHARGING capacity: 22 max: 1200[0m`

FireSon, 2025-04-25:
@FrColin what do you mean with 'manual mode'? The AC mode can only be used if the Zendure Manager is in operation mode OFF. The provided logging is however something related to Smart Matching.

closed 2025-04-25 by FireSon

FrColin, 2025-04-27:
Hi,
I was putting the zendure manager in "power manual" and set the power to a positive value ( charging ?? )
as documented in the README
Manual power; the 'Zendure Manual Power' number is used to set discharging (if negative) and charging if positive.