Zendure/Zendure-HA #104: Smat Modu Funktioniert ab Version 1.0.29 nicht mehr

MarcelF1, 2025-04-17:
Hallo zusammen

erstmal ein großes Lob ich finde die Integration echt klasse

nur habe ich seit Version 1.0.29 echte Probleme ich kann den Smart Modus gar nicht mehr benutzen oder er geht nicht mehr.

habe schon eine komplette Neuinstallation probiert bei Version 1.0.34 ging die auch kurz aber er hat immer genau 108 W eingespeist egal wieviel Bedarf da war 

ich wollte das Heute nochmal reproduzieren aber es mir gar nicht mehr gelungen den Smart Modus zu aktivieren wie ihr auf den Screenshots seht.

Ich hoffe das der Fehler irgendwo bei mir liegt kann diesen aber einfach nicht finden.

[image]

mangri01, 2025-04-18:
Hallo Marcel,

ich habe mir auch die Zendure in mein Home Assistant Installiert. WIe hats du deinen Hyper Eingestellt. Ich meinen welchen Energieplan benutzt du am Hyper wenn du es über Home Assistant smarte Leistungsregelung regeln lassen willst. 

Ich habe 4 x SolarFlow 1200, alle Daten von den Hubs sehe ich und diese sind immer Aktull. Ich benutze ebenfalls einen Shelly 3em pro für die Leistungsessung Stromverbrauch.

wenn ich diesen Sensor shelly 3em in der Zendure Hacs eingebe steht da immer sensor.actual_power. Jedesmal wenn ich meine Shelly speichere und die Konfiguration aufrufe wieder der sensor.actual_power

Habe mir einen Sensor mittels Helfer (Template) gebaut und den als sensor.actual_power benannt und als w festgelegt.

Danke für deine Antwort

Konstel, 2025-04-18:
Bei mir das gleiche Problem. Entweder stehe ich auf einer Leitung oder es ist tatsächlich ein Bug. 

Bei mir übernimmt er die Sensor Einstellungen / Benennung nicht und im smart Modus passiert gar nichts. 

Man könnte jetzt natürlich selbst Automationen schreiben, aber ich hoffe hier tatsächlich auf ein Helferlein. 

Ich bin ja schon froh, dass man generell vernünftige Werte und zwar zeitnah bekommt. Davor direkt über  mqtt war das ein Graus.

FireSon, 2025-04-18:
Die Modus sollte auch fur inem einzelnen Hyper 2000 funktionieren. Ich habe damit angefangen weil ich keinen Shelly hatte, und trotzdem mit meinen P1 meter, die gerate steueren mochte. 
Ich habe keinen SolarFlow 1200, und deswegend hat das gerat nur die 'reporting' funktionalitat und funktionier (noch) nicht im HA smart mode. Mochte das gerne hinzufugen wenn klar ist welche Befehle dafur verfugbar sind bei das gerat. Dass ware einfach mal zu testen mit das einschalten von die debug logging.

Es gibt ein par Probleme mit die HA smart mode, und ich bin dabei die zu beheben. Kasst du mal die version 35 ausprobieren? 

Es ist nich notwendig um "sensor.actual_power" zu benutzen. Beim konfiguration kannst du einen beliebigen sensor eingeben. Die einzige voraussetzung ist das diese sensor positiev ist in dem moment das energie gebraucht wird, und negativ bei energie production. Ich bin nicht sicher, aber ich glaube das Shelly auch schon so einen sensor hatt.

C0rd0n, 2025-04-18:
Ich habe leider das selbe Problem. Mein Sensor heißt:
sensor.virtueller_smart_meter

Er zeigt negative Werte bei Einspeisung und positive bei Netzbezug. Leider funktioniert das Smart Matching nicht.
In der Zendure App ist kein Energieplan ausgewählt. Was ist falsch?

Der virtuelle sensor zeigt bei mir tagsüber den normalen Netzbezug/ Einspeisung an. Da ich aber nachts einen anderen Strompreis habe, wird in der Nacht der Verbrauchswert auf 0 gesetzt bzw. bei einer zu schwach erwartetem PV Ertrag wird der Wert auf -1200 gestellt.

Liegt es eventuell daran?

[image]
[image]

FireSon, 2025-04-18:
Das sieht eigenlich alles OK aus! Wie viele Hyper2000 hast du? Mit einen sollte es eigenlich sofor funktionieren, mit mehrere braucht man die device cluster aus zu wahlen. 
[image]
Die ZendureManager verteilt die strom ueber die verfugbare gerate, abhangig von die aktuelle kapacitat = charging = (socSet - electricityLevel) * packNum und discharging = (electricityLevel - minSoc) * packNum. Weil es nicht erlaubt ist mehr als 800w zu lieferen (ohne massnamen) begrenst die ZendureManager dass auf das ausgewahlte maximum. In die readme steht einiges daruber.

Wenn es immer noch nicht funktioniert, kannst du die 'debug logging' einschalten und das anschauen in de Settings/System/Log die integration logt mit welche Werte er rechnet, und kannst du ueberprufen ob das richtig ist!

mangri01, 2025-04-18:
Alos bei mir geht es nicht. 
Mein System 4 x Solar Flow 1200 je ein AB2000 oder AB1000 Batterie.
Die Systeme sehe ich in Home Assistant alle Sensoren sind live da und Aktualisieren sich. 
Wenn ich meinen Shelly 3em pro als Strommesser hinterlege und dann in Home Assistant auf smarte Leistungsregelung umstelle passiert nichts.

Muss ich in einen Energieplan in der Zendure App einstellen, oder was soll man einstellen.

[image]
[image]

MarcelF1, 2025-04-18:
Hallo mangri01

probier mal folgendes aus 

schalten den Smart Modus auf aus in der Integration.

Dann gehst du in die App deaktivierst alle Energiepläne 

dann logst du dich aus der App aus und lädst einmal die Integration in Home Assistent neu.

und dann aktivierst du den Smart Modus in Home Assistent und wartest eine Minute und schaust was passiert. (**Nicht** wieder in der App Einloggen ).

Meine Theorie ist es sobald man mit einem Account (egal ob der 1. oder der 2.) eingeloggt ist geht der Smart Modus nicht mehr.

Ich kann es leider selber noch n nicht probieren weil der Akku leer ist.

mangri01, 2025-04-18:
Danke für deinen Tipp.

Ich habe alles so gemacht wie du es geschrieben hast. Leider ohne Erfolg

Aber meine Frage, wenn ich auf Konfiguration klicke kommt immer der sensor.actual_power, obwohl ich schon mehrfach meinen Shelly 3em pro hinterlegt habe. 

Hats du noch einen Tipp

[image]

MarcelF1, 2025-04-18:
Dort habe ich den Sensor Total Active power vom Shelly 3em eingegeben da dieser alle Vorraussetzungen erfüllt was nach dem ich den Sensor geändert habe dort drin steht habe ich mal ignoriert

Was zeigt er bei Hyper Energy Programm sensor an wenn du den Smart Modus auf Smarte Leistungsregelung aktivierst.

mangri01, 2025-04-18:
Ich habe jetzt nochmal bei Konfig den Sensor verbrauch_strom_shelly_3em_pro_total_active_power hinzugefügt.
Vorther die smarten Leistungsmodus auf aus und nach der Konfig wieder auf ein gestellt.

Geht aber nicht

[image]

C0rd0n, 2025-04-18:
> Das sieht eigenlich alles OK aus! Wie viele Hyper2000 hast du? Mit einen sollte es eigenlich sofort funktionieren, mit mehrere braucht man die device cluster aus zu wahlen. [image] Die ZendureManager verteilt die strom ueber die verfugbare gerate, abhangig von die aktuelle kapacitat = charging = (socSet - electricityLevel) * packNum und discharging = (electricityLevel - minSoc) * packNum. Weil es nicht erlaubt ist mehr als 800w zu lieferen (ohne massnamen) begrenst die ZendureManager dass auf das ausgewahlte maximum. In die readme steht einiges daruber.
> 
> Wenn es immer noch nicht funktioniert, kannst du die 'debug logging' einschalten und das anschauen in de Settings/System/Log die integration logt mit welche Werte er rechnet, und kannst du ueberprufen ob das richtig ist!

Kann man bei einem Hyper auch das folgende auswählen, wenn der Hyper an einem eigenen Stromkreis hängt?
Wenn diese Integration funktioniert, sind weitere Hyper nicht ausgeschlossen.

[image]

MarcelF1, 2025-04-18:
Irgendwie werde ich nicht so richtig schlau aus der Integration habe jetzt den Smart Modus auf aus geschaltet und den AC Mode auf Eingang und lasse meinen Akku aus dem Netz laden soweit so gut

Jetzt is mir auf gefallen das in unregelmäßigen abständen der Smart Modus (Smarte Leistungsregelung) für 5 -10 Sekunden aktiv ist (blau) aber wie man am untern Balken sieht der smart modus is immer auf aus gestellt ?

Ich Zweifel langsam an mir selbst

das alles ist bei Version 35

[image]

FireSon, 2025-04-18:
> Kann man bei einem Hyper auch das folgende auswählen, wenn der Hyper an einem eigenen Stromkreis hängt? Wenn diese Integration funktioniert, sind weitere Hyper nicht ausgeschlossen.
>
> [image]

Genau diese einstellung wird automatisch gewahlt wenn man nur einen Hyper2000 hat. Bei mehrere hypers wird diese einstellung sichtbar.

mangri01, 2025-04-18:
Ich habe 4 x SolarFlow 1200, was soll ich da pro SolarFlow einstellen??

DANKE

harrymayr, 2025-04-18:
> Ich habe jetzt nochmal bei Konfig den Sensor verbrauch_strom_shelly_3em_pro_total_active_power hinzugefügt. Vorther die smarten Leistungsmodus auf aus und nach der Konfig wieder auf ein gestellt.
> 
> Geht aber nicht
> 
> <img alt="Image" width="409" src="https://private-user-images.githubusercontent.com/170202755/435152700-c84ead43-9a59-4cc3-848b-89cd1fcf16d7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQ5ODY2MzgsIm5iZiI6MTc0NDk4NjMzOCwicGF0aCI6Ii8xNzAyMDI3NTUvNDM1MTUyNzAwLWM4NGVhZDQzLTlhNTktNGNjMy04NDhiLTg5Y2QxZmNmMTZkNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNDE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDQxOFQxNDI1MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OWI3MDE2ZjgxNDNkZDlmNmI1YjVkMjI2NGIyZTBmYWY2Y2U1ZDFkMDY2Zjg2NDljZjZmM2I3MmFhMDg3OGZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.5WkGPTYAWkUEpmpJD15MXeQxlD-sXJKmN3LoJSO2C-0">

Siehe Post von heute (18.04.25 9:16):
[image]

mangri01, 2025-04-18:
Jetzt ist alles klar, perfekt danke

FireSon, 2025-04-18:
@harrymayr das sind schone bilder 😄 

Ich habe gerade cersion 36 erstellt, wobei einige Probleme mit die SmartMode behoben sind. Haubsachlich wenn man mit mehrere hypers arbeitet. 

@mangri01
Es ganz schwierig zu wissen wass genau bei dir passiert. Ich habe es gerade ueberprufft aber mit HA Smart Mode = OFF schickt die Manager keinen befehle an die gerate. Ich habe keinen SolarFlow 1200, hat das Gerat auch einen 'Smart Plug Mode'? Ich habe die identische befehle von die Hyper2000 in version 36 jetzt auch auch an die SolarFlow 1200 hinzugefugt. Kannst du das mal probieren? Wenn etwas nicht funktioniert kann ich eigenlich nur ueber einen Logdatei, mir anschauen wass das gerat macht. Kannst du vielleicht dir die debug logging anschauen? Da gibt es wichtige information ueber die steuerung! 
Einig vorsicht mit die inhallt ist gut, weil Serial nummer und Devicekey darin stehen!!

mangri01, 2025-04-18:
Gerne, woher bekomme ich die Version 36??

harrymayr, 2025-04-18:
> Gerne, woher bekomme ich die Version 36?? In Hacs zeigt es kein Update an

In HACS auf "Erneut herunterladen" gehen.

mangri01, 2025-04-18:
Hat Funktioniet, musste die Version 36 Manuell auswählen. 
Kann aber heute nicht mehr testen da meine Akku leer sind. Meine Kinder Backen für Ostern und meine 6 PV Module bringen nur 50w.
Also wird es erst morgen Nachmittag wieder weitergehen.

Vielen Dank und bis Morgen

mangri01, 2025-04-18:
Aber eine Frage habe ich noch, was soll ich bei meine 4 SolrFlow mit je einem Akku einstellen.

[image]

harrymayr, 2025-04-18:
Meines Wissens haben nur die Hyper 2000 (und Nachfolger) eine Cluster-Funktionalität. Da du die HUB 1200 hast (wenn ich das richtig gesehen habe), ist die Einstellung "eigener Stromkreis..." schon OK.

FireSon, 2025-04-18:
Nee das stimmt nicht ganz 😄 Es hat damit zu tun das wenn man mehrere gerate anschliest, es gezatslich nicht erlaubt ist mehr als 800w zuruck zu liefern. Aushalb wenn gerate mit ''eigener Stromkreis." angeschlossen sind. Das bedeutet das man ein Gerat mit einen eigen 'automat' in Verteilerkasten anschliest. Dann ist 2400w erlaubt, in die Niederlande auf jedem fall. 
Die ZendureManager verteilt die maximale strom ueber alle gerate in einen HA cluster. In gesamt liefert ein HA cluster nie mehr als das maximum. 
Mit 4 SolarFlow 1200 brauchst du wahrsheinlich vergleichbare massnahmen!

mangri01, 2025-04-18:
Habe es gerade nochmal testen können. Es zeigt mir zwar an das die Smart Leistungsregelung ein ist, es wird aber viel zuviel eingespeist.

Geht also nicht :-(

harrymayr, 2025-04-18:
... wieder was gelernt :-)
Ich versuche gerade rauszufinden, warum im Grenzbereich laufend in den Ruhezustand geschaltet wird. Ich habe mal das Debug-Log und die Zeitstempel für die einzelnen Datenpunkte von HA angehängt. Aber aus dem Log und dem Source-Code erschließt sich mir nicht, wie der Ruhezustand zustande kommt :-|

[image]

[history.csv](https://github.com/user-attachments/files/19813797/history.csv)

[Zendure-HA.log](https://github.com/user-attachments/files/19813799/Zendure-HA.log)

harrymayr, 2025-04-18:
> <img alt="Image" width="427" src="https://private-user-images.githubusercontent.com/170202755/435235247-fe91f407-a90d-4213-b1f2-09f49fdecb87.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQ5ODk5ODksIm5iZiI6MTc0NDk4OTY4OSwicGF0aCI6Ii8xNzAyMDI3NTUvNDM1MjM1MjQ3LWZlOTFmNDA3LWE5MGQtNDIxMy1iMWYyLTA5ZjQ5ZmRlY2I4Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNDE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDQxOFQxNTIxMjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yYmI5NjQ0Njc0Yzg1OTNkNjhjZTdkNTE5NTMwYTk0MTVhZjc5ODBlZTcwNTdkYzgwY2YyMDk0Yjk4MDdkYWViJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sNZUmQuhpXvvNBb5hiUWN1JlavqaIcAMamJ3PSH4Pco">
> Habe es gerade nochmal testen können. Es zeigt mir zwar an das die Smart Leistungsregelung ein ist, es wird aber viel zuviel eingespeist.
> 
> Geht also nicht :-(

Mit der Erklärung von FireSon oben, solltest du dann mal alle auf "Cluster Max 800" stellen, ggf. auch alle auf "Part of ..." geht ja darum, maximal 800W einzuspeisen.

mangri01, 2025-04-18:
Habe die Cluster der Akku auf 800 gestellt. 
Er Regelt schon hoch, aber sehr langsam. Aber wenn weniger Strom benötigt wird regelt er nicht mehr herunter.

[image]

harrymayr, 2025-04-18:
Wenn du das Debugging aus machst, müsste dein Browser eine Datei runterladen (geladen haben). Dateiname müsste so lauten wie: home-assistant_zendure_ha_2025-04-18T15-29-40.639Z.log. In der Datei nur die Zeilen mit "zendure_ha" nehmen, die Account-Relevanten Sachen rauslöschen wie z.B. "[custom_components.zendure_ha.zendurermanager] Receive: **K......5** => ready!" und dann hier reinstellen.

harrymayr, 2025-04-18:
Hmm, also Textdateien kann man sehr gut mit Notepad++ bearbeiten (https://notepad-plus-plus.org/downloads/)
Und die Log-Datei hat Zeitstempel, da kann man schon ganz schnell nur den relevanten Zeitraum auswählen. Und mit Suchen/Ersetzen (Strg. H) sind Einträge wie die Account-Daten schnell unkenntlich gemacht.
Eine private email stelle ich ungern ins Netz :-| und das Log kann FireSon wahrscheinlich besser auswerten wie ich ;-)

FireSon, 2025-04-18:
Kein problem: j.j.fireson at gmail.com

MarcelF1, 2025-04-18:
@FireSon 
So bei mir geht der Smart Modus Jetzt aber irgendwo ist in der Integration ein Fehler ich hoffe ich bekomme den gut beschrieben.

Wenn die integration neu gestartet oder ein update erfolgt ist wird ist der Punkt **select.hyper_2000_cluster** anwählbar und in dem Moment geht die Smarte Leistungsregelung nicht

Wenn ich dann den Home Assistant neu gestartet habe ist die Entität **select.hyper_2000_cluster** Nicht verfügbar und siehe da die Smarte Leistungsregelung tut genau was sie soll.

Ich habe 1. Hyper 200 also ohne Cluster

ich hoffe hier vielen damit zu helfen und vielleicht bekommt es @FireSon hin das irgendwann zu fixen das man nicht immer den Home Assistent Neustarten muss.

Nochmal Riesen Dank an dich für die Mühe

C0rd0n, 2025-04-18:
Bei mir geht’s jetzt mittlerweile auch. Ich habe meinen Sensor auf sensor.power_actual umbenannt. Jetzt scheint es wie gewünscht zu laufen.

mangri01, 2025-04-19:
> Kein problem: j.j.fireson at gmail.com

Hallo, gibt es schon etwas neues. Danke

mangri01, 2025-04-19:
Hallo Fireson, 

ich hatte ihnen eine Nachricht mit dem Debugging File gesenet. Gibt es schon etwas neue für die SolarFlow 1200.

DANKE

FireSon, 2025-04-19:
@mangri01 I had a look at your logfile. Everything seems to work properly, except that the SF1200 does not react to the power command. Fortunately Idid get some information on the specifics of these commands from Zendure. I will try to see if I get get them info the integation. I will let you know if you can test them!

mangri01, 2025-04-20:
Thanks for the reply. Let me know when you have a new version to test. Best regards, Manfred

mangri01, 2025-04-27:
Ich habe es heute mit der Versiojn 1.0.40 ausprobiert. Bei mir geht es nicht.  4 x SolarFlow 1200 Hub.

FireSon, 2025-05-03:
Wir sind jetzt in einen anderen Issue dabei. Schliesse diese Issue jetzt.

closed 2025-05-03 by FireSon