Zendure/Zendure-HA #111: Problem with Cluster

technoman63, 2025-04-18:
Hello,

I'm having a problem. As soon as I enable smart pairing on the integration, my cluster only uses half the value.

Example:

I output 1000W to the network. If I enable it, I'll have 500W in the network and 500W in the battery.

I've attached the logs.

Sincerely,

[home-assistant_zendure_ha_2025-04-18T15-53-29.165Z.log](https://github.com/user-attachments/files/19814168/home-assistant_zendure_ha_2025-04-18T15-53-29.165Z.log)

FireSon, 2025-04-18:
Which device cluster option did you select for the devices?

technoman63, 2025-04-18:
for the hyper n°2 Cluster max 2400w 
for the hyper n°1 part of hyper n°2

In the zendure app all my hyper are into a cluster

FireSon, 2025-04-18:
Ah ok, that is the problem, the integration does not (yet) support Zenlink clusters. The integration actually does exactly the same as the ZenCluster, distribute the power over all devices of the same cluster. So if you want to use the integration in that way, you have to take them out of the Zenlink cluster in the Zendure App

technoman63, 2025-04-19:
OK, I disconnected my cluster on the application, but it doesn't work at all. I have a request for electricity from my house and the battery doesn't take over. I put the logs

[home-assistant_zendure_ha_2025-04-19T06-41-30.916Z.log](https://github.com/user-attachments/files/19820061/home-assistant_zendure_ha_2025-04-19T06-41-30.916Z.log)

FireSon, 2025-04-19:
That is strange indead. Can you try to restart HA?

technoman63, 2025-04-20:
Yes i have restart HA restart but don't work 

What I'm seeing:

- No charging when there's a demand
- Charging works, but with an excess usage of about 150W from the network

technoman63, 2025-04-30:
For me with the new version it's OK. thank you for your work

closed 2025-04-30 by technoman63