Zendure/Zendure-HA #112: Wrong calculation of discharging power

hblockx, 2025-04-18:
I got 2 hyper, one with 2 AB2000 and one with one AB1000. Linked both in HA with the first hyper to be the cluster and the second one using it. While the config was charging perfectly fine with 1.0.35 but not discharging at all now it is discharging but the calculation is all wrong. Even with the limitation of 800w it is only discharging 450W although there are 2kw coming from the grid...

FireSon, 2025-04-19:
@hblockx Sorry to hear that. Unfortunately that is insufficient information for me to understand what your problem is. You cloud enable the debug logging and look at it. It shows which commands are sent to the devices, and information about the power distribution...

djodjo02130, 2025-04-19:
@FireSon Maybe same issue for me but i am using 1.0.36.

I got spikes and cannot be possible because i have only one AB2000S

[image]
[image]

GeneralRurra, 2025-04-20:
@FireSon 
[ENG] Yes, I also noticed that a sensor had not reset itself at 0 o'clock 
[DE] Ja was ich bei mir auch gemerkt habe das sich ein sensore nicht um 0 uhr zurück gesetzt hatte

[image]
[image]

hblockx, 2025-04-20:
[image]

#
`{'packInputPower': 158, 'hyperTmp': 3181}}
2025-04-20 19:54:05.921 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/SNAt0LaQ/properties/report => {'messageId': 5185436, 'product': 'solarFlowPro', 'deviceId': 'SNAt0LaQ', 'timestamp': 1745171643, 'properties': {'outputHomePower': 520}}
2025-04-20 19:54:06.540 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/properties/report => {'messageId': 1139129, 'product': 'solarFlowPro', 'deviceId': 'j143TaKV', 'timestamp': 1745171645, 'properties': {'packInputPower': 158}}
2025-04-20 19:54:07.569 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/properties/report => {'messageId': 1139131, 'product': 'solarFlowPro', 'deviceId': 'j143TaKV', 'timestamp': 1745171646, 'properties': {'packInputPower': 158, 'hyperTmp': 3171}}
2025-04-20 19:54:07.968 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/SNAt0LaQ/properties/report => {'messageId': 5185440, 'product': 'solarFlowPro', 'deviceId': 'SNAt0LaQ', 'timestamp': 1745171645, 'properties': {'remainOutTime': 316}}
2025-04-20 19:54:08.596 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/properties/report => {'messageId': 1139133, 'product': 'solarFlowPro', 'deviceId': 'j143TaKV', 'timestamp': 1745171647, 'properties': {'packInputPower': 158}}
2025-04-20 19:54:09.632 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/properties/report => {'messageId': 1139135, 'product': 'solarFlowPro', 'deviceId': 'j143TaKV', 'timestamp': 1745171648, 'properties': {'packInputPower': 158, 'remainOutTime': 483}}
2025-04-20 19:54:10.661 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/properties/report => {'messageId': 1139137, 'product': 'solarFlowPro', 'deviceId': 'j143TaKV', 'timestamp': 1745171649, 'properties': {'outputHomePower': 158}}
2025-04-20 19:54:10.699 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: -204 power: 676 operation: 2
2025-04-20 19:54:10.700 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000 136 = 2 * (73 - 5)
2025-04-20 19:54:10.700 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000s 2 73 = 1 * (78 - 5)
2025-04-20 19:54:10.700 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 472 stateManagerState.DISCHARGING capacity: 209 max: 1600
2025-04-20 19:54:10.700 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000s 2 => 164 capacity 73
2025-04-20 19:54:11.048 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/function/invoke/reply => {'messageId': 2873, 'deviceId': 'j143TaKV', 'timestamp': 1745171649, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}
2025-04-20 19:54:11.049 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Receive: j143TaKV => ready!
2025-04-20 19:54:11.871 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/properties/report => {'messageId': 1139139, 'product': 'solarFlowPro', 'deviceId': 'j143TaKV', 'timestamp': 1745171650, 'properties': {'packInputPower': 158, 'outputHomePower': 157}}
2025-04-20 19:54:12.555 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/SNAt0LaQ/properties/report => {'messageId': 5185446, 'product': 'solarFlowPro', 'deviceId': 'SNAt0LaQ', 'timestamp': 1745171648, 'properties': {'packInputPower': 520, 'outputHomePower': 521}}
2025-04-20 19:54:12.559 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/SNAt0LaQ/properties/report => {'messageId': 5185448, 'product': 'solarFlowPro', 'deviceId': 'SNAt0LaQ', 'timestamp': 1745171649, 'properties': {'electricLevel': 73, 'packInputPower': 520, 'outputHomePower': 520, 'packInputPowerCycle': 518, 'outputHomePowerCycle': 519}}
2025-04-20 19:54:12.902 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/properties/report => {'messageId': 1139141, 'product': 'solarFlowPro', 'deviceId': 'j143TaKV', 'timestamp': 1745171651, 'properties': {'packInputPower': 158}}
2025-04-20 19:54:13.109 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/SNAt0LaQ/properties/report => {'messageId': 5185450, 'product': 'solarFlowPro', 'deviceId': 'SNAt0LaQ', 'timestamp': 1745171650, 'properties': {'packInputPower': 520, 'remainOutTime': 308}}
[... cut ...]
2025-04-20 19:54:20.715 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000s 2 => 171 capacity 73
2025-04-20 19:54:21.062 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Topic: /gDa3tb/j143TaKV/function/invoke/reply => {'messageId': 2875, 'deviceId': 'j143TaKV', 'timestamp': 1745171659, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}
2025-04-20 19:54:21.062 INFO (paho-mqtt-client-75501:48842539-f179-438d-b578-0e0458691ae2) [custom_components.zendure_ha.zendurermanager] Receive: j143TaKV => ready!`

hblockx, 2025-04-20:
Seems like it is updating it and even mentioning that it is discharging too much "Update p1: -198 power: 688" other times it was not enough....

harrymayr, 2025-04-20:
For some reason, it seems that it don't update the 1st Hyper with the 2 AB2000. This output is always at 520W. The 2nd Hyper seems to updated correctly. 
I have only one Hyper in use, so this is a job for FireSon 😉

BTW, it's not the best idea to post the deviceId.

hblockx, 2025-04-20:
Ok I didn't know it is some kind of secret. I restarted the device, lets see, maybe that does change the issue.

hblockx, 2025-04-20:
Sadly the same issue. Either the calculation is wrong when two hyper are clustered or it is related to the first one having several packs, idk

harrymayr, 2025-04-25:
> Sadly the same issue. Either the calculation is wrong when two hyper are clustered or it is related to the first one having several packs, idk

Is this still an issue with the latest version?

closed 2025-05-05 by FireSon