Zendure/Zendure-HA #113: Hub2000 with Ace 1500 connect and control over Hub200

Coyotex1, 2025-04-18:
Hi,

So start my test system againg to test your hard work.

The Ace 1500 is after all updates not here in the log is no comment about fail device or so on.

The Hub2000 make nothing on any change

The energy managend do nothing.

Can you add a drop down menu to select the energie modus / plan witch all options (nothing,smart ct modus, intelligent modus, etc) what people want to use? When you add it and it do nothing in the app must on the phone must be all energy mode unselect then it work and it stay on the select mode. Over the iOBroker it works so perfecty.

Because add this state you integration is nice and you work very hard but at all its nice to have and see state but not more. But for people with know knowledge to make selfe a control its very good and i thing many people have wait for it. I hope you get much spends for it.

FireSon, 2025-04-19:
@Coyotex1 I will try to add the nothing,smart ct modus, intelligent modus. It is however quite difficult since I cannot test it properly. So it is more or less 'blind' developement. Fortunately I got some info from Zendure about the device specific commands which I probably can use to achieve it!

z-master42, 2025-04-20:
The commands for the different device types should be the same, only the product category differs.

Hub1200: 73bkTV
Hub2000: A8yh63
Hyper 2000: ja72U0ha

FireSon, 2025-05-10:
Can you test the pre-6 release, the necessary commands have been added.

closed 2025-05-11 by FireSon