Zendure/Zendure-HA #123: Make entities more human-readable

z-master42, 2025-04-21:
I only have a Hub1200, but that's where I come from.

autoRecover contains the information whether the bypass should be reset to the original setting the next day. 0: Off. 1: On.

SoH for the batteries stands for State of Health and is therefore a percentage value. The transmitted value must be divided by 10. Example 994 = 99.4 % (100 % is logically the maximum value)

hubState contains the information on what the hub should do when the discharge limit is reached. 1: Stop output and switch off. 0: Stop output and standby.

pass contains the information as to whether the hub is in bypass mode. 1: On. 0: Off.

passMode contains information on which bypass mode is set. 0: Automatic, 1: Always off, 2: Always on.

FireSon, 2025-04-23:
@z-master42 Thanks for the info! Fixed this in the next release. Please reopen the issue if something is not correct.

closed 2025-04-23 by FireSon

z-master42, 2025-04-27:
You have currently created a `soh` entity in `hub1200.py` and `hub2000.py`. This does not exist in Zendure (therefore also shows permanently unknown).

What I meant in the markup for this issue is one soh entity per battery. This does not appear in your battery definition in `zenduredevice.py` either, but it is created for me.

Question in this context: 
 Does `batcur` really exist for batteries? It is always unknown to me. I haven't seen the battery currents anywhere else either.

As I said, I only have one Hub1200 (`hub1200.py`):
What is `BatVolt` supposed to represent? It is always unknown to me. Every battery has its own voltage entities.
What does `reverseState` represent?
What does `energyPower` represent?
`strength`, i.e. the WLAN signal strength is not available on the Hub1200. It is always unknown.

Linus86529, 2025-04-27:
Hello, I have a Hub1200 with 4 x AB1000. And a Hyper2000 with 1x AB2000. I get the soh value for every battery attached to the Hub1200. 
Battery #1 = 991, Battery #2 = 989, Battery #3 = 991 and Battery #4 = 991. 

I would like to share a screen shot with you but unfortunately I don't see the attachment button on this github page.

But I don't get the soh for the AB2000 attached to the Hyper2000.

Linus86529, 2025-04-27:
Hello, these sensors show "unknown" for AB1000 batteries attached to the Hub1200:
sensor.hub_1200_bat_volt
sensor.hub_1200_battery_1_batcur
sensor.hub_1200_battery_2_batcur
sensor.hub_1200_battery_3_batcur
sensor.hub_1200_battery_4_batcur 

The battery attached to the Hyper2000 has a value for:
sensor.hyper_2000_battery_1_batcur = 9,7A