Zendure/Zendure-HA #124: strange battery state

FrColin, 2025-04-21:
Hi,
I have one hyper 2000 stacked on two ab2000 battery.
This morning i find the system in a strange state.
Mode smart but no output to home, one battery charging , one discharging , temperature alarm on the zendure app.
Difficulties to recover a fonctional system 
Manual mode didn't change the state, smart do not produce output , if output limit is set the output goes to the maximum and régulation as no effect on output.

Should the input and output set to 0 on smart mode
The zendure is blocking those limit when smart ct mode ?
Should the smart ct mode be activated on the zendure app ?

Could you clarify the interaction between the zenapp and the integration ?

Thanks a lot for your work and help

FrColin, 2025-04-23:
Ok if the trouble is not fréquent.
I dont think that the bluetooth would be a solution because the ha host may not have bluetooth and not be close to the hyper 2000.
In my case ha is at home and hyper is located in the pool housse.

Milvus11, 2025-04-25:
Good morning, thank you @FireSon for the info. I also faces issues with communication two days in a row. It gets stuck/stops functioning properly and shows at night, when dark, still that it is receiving solar power. Now I know it is not an issue with home assistant:)

closed 2025-05-11 by FireSon