Zendure/Zendure-HA #129: smart mode will not switch back to discharging until soloar production

FrColin, 2025-04-22:
Hi ,
with one device the smart mode discharging state will regulate the home output as expected , but when its goes to charging it will never return to DISCHARGING until battery full.
when charging  the `d.powerAct = d.asInt("packInputPower") - d.asInt("outputPackPower")` will be negative since no output from battery pack.
as it's negative the mode will stay CHARGING unless the input power stop ie battery full

`2025-04-22 12:45:56.346 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 72 stateManagerState.DISCHARGING capacity: 180 max: 1200
2025-04-22 12:46:09.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 74 power: 0 operation: 2
2025-04-22 12:46:09.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update state: p1: 74
2025-04-22 12:46:09.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000 180 = 2 * (100 - 10)
2025-04-22 12:46:09.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 74 stateManagerState.DISCHARGING capacity: 180 max: 1200
2025-04-22 12:46:23.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 73 power: 0 operation: 2
2025-04-22 12:46:23.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update state: p1: 73
2025-04-22 12:46:23.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000 180 = 2 * (100 - 10)
2025-04-22 12:46:23.334 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 73 stateManagerState.DISCHARGING capacity: 180 max: 1200
2025-04-22 12:46:32.324 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: -94 power: 0 operation: 2
2025-04-22 12:46:32.324 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update state: p1: -94
2025-04-22 12:46:32.324 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000 0 = 2 * (100 - 100)
2025-04-22 12:46:32.324 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: -94 stateManagerState.CHARGING capacity: 0 max: 1200
2025-04-22 12:46:40.319 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: -86 power: 0 operation: 2
2025-04-22 12:46:40.319 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update state: p1: -86
2025-04-22 12:46:40.319 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000 0 = 2 * (100 - 100)
2025-04-22 12:46:40.319 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: -86 stateManagerState.CHARGING capacity: 0 max: 1200
2025-04-22 12:46:52.434 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
`

FrColin, 2025-04-22:
May be the following test on powerActual should be on `powerActual + p1`
` elif powerActual < 0:
                self.updateSetpoint(powerActual + p1 + 50, ManagerState.CHARGING)
            elif powerActual > 0:
                self.updateSetpoint(powerActual + p1, ManagerState.DISCHARGING)`

FireSon, 2025-04-23:
@FrColin well, I cannot reproduce taht behaviour. I once saw that my battery was doing a auto calibration, and since my limit is 96% it never reached 100% and was not able to discharge. When I saw it I added a better description to socStatus, where it indicates it it does auto-calibration.
The code you are referring to is correct (as far as I can see) because if it is positive and the hypers are charging, first the hypers will be set to zero, and after 10 seconds they should start to discharge. I have seen it many time function correctly. It might me a problem if one of your hypers does not respond properly, then the poweractual will never be zero.

FrColin, 2025-04-23:
Today the same behavior batterie left 78%
And output never come to discharge the batterie.
I have only one hyper and it is responding well.
Since the actual power is négative the system stay in charging state

FrColin, 2025-04-24:
hi, an other point of this trouble:
I use automation to switch zendure_manager_operation from smart to off in different situation.
restoring state from stop to smart didn't make smart mode to work effectively, 
switching from stop -> manual -> smart seems to re-engage smart mode correctly.
Also I see in comment that the solarInput is not take in account in the calculation.

I think in case off a single device the outputPower as no need to be balanced on multiple devices and so no need to do complexe compute on battery state. a simple method used was to keep the last outputPower and adjust the output depending on p1meter

something like 
`variables:
  ancienne_limite: "{{ states('sensor.hyper_2000_output_home_power') | float(0) }}"
  em0_p: "{{ states('sensor.shellyproem50_08f9e0e60c18_em0_power') | float(0) }}"
  em1_p: "{{ states('sensor.shellyproem50_08f9e0e60c18_em1_power') | float(0) }}"
  grid_sum: "{{ (em0_p +  em1_p) }}"
  nouvelle_limite: "{{ (grid_sum + ancienne_limite )  | round(0) }}"
  setpoint: "{{ ( [ 0.0,nouvelle_limite,1200.0]|sort)[1] }}"
alias: Calcul variables`

hope this can help

FireSon, 2025-04-25:
@FrColin Can you check if your problem still exists in version 40? Please reopen this issue if it is still the case.

closed 2025-04-25 by FireSon

FrColin, 2025-04-27:
Hello, the problem is solved in version 40 , thanks a lot.
I was facing to another problem , I doesn't known if it is due to the integration or the hyper200
the hyper was not responsive from 17h yesterday until a switch off the hyper2000 this morning

[image]