Zendure/Zendure-HA #132: Batteries as separate HA devices

danez, 2025-04-23:
I was playing around and wondered if it would make sense to have Batteries as separate devices in HomeAssistant. 

Do you think this makes sense and would be a good change?

[image]

[image]

FireSon, 2025-04-23:
That was the original idea 😄 
Since you cannot have parent-child devices in HA (I could not fin it), the problem is the 'simple' link between the two, which can than only be done by serial number... There is not much between the two solutions.

danez, 2025-04-23:
There is the `via_device` link in a device, to tell through which device it is connected:

> Identifier of a device that routes messages between this device and Home Assistant. Examples of such devices are hubs, or parent devices of a sub-device. This is used to show device topology in Home Assistant.

From the docs

But other than the simple link on the device page `Connected via Hub 2000` (see screenshot) nothing else happens when setting it.

Is the link important?

danez, 2025-04-23:
Oh and there is a small box on the device page: 

[image]

danez, 2025-04-23:
I have a prototype already and can submit it.

FireSon, 2025-04-23:
That would be great! It can be that other Zendure devices can have more than 4 batteries, which would lead to more properties and difficulties 😄

FireSon, 2025-04-25:
@danez Can you make a pr from your prototype? I think your solution is the better way to move forward...

closed 2025-04-25 by FireSon