Zendure/Zendure-HA #135: During charging, the Gridinput is still being used :-(

Wunni73, 2025-04-23:
[image]

Why gridInputPower ???

FireSon, 2025-04-23:
@Wunni73 As alway I am happy to help, but I cannot see exactly what you are doing and with which settings. I have not connected any solarpanels to my hyper (yet), so it is difficult for me to test. If you use the smart matching mode, it will use the available electricity and store it, also if the hyper has its own solarpanel. 
But please explain it in a bit more detail ☺️

harrymayr, 2025-04-23:
@FireSon: I assume the outputPackPower includes the solarInputPower, so the calculation of the powerActual (which is in the moment more or less the input or output from/to the grid) is wrong. For calculating the setpoint, the solarInputPower may be substracted.
I also have no panels connected to the hyper, so I can't test this assumption.

harrymayr, 2025-04-23:
And also the priority for the power from connected panels is not yet implemented. It charges at the beginnig with the full power from the panels. The App has the priority House, Battery, feed into grid.

FrColin, 2025-04-24:
> [@FireSon](https://github.com/FireSon): I assume the outputPackPower includes the solarInputPower, so the calculation of the powerActual (which is in the moment more or less the input or output [...]

yes the outputPackPower includes solarInputPower and may also includes gridInputPower

Wunni73, 2025-04-24:
@FireSon I have a Hyper2000 with 4x800W solar power. and I actually want 100% solar for the Hyper. and not gidinput for 35 cents. 
and secondly he stops with Smart-Matching-Modus when the sun comes out.

Wunni73, 2025-04-24:
[image]
And there's also an error in the formula.
Grid is the same as solar??
Actually, it should be grid zero, and solar + battery = home.
Smart-Matching-Modus

FireSon, 2025-04-24:
In version 37, I have substracted the solarInputPower from the poerActual calculation.

Wunni73, 2025-04-25:
@FireSon  I Use 39 and it still works perfekt 👍 
, the is no Solar problem more .

FireSon, 2025-04-25:
Ahh great news!! Also thanks to @harrymayr for the help 😄 
Close the issue for now!

closed 2025-04-25 by FireSon