Zendure/Zendure-HA #141: smart mode produce wrong  outPower

FrColin, 2025-04-24:
in the state battery full , home grid ask for 467W, ouput power is only 150W
as you can see in the log
`zendurermanager] Update p1: 467 power: 0 operation: 2`
`zendurermanager] Update state: p1: 467`
`zendurermanager] Update capacity: Hyper 2000 178 = 2 * (99 - 10)`
`zendurermanager] Update setpoint: 467 stateManagerState.DISCHARGING capacity: 178 max: 1200`
`devices.hyper2000] Update power Hyper 2000 => 467 capacity 178`
`zendurermanager] Topic: /gDa3tb/xxxxxxxx/properties/report => {'messageId': 7330664, 'product': 'solarFlowPro', 'deviceId': 'xxxxxxxx', 'timestamp': 1745509991, 'properties': {'electricLevel': 99, 'solarPower1Cycle': 86, 'solarPower2Cycle': 40, 'outputHomePowerCycle': 104}}`
`zendurermanager] Topic: /gDa3tb/xxxxxxxx/properties/report => {'messageId': 7330666, 'product': 'solarFlowPro', 'deviceId': 'xxxxxxxx', 'timestamp': 1745509992, 'properties': {'outputHomePower': 149}}`
`zendurermanager] Topic: /gDa3tb/xxxxxxxx/properties/report => {'messageId': 7330668, 'product': 'solarFlowPro', 'deviceId': 'xxxxxxxx', 'timestamp': 1745509993, 'properties': {'solarInputPower': 184, 'solarPower1': 126}}`
`zendurermanager] Topic: /gDa3tb/xxxxxxxx/properties/report => {'messageId': 7330670, 'product': 'solarFlowPro', 'deviceId': 'xxxxxxxx', 'timestamp': 1745509994, 'properties': {'outputHomePower': 150}}`
```
zendurermanager] Topic: /gDa3tb/xxxxxxxx/function/invoke/reply => {'messageId': 13448, 'deviceId': 'xxxxxxxx', 'timestamp': 1745509997, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}
zendurermanager] Receive: xxxxxxxx => ready!
```

JannickWe, 2025-04-24:
I had the same issue using version 1.0.36 in manual mode. Reinstalling (remove via HACS, reboot and clean install) resolved the issue.

FireSon, 2025-04-24:
@FrColin I have had the same issue mentioned by multiple users. So it would be very helpful to know what causes this!! The calculated power 467w is correct looking at the value from the p1 meter. Your hyper however is only producing 150w allthough 467 is sent. Is that the correct conclusion?

Linus86529, 2025-04-24:
Ich habe die selbe Erfahrung gemacht. Nachdem ich die Version 1.0.36 komplett entfernt habe und HA komplett rebootet habe. Hat eine Neuinstallation das Problem behoben. Eine Unklarheit die mich noch immer beschäftigt ist der Konfigurationspunkt "lokaler Broker" ? Was muss dort eingegeben werden? Ich habe dort alles mögliche eingegeben. Die IP des RasPi auf dem HA installiert ist, den Full Quallified Domain Name, und den mosquiito. Mit letzterem läuft es gerade interessanterweise wie ich das erwarte  

Also, wenn wie heute beobachtet der Leistungsbezug konstant bei 2000Watt über längere Zeit konstant ist, dann speist der Hyper2000 (1xAB2000, 3x210Wp) nach ca. 10 Sekunden 400Watt ein. Nach weiteren 10 Sekunden fängt der Hub1200 (4xAB1000, 2x415Wp) an einzuspeisen. Es dauert aber sehr lange bis der Hub1200 seine 400Watt einspeist.

FrColin, 2025-04-24:
> [@FrColin](https://github.com/FrColin) I have had the same issue mentioned by multiple users. So it would be very helpful to know what causes this!! The calculated power 467w is correct looking at [...]

Yes this is correct but i can't see in the log if the transmited value is correct, looking at the code it should be ok. The log show the succes reply but i dont know if  it is the reply of the previous sent message or an other because we dont know the message id of the sent message . I see That the message id of the invoke/reply is far away from the properties one.

FireSon, 2025-04-25:
@FrColin I think this should be fixed in version 39, can you test it?

FrColin, 2025-04-27:
hi, this is fine now