Zendure/Zendure-HA #149: 'zendure_ha' closes the Home Assistant aiohttp session

hblockx, 2025-04-25:
5-04-25 09:04:13.154 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'zendure_ha' closes the Home Assistant aiohttp session at custom_components/zendure_ha/api.py, line 91: await self.session.close(). Please create a bug report at https://github.com/FireSon/Zendure-HA/issues

hblockx, 2025-04-25:
1.0.38 fresh installed

FireSon, 2025-04-25:
That is strange, it closes the http connection to the ZendureApi, since it is not needed anymore. This connection was created by the api itself...

AndreR84, 2025-04-25:
From 1.0.36 to 1.0.38. Before no Problems

WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'zendure_ha' closes the Home Assistant aiohttp session at custom_components/zendure_ha/api.py, line 91: await self.session.close(). Please create a bug report at https://github.com/FireSon/Zendure-HA/issues

FireSon, 2025-04-25:
Should be fixed in version 40

closed 2025-04-25 by FireSon