Zendure/Zendure-HA #151: duplicate tension battery on Hyper2000

FrColin, 2025-04-25:
Hi,
thanks a lot for the version 1.0.38, it works much better .

i discover a duplicate entities sensor.hyper_2000_bat_volt (mdi:sine-wave) and sensor.hyper_2000_bat_volt_2 (mdi:eye)
the _2 as no unit it could be the old sensor
I cannot delete it, reinstall Integration did not solve the problem
looking at the code I see a minor difference between other devices

devices/ace1500.py:            self.sensor("batVolt", None, "V", "voltage"),
devices/aio2400.py:            self.sensor("batVolt", None, "V", "voltage"),
devices/hub1000.py:            self.sensor("batVolt", None, "V", "voltage"),
devices/hub1200.py:            self.sensor("batVolt", None, "V", "voltage"),
devices/hub2000.py:            self.sensor("batVolt", None, "V", "voltage"),
devices/hyper2000.py:            self.sensor("batVolt", None, "V", "voltage", **"measurement"**),
devices/solarflow2400ac.py:            self.sensor("batVolt", None, "V", "voltage"),

FireSon, 2025-04-25:
@FrColin thanks again for the sharp eye! I made all sensors the same as the Hyper2000. I also have both sensors and will look into it.

FireSon, 2025-04-25:
@FrColin should be fixed in version 40

closed 2025-04-25 by FireSon