Zendure/Zendure-HA #152: Zendure-HA stopped working with 1.0.39

hblockx, 2025-04-25:
stopped working
-04-25 18:36:28.758 ERROR (paho-mqtt-client-75501:f24efbac-3ced-4dab-9e67-4ce6e5a65b9c) [custom_components.zendure_ha.zenduredevice] 'NoneType' object has no attribute 'default_language_platform_translations'
2025-04-25 18:36:28.763 ERROR (paho-mqtt-client-75501:f24efbac-3ced-4dab-9e67-4ce6e5a65b9c) [custom_components.zendure_ha.zenduredevice] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 141, in message
    self.updateProperty(key, value)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/devices/hyper2000.py", line 94, in updateProperty
    if not super().updateProperty(key, value):
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 212, in updateProperty
    self.update_energy(1, int(value))
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 227, in update_energy
    kWh += float(self.powerSensors[idx].state) if self.powerSensors[idx].state is not None and time.day == self.totaltime[idx].day else 0
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 520, in state
    unit_of_measurement = self.unit_of_measurement
                          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 501, in unit_of_measurement
    := self.platform.default_language_platform_translations.get(translation_key)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'default_language_platform_translations'

2025-04-25 18:36:29.787 ERROR (paho-mqtt-client-75501:f24efbac-3ced-4dab-9e67-4ce6e5a65b9c) [custom_components.zendure_ha.zenduredevice] 'NoneType' object has no attribute 'default_language_platform_translations'
2025-04-25 18:36:29.791 ERROR (paho-mqtt-client-75501:f24efbac-3ced-4dab-9e67-4ce6e5a65b9c) [custom_components.zendure_ha.zenduredevice] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 141, in message
    self.updateProperty(key, value)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/devices/hyper2000.py", line 94, in updateProperty
    if not super().updateProperty(key, value):
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 212, in updateProperty
    self.update_energy(1, int(value))
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 227, in update_energy
    kWh += float(self.powerSensors[idx].state) if self.powerSensors[idx].state is not None and time.day == self.totaltime[idx].day else 0
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 520, in state
    unit_of_measurement = self.unit_of_measurement
                          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 501, in unit_of_measurement
    := self.platform.default_language_platform_translations.get(translation_key)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

FireSon, 2025-04-25:
Have you deleted and added the configuration?

hblockx, 2025-04-25:
Yes

FireSon, 2025-04-25:
@hblockx fixed this in version 40. It might still produce an error but it should continue. Please reopen this issue if the problem is still there!

closed 2025-04-25 by FireSon