Zendure/Zendure-HA #154: 1.0.40 does nothing

hblockx, 2025-04-25:
it actually stopped doing anything now...

5-04-26 00:17:22.665 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-04-26 00:17:22.668 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching zendure_ha 
[custom_components.zendure_ha.zenduredevice] Receive:  ... => Ready (a few times)

then nothing in the logs anymore

GeneralRurra, 2025-04-26:
What kind of system do you have Hyper or something similar. 
Have you already tried to redo the configuration?

hblockx, 2025-04-26:
Yeah 2 Hyper in a cluster. Updated from 39

FireSon, 2025-05-03:
This should be solved in version 41-pre4 please check if that one works. Closing this issue for now, but please reopen if the problem persists

closed 2025-05-03 by FireSon