Zendure/Zendure-HA #156: Zendure Manager does not anything at all

Kieft-C, 2025-04-26:
May I got done something wrong from the scratch.
I was using the integration from very the beginning and updating until now the newest version 1.0.40
But never the Zendure Manager works with my Hyper2000.

I use normally in the zendure App the "intelligent mode" (or named ct mode). And all work fine.
I got a shelly 3em pro as power sensor and put "sensor.shellypro3em_total_active_power" into the configuration.

Then I try to stop the intelligent mode from the zendure manager with "Off" and nothing happens.
Then I try it the other way around, stop the intelligent mode in the zendure app.
And try to start smart mode in the HA Integration, nothing happend.

Then I try manual regulation in the HA/ZA with 50W, then the app gives me feedback, that "smart mode" is starting with the 50W.
But i got in the app to check the mode, no mode is active there..

I'm confused and can't control really anything.
The SoC Max / Min works totally good. But nothing more.

Any Idea? or how I can make a log report, in the "debug protocol" I found not really points which interact with Zendure-HA.

pjeterinfo, 2025-04-26:
If you want to use the HA integration you should not use any mode from the app. Because you see a usage of 50W the intergration seems to work at least, but sometimes the app is not showing realtime update , at least for me. Sometime I have to restart the app to be able to show the actual state.

For smart mode I see the same issue on 1.0.40 smart mode does not start charging or uncharge. 

You could create a debug logging in HA selecting the intergration in devices/services. Then enable debug logging in the main page.
go to Settings / system / logs and select "show raw logs"

Linus86529, 2025-04-26:
@pjeterinfo and @fireson
When I turn off all modes in the Zendure app for the Hyper2000 and the Hub1200, which gets the power values from the Shelly 3EM Pro. Neither the Hub1200 nor the Hyper2000 do anything. Both are just charging their batteries.

In the Zendure-Ha integration both devices show in the sensor “sensor.hyper_2000_auto_model” = NONE and “sensor.hub_1200_auto_model” = NONE!
The Zendure Manager “select.zendure_manager_operation” is set to ‘SMART’ and nothing happens until the Hub 1200 is set to “Smart CT Mode” again in the Zendure App. I also have to change the Hyper2000 to “Smart CT”. 

Then set both devices in the sensor “sensor.hyper_2000_auto_mode” and in the sensor “sensor.hub_1200_auto_model” to “Smart CT power control”.

Who controls my Zendure HW? The Zendure Manager or the app?

It would be time to clarify for each user how the Zendure devices must be set in the APP so that the Zendure Manager functions work correctly.

Translated with DeepL.com (free version)

Kieft-C, 2025-04-26:
```
2025-04-26 13:30:26.331 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-04-26 13:30:26.332 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching zendure_ha (Zendure Integration - t***.ipad@gmail.com) data in 0.001 seconds (success: True)
2025-04-26 13:30:27.355 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:30:43.589 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:30:46.417 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:30:55.540 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 2
2025-04-26 13:30:55.540 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => no action [power 0 capacity 156]
2025-04-26 13:31:01.243 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 2 from: 0
2025-04-26 13:31:36.656 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:31:43.987 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:31:56.331 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-04-26 13:31:56.333 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching zendure_ha (Zendure Integration - t***.ipad@gmail.com) data in 0.001 seconds (success: True)
2025-04-26 13:31:57.355 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:32:17.603 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:32:32.017 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 2
2025-04-26 13:32:32.017 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => no action [power 0 capacity 156]
2025-04-26 13:32:42.781 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 2 from: 0
2025-04-26 13:33:13.769 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 2
2025-04-26 13:33:13.769 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => no action [power 0 capacity 156]
2025-04-26 13:33:26.331 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-04-26 13:33:26.333 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching zendure_ha (Zendure Integration - t***.ipad@gmail.com) data in 0.002 seconds (success: True)
2025-04-26 13:33:27.117 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 1 from: 0
2025-04-26 13:33:27.118 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => no action [power 0 capacity 156]
2025-04-26 13:33:27.400 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:33:34.656 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update capacity: Hyper 2000 219 = 3 * (78 - 5)
2025-04-26 13:33:34.657 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 50 stateManagerState.DISCHARGING capacity: 219 max: 1200
2025-04-26 13:33:34.657 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update power: Hyper 2000 50
2025-04-26 13:33:34.657 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Function invoke Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 50}, "msgType": 1, "autoModel": 8}], "deviceKey": "K*******", "function": "deviceAutomation", "messageId": 1, "timestamp": 1745667214}
2025-04-26 13:33:35.738 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:34:04.820 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 2 from: 1
2025-04-26 13:34:16.017 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:34:56.331 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-04-26 13:34:56.332 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching zendure_ha (Zendure Integration - t***.ipad@gmail.com) data in 0.001 seconds (success: True)
2025-04-26 13:34:57.355 INFO (paho-mqtt-client-53937:063ecca6-60b7-4150-a011-3b97c489c38c) [custom_components.zendure_ha.zenduredevice] Receive: K******* => ready!
2025-04-26 13:35:08.991 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 1 from: 2
[... cut ...]
Setting up the sensor to "sensor.shellypro3em_***_total_active_power" but it feels like, that this allready not work, because always when i click again on "configurate" it is not shown my given sensor. So may there is a chance to get in the integration a feedback sensor which is used?

But general after it, and go from "off" to "smartmode" nothing happens.
Only with manual mode after put a value in, it starts with it, but the smartmode not work for me.
I jump a little bit back and for to give more data in the log, and then go back in the zendure app to the standard ct-mode, because nothing really work.

FireSon, 2025-04-26:
It looks like the operation mode if the Zendure Manager changes all the time:
2025-04-26 13:35:52.466 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 2 from: 1
2025-04-26 13:35:56.270 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 2
That should NOT be the case. This means that manually or through some automation it is manipulated. Only in the operation mode 2 the smart matching is active. Changing to 0 (OFF) will set power = 0

closed 2025-05-12 by FireSon