Zendure/Zendure-HA #158: ACE1500 wird leider nicht angezeigt

Rubberduck73, 2025-04-26:
@FireSon 
Hi, danke schon mal für die bisherige Arbeit, ist echt Super.🤩
Das hat echt was gebracht für HA.

Leider wird auch bei mir kein ACE angezeigt, habe denn Debuglog  mal gestartet und da ist der ACE zu finden. 
Aber kein "Device XXXX is not supported" ... woran kann das liegen ?

Ich hätte noch Verbesserungsvorschläge.
Bei mir läuft jetzt die .40 für meinen Hub2000.
Soweit auch echt gut. 
Nur 4 Sensoren fehlt noch die Einheiten. 
Und zwar  bei "exitPassTime" , "gridPower" , "solarPower1Cycle" und "..2Cycle" da wäre es klasse wenn das noch nach kommt ☺️

Auch im Fenster der Steuerelemente... da wäre es schön wenn unter jedem Einsteller der aktuelle Wert stehen würde. Oder statt des Schiebereglers, der eingestellt Wert da stände und man dann ändern kann in dem man bei soc anstatt 100 da 90 einträgt. Dann hätte man einen schnelleren Überblick, auch lässt es sich besser so einstellen. Wie ich persönlich finde, da es übers Handy immer etwas fummelig ist.

Aber nochmal ein großes Lob für die bisherige Arbeit und macht bitte weiter so

Rubberduck73, 2025-05-02:
Hi, hatte mich gewundert das ich keine Antwort bekommen habe.. bis ich denn Link im Postfach gesehen habe !

Wen ich das Richtig Verstanden habe wird der ACE nicht als einzelnen Gerät angezeigt wenn er mit einem Hub verbunden ist.
OK, aber müsste ich denn nicht Sensoren vom ACE sehen können, zb. die Verbraucher an denn 2 AC Steckdosen oder denn DC/XT Aus/Eingang. Auch sind keine Steuerelemente vorhanden. 

MfG Rubberduck

FireSon, 2025-05-02:
Das stimmt! Es gibt schon eine losung dafur, aber ich habe noch keinen Zeit dafur gehabt. Ist aber auf die ToDo Liste 😄

Rubberduck73, 2025-05-02:
Ah das hört sich gut an... ist er dann so steuerbar wie über die APP was die Steckdosen und denn XT angeht ?

FireSon, 2025-05-03:
@Rubberduck73 kannst du das mal versuchen mit version 41-pre4?

kutzera, 2025-05-04:
> [@Rubberduck73](https://github.com/Rubberduck73) kannst du das mal versuchen mit version 41-pre4?

Ich habe es mal ausprobiert. Leider wird der ACE nach wie vor nicht angezeigt.

Rubberduck73, 2025-05-04:
@FireSon habe es grade mal ausprobiert, aber die Version zeigt mir meine Hub überhaupt nicht an. Nur der Manager ist da !

FireSon, 2025-05-05:
@Rubberduck73 Um das zu beheben ware es wichtig die logdatei, hauptsächlich die initialisierung, zu sehen, kannst du das teilen?

Rubberduck73, 2025-05-05:
@FireSon ja das würde ich hin bekommen, wie muss ich da vorgehen, bin da noch nicht so tief drin ?

kutzera, 2025-05-06:
Kann mir vielleicht erklären, was es mit dem local MQTT Schalter auf sich hat?

[image]

FireSon, 2025-05-06:
@Rubberduck73 
Select the Enable debug logging here:
[image]

Restart HA and after that click disable debug logging, and a file is generated. Since it contains some elements which you should share on github, you can sent it to j.j.fireson at gmail.com

Rubberduck73, 2025-05-08:
@FireSon Hi,

so habe Log Datei erstellt und an die Mail geschickt. 

MfG Michael

Rubberduck73, 2025-05-09:
@FireSon  danke für die Antwort, dann bin ich mal auf die Nächte Verson gespannt :)

Rubberduck73, 2025-05-11:
@FireSon Hi,
habe gestern mal kurz die pri6 installiert.
Angezeigt wird mir jetzt der Hub2000, die 3 Akkus als eigene Geräte und er ACE als eigenes Gerät, wobei viele Sensoren 
gleich sind wie im Hub. Was noch fehlt sind hier und da Einheiten, wie Watt bei der AC Steckdose des ACE.
Und die Steuerelemente sind ebenfalls die gleichen wie/die beim/vom Hub, aber die vom ACE fehlen noch.
Danke für die ständige weiter Entwicklung.  :)

kutzera, 2025-05-11:
Das as verstehe ich nicht. Wieso wird bei dir der ACE angezeigt und bei mir nicht. Bei mir fehlen auch die Batterien. Habe bei nach der Installation die konfig neu gestartet. Gibt es die Probleme, wenn man mehr als ein Gerät hat? Ich verstehe es einfach nicht

Rubberduck73, 2025-05-12:
@kutzera hat bei mir vor der 6er auch nicht geklappt. Frag mal direkt bei FireSon an. 
Eventuell kann er da weiter helfen !

FireSon, 2025-05-12:
Nice, close this one for now!

closed 2025-05-12 by FireSon