Zendure/Zendure-HA #161: V1.0.40 | api.py |2 KRITISCHE FEHLER |  Siehe mqtt Methode:

Linus86529, 2025-04-26:
KRITISCHER FEHLER GEFUNDEN: Diese Methode ruft sich selbst rekursiv auf (return self.mqtt(...)). Das ist ein Implementierungsfehler. An dieser Stelle sollte nicht self.mqtt aufgerufen werden, sondern der Konstruktor des paho.mqtt.client (return mqtt_client.Client(...)). Mit dem Code, wie er hier steht, kann der MQTT-Client nicht korrekt erstellt werden!

KRITISCHER FEHLER GEFUNDEN: Weiter unten werden client.loop() und client.loop_start() gleichzeitig aufgerufen. Dies ist eine falsche Verwendung der paho-mqtt-Bibliothek. loop_start() startet den Loop in einem Hintergrundthread und ist die korrekte Methode für die Integration in eine Event-Loop wie in Home Assistant. 1 Der Aufruf von client.loop() danach ist redundant und potenziell problematisch. Auch dieser Fehler kann dazu führen, dass der MQTT-Client nicht zuverlässig funktioniert oder die Integration blockiert!   

Dokumentation: https://eclipse.dev/paho/files/paho.mqtt.python/html/client.html

Die api.py-Datei ist für die Übertragung der Befehle zuständig. Die kritischen Fehler hier könnten erklären, warum die Kommunikation manchmal generell Probleme macht oder abbricht.

FireSon, 2025-04-26:
Ein bischen ruhiger bitte!

closed 2025-04-26 by FireSon

Linus86529, 2025-04-26:
Gerne doch

Linus86529, 2025-04-26:
def mqtt(self, clientId: str, username: str, password: str, onMessage: Callable) -> mqtt_client.Client:
        _LOGGER.info(f"Create mqtt client!! {clientId}")
        # Fehler behoben: paho.mqtt.client.Client Konstruktor aufrufen
        client = mqtt_client.Client(client_id=clientId, clean_session=False)
        client.username_pw_set(username=username, password=password)
        client.on_connect = self.onConnect
        client.on_disconnect = self.onDisconnect
        client.on_message = onMessage
        
        try:
            _LOGGER.info(f"Connecting to MQTT broker: {self.mqttUrl}:1883")
            client.connect(self.mqttUrl, 1883)
        except Exception as e:
             _LOGGER.error(f"Failed to connect to MQTT broker {self.mqttUrl}: {e}")
             # Fehlerbehandlung hier ist wichtig, falls die Verbindung fehlschlägt
             raise # Leite den Fehler weiter, falls Verbindung nicht möglich ist

        # Fehler behoben: Nur loop_start() verwenden, um den Loop im Hintergrund zu starten
        client.loop_start()
        
        # Hinweis: suppress_exceptions kann Fehler verbergen.
        # Für Debugging könnte man das temporär entfernen oder Fehler im Callback loggen.
        client.suppress_exceptions = True 

        _LOGGER.info("MQTT client created and loop started.")
        return client

    def onConnect(self, _client: Any, _userdata: Any, _flags: Any, _rc: Any) -> None:
        # Hier könntest du die Subscriptions neu setzen, falls die Verbindung verloren ging und wieder aufgebaut wurde
        if _rc == 0:
            _LOGGER.info("Client has been connected to MQTT broker")
            # TODO: Subscriptions hier wiederholen nach erfolgreicher Wiederverbindung, falls nötig
        else:
            _LOGGER.error(f"Failed to connect to MQTT broker. Return code: {_rc}")

    def onDisconnect(self, _client: Any, _userdata: Any, _rc: Any) -> None:
        if _rc != 0:
            _LOGGER.warning(f"Client disconnected from MQTT broker with return code {_rc}")
        else:
             _LOGGER.info("Client disconnected cleanly from MQTT broker")
        
        # reconnect() wird automatisch von loop_start() versucht, wenn clean_session=False
        # Ein explizites reconnect() hier kann in manchen Szenarien zu Problemen führen,
        # da loop_start() bereits einen Reconnect-Mechanismus hat.
        # Das explizite reconnect() und loop_start() am Ende von onDisconnect
        # in der Originalversion war wahrscheinlich auch Teil des Fehlers.
        # Es ist oft besser, sich auf den Reconnect-Mechanismus von loop_start() zu verlassen.
        # _client.reconnect() # Vermutlich nicht nötig, wenn loop_start() läuft
        # _client.loop_start() # Nicht aufrufen, wenn loop_start() bereits läuft/lief

FireSon, 2025-04-26:
Ich habe deine Nachricht gelesen, es könnte sinnvoll sein, nicht gleich alles in Großbuchstaben zu schreiben. 

1) Nirgendwo in der Dokumentation steht, dass es nicht zulässig ist, sowohl loop_start als auch loop zu verwenden. Aber wenn nur loop_start benötigt wird, entferne ich loop gerne.

2) Es ist in keiner Weise rekursiv, da der MQTT-Client einmal erstellt wird. 

3) Die api.py ist nicht für die MQTT-Kommunikation verantwortlich, sie erstellt nur den Client.

4) Es hat auch nichts mit dem Kommunikationsproblem mit den Zendure-Geräten zu tun, da diese auch beim Neustart der Integration nicht auf die Kommunikation reagieren. Darüber hinaus habe ich festgestellt, dass, wenn die MQTT-Verbindung im Hyper2000 über Bluetooth zurückgesetzt wird, alles wieder „normal“ funktioniert

FireSon, 2025-04-28:
Auf basis von deine feedback have ich die initalisierung von MQTT uberarbeitet. Ich habe es aus die api.py entfernt und im ZendureManager.py hinzugefugt. 
Wenn die integration von HA entfernt wird, benutze ich jetzt die loop_stop und disconnect. Kannst du das nochmals kontrolieren ob jetzt alles stimmt? Beim connect werden jetzt auch die subscriptions hinzugefugt.

Linus86529, 2025-04-28:
Ich war mir nicht sicher, ob du lieber auf Englisch kommunizieren möchtest. Entschuldige bitte den Sprachmix.

Linus86529, 2025-04-28:
Summary of Changes in zenduremanager.py:

    Major Improvement: The MQTT client creation and loop management (mqtt_client.Client(...), connect, loop_start()) have moved from api.py to the initialize method of ZendureManager. This fixes the critical bugs identified in api.py related to MQTT setup. The manager now directly owns and manages the MQTT client, which is a better architectural fit.
    Improved Subscription Handling: MQTT subscriptions are now initiated in the mqttConnect callback, ensuring they happen only when the client is connected. This is a robust pattern. Requires a new initMqtt method in ZendureDevice.
    Improved Cleanup: Explicit clearing of class-level device lists and calling _mqtt.loop_stop() have been added to unload.

Summary of Changes in zenduredevice.py:

    Major Improvement: The async def initialize method is removed and replaced with a simpler def initMqtt method. This new method handles storing the MQTT client and subscribing to device-specific topics.
    Better MQTT Lifecycle: The MQTT initialization (client creation, connection, loop start) is now centralized in the manager. The device is responsible for subscribing when told by the manager (via initMqtt called from mqttConnect), which correctly links subscriptions to the client's connected state.
    Added BLE Placeholder: A bleClient attribute was added, indicating future or incomplete BLE functionality.

Summary of Changes in api.py:

    Major Improvement: The critical bugs related to MQTT client creation and management are removed because the entire MQTT client handling logic was moved out of this file. api.py is now exclusively for the HTTP communication part. This is a significant simplification and improvement in the role of this class.

Summary of Changes in config_flow.py:

    Major Change: A new multi-step flow is introduced in async_step_user. Based on the CONF_USEBROKER checkbox, the user is either sent to a new async_step_mqtt (for local broker setup) or the cloud setup proceeds directly.
    Local Broker Options Activated: The local broker configuration fields (CONF_BROKER, etc.) are now active and collected in the new async_step_mqtt.
    Inconsistent Validation (Problem): The validate_input function, which performs a cloud API login check, is called in both the cloud path (when CONF_USEBROKER is false/not set) and the local broker path (async_step_mqtt). This means setup for the local broker option still requires successful cloud API login validation, which is incorrect logic.
    Inconsistent Reconfiguration: The local broker fields are not added to the async_step_reconfigure or the ZendureOptionsFlowHandler, meaning these settings can only be entered during the initial setup's async_step_mqtt.