Zendure/Zendure-HA #166: V1.0.40 - viele Batterie-Werte wurde über mehrer Stunden nicht mehr aktualisiert | Auf "Integrationseinträge", im "3-Punkte-Menü" ein "Neu laden" ausgelöst >>> viele Entities = unknown

Linus86529, 2025-04-27:
Gestern und auch heute wurden verschiedene Sensorwerte nicht mehr aktualisiert.

Irgendwann nach längerer Laufzeit werden verschiedene Batteriesensorwerte des Hub1200 und Hyper2000 nicht mehr aktualisiert und bleiben auf dem letzten Wert stehen.

Dies ändert sich auch nicht, wenn man die Betriebsmodi zwischen "AUS", "Manuell Leistungsregelung" und "Smarte Leistungsregelung" umschaltet und dazwischen genügend lange wartet.

Auf der HA-Seite "Integrationseinträge" im "3-Punkte-Menü", wo auch "Neu konfigurieren" steht, habe ich "Neu laden" ausgelöst. Danach wurden die Sonsor-Daten für alle Batterien wieder aktualisiert. Die Entitäten der Geräte Hub1200 und Hyper2000 bleiben jedoch auf "unbekannt" stehen. Auch nach einer Wartezeit von mehr als 10 Minuten bleiben die Werte auf "unknown".

(In der Version v1.0.36 hat der Menüpunkt "Neu laden" keinen Fehler ausgelöst.)

Das Auslösen von "Neu laden" erzeugte ein Fehlerprotokoll mit 4000 Zeilen. Hier ein kleiner Auszug. Vielleicht gibt das schon einen Hinweis auf die Ursache.

Snip ----
2025-04-27 15:04:20.337 ERROR (MainThread) [homeassistant.components.select] Platform zendure_ha does not generate unique IDs. ID Hub 1200-cluster is already used by select.hub_1200_cluster - ignoring select.Hub 1200-cluster
2025-04-27 15:04:20.337 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-aggrChargeDaykWh is already used by sensor.hub_1200_aggr_charge_dayk_wh - ignoring sensor.Hub 1200-aggr_charge_dayk_wh
2025-04-27 15:04:20.338 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-aggrDischargeDaykWh is already used by sensor.hub_1200_aggr_discharge_dayk_wh - ignoring sensor.Hub 1200-aggr_discharge_dayk_wh
2025-04-27 15:04:20.343 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-masterSwitch is already used by binary_sensor.hub_1200_master_switch - ignoring binary_sensor.Hub 1200-master_switch
2025-04-27 15:04:20.344 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-buzzerSwitch is already used by binary_sensor.hub_1200_buzzer_switch - ignoring binary_sensor.Hub 1200-buzzer_switch
2025-04-27 15:04:20.344 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-wifiState is already used by binary_sensor.hub_1200_wifi_state - ignoring binary_sensor.Hub 1200-wifi_state
2025-04-27 15:04:20.344 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-heatState is already used by binary_sensor.hub_1200_heat_state - ignoring binary_sensor.Hub 1200-heat_state
2025-04-27 15:04:20.344 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-reverseState is already used by binary_sensor.hub_1200_reverse_state - ignoring binary_sensor.Hub 1200-reverse_state
2025-04-27 15:04:20.345 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-pass is already used by binary_sensor.hub_1200_pass - ignoring binary_sensor.Hub 1200-pass
2025-04-27 15:04:20.345 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-autoRecover is already used by binary_sensor.hub_1200_auto_recover - ignoring binary_sensor.Hub 1200-auto_recover
2025-04-27 15:04:20.346 ERROR (MainThread) [homeassistant.components.number] Platform zendure_ha does not generate unique IDs. ID Hub 1200-inputLimit is already used by number.hub_1200_input_limit - ignoring number.Hub 1200-input_limit
2025-04-27 15:04:20.346 ERROR (MainThread) [homeassistant.components.number] Platform zendure_ha does not generate unique IDs. ID Hub 1200-outputLimit is already used by number.hub_1200_output_limit - ignoring number.Hub 1200-output_limit
2025-04-27 15:04:20.346 ERROR (MainThread) [homeassistant.components.number] Platform zendure_ha does not generate unique IDs. ID Hub 1200-socSet is already used by number.hub_1200_soc_set - ignoring number.Hub 1200-soc_set
2025-04-27 15:04:20.346 ERROR (MainThread) [homeassistant.components.number] Platform zendure_ha does not generate unique IDs. ID Hub 1200-minSoc is already used by number.hub_1200_min_soc - ignoring number.Hub 1200-min_soc
2025-04-27 15:04:20.347 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-hubState is already used by sensor.hub_1200_hub_state - ignoring sensor.Hub 1200-hub_state
2025-04-27 15:04:20.348 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-solarInputPower is already used by sensor.hub_1200_solar_input_power - ignoring sensor.Hub 1200-solar_input_power
2025-04-27 15:04:20.348 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-BatVolt is already used by sensor.hub_1200_bat_volt - ignoring sensor.Hub 1200-bat_volt
2025-04-27 15:04:20.351 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-packInputPower is already used by sensor.hub_1200_pack_input_power - ignoring sensor.Hub 1200-pack_input_power
2025-04-27 15:04:20.351 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-outputPackPower is already used by sensor.hub_1200_output_pack_power - ignoring sensor.Hub 1200-output_pack_power
2025-04-27 15:04:20.352 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-outputHomePower is already used by sensor.hub_1200_output_home_power - ignoring sensor.Hub 1200-output_home_power
2025-04-27 15:04:20.352 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-remainOutTime is already used by sensor.hub_1200_remain_out_time - ignoring sensor.Hub 1200-remain_out_time
2025-04-27 15:04:20.352 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-remainInputTime is already used by sensor.hub_1200_remain_input_time - ignoring sensor.Hub 1200-remain_input_time
2025-04-27 15:04:20.352 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-soH is already used by sensor.hub_1200_so_h - ignoring sensor.Hub 1200-so_h
2025-04-27 15:04:20.352 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-packNum is already used by sensor.hub_1200_pack_num - ignoring sensor.Hub 1200-pack_num
2025-04-27 15:04:20.353 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-electricLevel is already used by sensor.hub_1200_electric_level - ignoring sensor.Hub 1200-electric_level
2025-04-27 15:04:20.353 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-energyPower is already used by sensor.hub_1200_energy_power - ignoring sensor.Hub 1200-energy_power
2025-04-27 15:04:20.353 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-inverseMaxPower is already used by sensor.hub_1200_inverse_max_power - ignoring sensor.Hub 1200-inverse_max_power
2025-04-27 15:04:20.353 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-solarPower1 is already used by sensor.hub_1200_solar_power1 - ignoring sensor.Hub 1200-solar_power1
2025-04-27 15:04:20.353 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-solarPower2 is already used by sensor.hub_1200_solar_power2 - ignoring sensor.Hub 1200-solar_power2
2025-04-27 15:04:20.353 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hub 1200-strength is already used by sensor.hub_1200_strength - ignoring sensor.Hub 1200-strength
[... cut ...]
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    unit_of_measurement = self.unit_of_measurement
                          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 501, in unit_of_measurement
    := self.platform.default_language_platform_translations.get(translation_key)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'default_language_platform_translations'

... Und so weiter ...

snip ----

Linus86529, 2025-04-27:
Durch einen Neustart von Homeassistant haben alle Entitäten wieder gültige Werte.

FireSon, 2025-04-28:
Thanks for the info. There was a problem with the 'unload' of the integration, which resulted in basically duplicating the devices. Also instead of closing the mqtt connection, it reconnected it. This resulted in very! stange behaviour with smart matching and probably also the mqtt connection. Also many of these errors were in the logging:

`2025-04-28 09:40:48.140 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs.`

I was not able to reproduce all of your logging, but fixed the unload, at least on my dev machine. Once the next release is out can you check if it is working properly?

Linus86529, 2025-05-06:
> @Linus86529 Is this working now? Or can we close this issue?

Hi, close it please.--Gesendet mit der GMX Mail App

closed 2025-05-06 by FireSon