Zendure/Zendure-HA #170: no control through Zendure Manager?

Bimmerfriend, 2025-04-29:
Hi, 

First of All, Thank you for your great work! I appreciate all the time you have invested to get the Integration working and to support it!

In lack of any other place (is there a User forum, where do we all exchange our ideas and questions?) I am asking here. Please let me know if there is a better place somewhere...

I am using three Zendure Hubs with my solar panels. They are all connected to different phases. Monitoring is done with a Shelly 3EM Pro.

I am trying to do a simple setup: Charging when there is power on the panels, bypassing when full, discharging when the sum of the power of all three phases gets positive.

While this does not work well with the Zendure App, I understood that the Manager would do the trick. Put in the Shelly as Smart Meter, and have the three Zendures contributing as a Cluster (not the Zendure cluster, as that does not work across phases for some reason - therefore the local cluster of the integration). 

1. Is the idea wring or bad? If so, why, what does not work the way I described it?
2. If not a bad idea, how do I set it up in the integration?

Thanks for your thoughts and have a nice day!

Apollo2022, 2025-04-29:
Hello,

I would join you here. I have a similar setup with 3 x Hyper2000 on each phase and Shelly Pro3EM (each with 4 AB2000). 
I currently only use the integration to display the entities in the dashboard and calculate the battery values in the energy dashboard. At this point, many THANKS and respect for your work Fireson! 
I have a Hyper2000 in CT mode. This is based on Pro3EM and dynamically regulates power consumption. I switch on the other two Hyper2000s as follows: 

from 19:45 with 100W each 
from 23:00 I reduce them to 85W, 
from 01:30 to 65W 
and from 07:00 to 0W.

I only use AC charging because I have an existing photovoltaic system. 
I imagine that all 3 Hyper2000s charge and discharge dynamically at the same time if there is power left over or power is needed in the house and that this is distributed evenly across all 3 systems. Actually a cluster for 3 devices (distributed over three phases) without power restrictions. So all three devices charge and discharge with max. 1200W. And if, for example, a total of 600W is needed, all devices supply 200W each to the house network or, conversely, if 900W is left over, all devices charge 300W each. Is this even possible? As my previous commenter wrote, this entry is not an issue. I am totally satisfied with what you have achieved so far.

Many THANKS for your feedback in advance.

FireSon, 2025-04-29:
@Apollo2022 I have the same setup as you have. The whole idea why I started this was to use my p1 meter, same as you shelly, and manage all hypers in the same way by distributing the power equally (or actually by relative capacity). 

In order to distribute the allowed power as efficient as possible, I added the 'device cluster' per device. It is legaly possible to add device(s) to the same phase if they have their own circuit. The device cluster gives you the possibility to add them as legally allowed. The Zendure Manager will make sure that the allowed power per cluster is never exceeded. It is good to say that these device clusters are not Zenlink clusters.

In smart matching mode, the devices are charged and discharged in this way.

@Bimmerfriend you are right that the Zendure Manager should be able to do the trick. Since I do not own a Hub 1200 or 2000, I have not tested the commands needed for the manager to distribute the power. I did get however get the description from Zendure, so it can be done. The other problem is that I have not tested it in a scenario with solarpanels. We did some inprovements, but I have not tested that myself.

Bimmerfriend, 2025-04-30:
Hi Fireson, thank you for your response! I understand the restriction of not being able to test any programming. 
But what does that mean in consequence - you won't be able to integrate it at all or we need to find a way together to support you in your efforts?

If the second choice is the answer, is there anything that I can do or that you would need from somebody to move forward?

FireSon, 2025-05-03:
Well there are a few people looking into this. Appearantly there are quite a few Hubs out there. So hang tight, we will get there 😄

Bimmerfriend, 2025-05-03:
That sounds good - let me know if I can support you with anything for the hub integration!

FireSon, 2025-05-12:
This should work in version 41-pre8 can you check it there?

closed 2025-05-12 by FireSon

Bimmerfriend, 2025-05-17:
Mot rallye sure I und derstand - what should work and how should it work? 
I updated to the latest pre-release, and the value read out seem to be correct. What I do not have / see is any choice of the mode for each hub. It only says (as a sensor) the mode - mostly beeing smart mode. 
What I would expect from my outside perspective is a mode selector to got to smart mode or fix output mode to a certain output level of let's say 100 W. 
Or do I think too complicated?