Zendure/Zendure-HA #174: Lost Control on Hyper 200

FrColin, 2025-04-30:
Hello,
After a long time of trouble-free operation, the system no longer responds.
the zendure manager continue to send msg but all the sensor gets stuck.
zenapp is also in the same state.
the log show that's "Receive: xxx => ready" are not received
[custom_components.zendure_ha.zenduredevice] Receive: xxx => ready! are lost durring that time form 13:41 to 16:43
until I do a disconnect reconnect of the hyper200 switch wifi on off 
during this blackout the device is pingable and some traffic is present on port 1883 ( mqtt ) but no real data 

may be the invoke/function  is too fast ?
in my case the system was at maximun output 1200W during this trouble , zendure manager try to demand outPower 1200 from 13:21 to 13:36 near every 10s without change.

a good functionality will be to detect this state ( no ready reply ) and perhaps wait the reply of the invoke/function before sending new one

thanks a lot

harrymayr, 2025-04-30:
Zendure Server down?

hblockx, 2025-04-30:
not just that, the issues began with .40

FrColin, 2025-04-30:
> Zendure Server (partly) down?

No because disconnect reconnect wifi , make it work again

harrymayr, 2025-04-30:
Then perhaps is it the same reason that nograx (developer of the iobroker zendure adapter) has tried to solve with restarting his adapter. This may not be a problem with FireSons integration ([see also](https://forum.iobroker.net/topic/69579/test-adapter-zendure-solarflow/1114))

FrColin, 2025-04-30:
I have try to restart the integration but this doesn't solve the trouble the sensor became unknown instead of keeping the last value.
Since the zendure Android app is also affected , the only working solution is to force the hyper2000 to reconnect it self to the broker.

FireSon, 2025-05-01:
Well congratulations 😉 I think you just had your first (really) frozen hyper. This happens for me every 2-3 days depending on the activity. It is really annoiing and I have asked Zendure to look into it. As far as I can see, the hyper just stops listing to mqtt commands. The Zendure app basically uses also mqtt to communicate with the hyper. The reason you (probably) see something in the app is due to caching. Also all the commands of the app are sent, but never processed. The only way I have found so far to deal with this problem is to use bluetooth to connect to the device and reset the mqtt server. After that they are working properly.

The downside of coarse is that not everybody has the ability to connect via bluetooth. Currently I am testing a new version where this bluetooth communication is used in the integration to reset the device...

FrColin, 2025-05-01:
@FireSon Ok understand but before zendure detect and correct the trouble.
I think it could be usefull to catch the fact we didn't receive reply message for a long periode and trigger an automation to workaround this trouble.
A kind of binary sensor ( device reponsive) resetting by a timeout.

FireSon, 2025-05-01:
Yes you are right about that, will try to add  that to the next pre-release

FrColin, 2025-05-01:
👍 thanks a lot

FrColin, 2025-05-02:
Hi, 
after installation of v1.0.41-pre1 , I noticed that the ready message appear twice quicly between P1 update
here is log extract 
`2025-05-02 09:28:39.087 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 09:28:51.936 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 09:28:52.054 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 09:29:02.193 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 09:29:02.304 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 09:29:13.921 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 09:29:14.035 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:02.855 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:02.923 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:20.605 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:20.721 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:35.571 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:35.684 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:49.525 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 11:59:49.645 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 12:00:09.558 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 12:00:09.666 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 12:00:23.552 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 12:00:23.670 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 12:01:05.805 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!
2025-05-02 12:01:05.920 INFO ... [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!`

once more offline today at 9:29 reset wifi to force reconnect at 11:59 no ready message in this slot
hope on day we got a real local API

FrColin, 2025-05-02:
@FireSon  from the mqtt log i Noticed 7 invoke  request before receiving answer from zendure about 2s lags .
maybe we need to wait for the reply before sending new request i noticed also that the outPower is not different

here is the log
 `2025-05-02 17:27:07.951 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21827, "timestamp": 1746199627}`
`2025-05-02 17:27:07.954 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21828, "timestamp": 1746199627}`
`2025-05-02 17:27:07.955 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21829, "timestamp": 1746199627}`
`2025-05-02 17:27:07.957 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21830, "timestamp": 1746199627}`
`2025-05-02 17:27:07.958 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21831, "timestamp": 1746199627}`
`2025-05-02 17:27:07.960 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21832, "timestamp": 1746199627}`
`2025-05-02 17:27:07.961 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21833, "timestamp": 1746199627}`
`2025-05-02 17:27:07.962 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21834, "timestamp": 1746199627}`
`2025-05-02 17:27:07.965 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 545}, "msgType": 1, "autoModel": 8}], "deviceKey": "xxxxxxxx", "function": "deviceAutomation", "messageId": 21835, "timestamp": 1746199627}`
`2025-05-02 17:27:09.483 INFO (paho-mqtt-client-56519:36242c81-3ae6-4f48-acc7-2df6b4791cb3) [custom_components.zendure_ha.zenduredevice] Topic: Hyper 2000 /gDa3tb/xxxxxxxx/function/invoke/reply => {'messageId': 21828, 'deviceId': 'xxxxxxxx', 'timestamp': 1746199628, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}`
`2025-05-02 17:27:09.604 INFO (paho-mqtt-client-56519:36242c81-3ae6-4f48-acc7-2df6b4791cb3) [custom_components.zendure_ha.zenduredevice] Topic: Hyper 2000 /gDa3tb/xxxxxxxx/function/invoke/reply => {'messageId': 21827, 'deviceId': 'xxxxxxxx', 'timestamp': 1746199627, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}`

harrymayr, 2025-05-02:
Please show a little bit more, so that these lines are visible:

Milvus11, 2025-05-09:
Hey there! Regarding the frozen Hyper, i faced it too several times now, is there a data point in this Integration, which one could use in an automation to e.g. notify that the Hyper is frozen?

Thanks in advance!

FrColin, 2025-05-09:
Hi,
I already suggesdted @FireSon to create a sensor for this trouble
for now I survey the last_updated time on sensor to fire a notification.
i have created a group sensor of all the power sensor off the hyper

```
state_class: measurement
entity_id:
  - sensor.hyper2000_solar_input_power
  - sensor.hyper2000_output_home_power
  - sensor.hyper2000_output_pack_power
  - sensor.hyper2000_pack_input_power
  - sensor.hyper2000_grid_input_power
last_entity_id: sensor.hyper2000_solar_input_power
unit_of_measurement: W
device_class: power
friendly_name: Hyper_ survey
```

then I created a last_update template sensor base on survey

```
{% set device_entities = state_attr('sensor.hyper_survey', 'entity_id') %}
{% set last_updated_times = [] %}
{% set data = namespace(last_updated_times=[]) %}
{% for entity in device_entities %}
   {% if states[entity] and states[
   entity].last_updated %}
      {% set data.last_updated_times = data.last_updated_times + [as_timestamp(states[entity].last_updated)] %}
    {% endif %}
{% endfor %}
{% set max_last_updated = data.last_updated_times | max %}
{{ ((as_timestamp(now()) - max_last_updated) / 60) | int }}
```

and last an automation to trigger the notification

```
alias: alerte hyper 2000
description: ""
triggers:
  - entity_id:
      - sensor.hyper_last_update
    above: 15
    trigger: numeric_state
  - trigger: state
    entity_id:
      - sensor.hyper_survey
    attribute: last_entity_id
    enabled: false
conditions:
  - condition: time
    after: "07:00:00"
    before: "22:00:00"
actions:
  - action: notify.persistent_notification
    metadata: {}
    data:
      message: "Hyper 2000 hors ligne "
      title: Alerte hyper 2000
```

closed 2025-05-12 by FireSon