Zendure/Zendure-HA #180: v1.0.41-pre1 | No "Hyper Energy Programm" active on Hub 1200 and Hyper 2000 since installation of pre1

Linus86529, 2025-05-01:
I'm trying the pre1, unfortunately both devices keep the "Hyper Energy Program" = kein

On both devices none of the possible "Eneryplan" is selected in the Zendure App.

The 1st screen shot: 14:58(L)
[image]

The 2nd screen shot: 15:30(L)
[image]

Linus86529, 2025-05-01:
I completely removed the v1.0.40, restarted HA, installed the pre1, restared HA, added the integration in "Geräte & Dienste", entered the required data in the configuration dialog. Then I switch on the "Smarte Leistungsregelung".

Linus86529, 2025-05-01:
What about the mqtt logs, were are they created? I could not find them neither in /var/log nor in the Studio Code Server log.

Linus86529, 2025-05-01:
Incidentally, I have the problem that the Hub1200 does nothing when it is switched to "Smart Power Control" via zenduremanager.py, even in version v1.0.40.

In version v1.0.40, however, the Hyper 2000 reacts as soon as the P1 sensor, in my case the "Shelly 3EM Pro", measures a very high power demand from the house.
Then the Hyper 2000 delivers its share of 400W within 1..2 seconds. But the Hub 1200 remains idle at 0W. Even though the P1 sensor measures 1700W, for example.

Previously, with all versions of the Z-HA integration, I had to set the Hub 1200 to "Smart CT mode" in the energy plan in the Zendure app in order for the Hub 1200 to respond to the
Shelly 3EM Pro" at all. In this mode, the 3EM delivers the power values to the Hub 1200 via the cloud.

Either the power values from the P1 sensor of the Z-HA Intergation are not reaching the Hub 1200 or something else is wrong.

I was wondering why the Hub 1200 seems to make an exception in the product name. Is the "solarflow2.0" in the zenduremanager.py correct?

For the others, the device name and product name are identical:
  case "solarflow2.0":
      device = Hub1200(self._hass, deviceKey, deviceDef)

FireSon, 2025-05-01:
The mqtt logging is now configurable. It is sometimes too verbose. When you switch on the debug logging for the integration, there is very helpfull logging under url/config/logs when you select show raw logs. Especially th power distribution in your case is of interrest, since nothing seems to work.

Linus86529, 2025-05-01:
Some minutes ago I reentered the configuration dialog and activated the check box for mqtt logging.
Usually I'm jump to HA via an ssh session from my Linux machine. In the folder config there is no folder called logs and no file with this name. 

Do you think I have to redo the complete Z-HA installation from scratch to get the logging to work?

FireSon, 2025-05-01:
Ahh sorry, what I meant is via the UI of HA, navigate to settings/System/Log this will get you in the HA logging

[image]

Before you see anything you must enable the debug logging here:

[image]

mangri01, 2025-05-01:
Ich nutze 4 x SolarFlow 1200 mit je eine Akku von Zendure und zwei Hoymiles HMS 1800 4T und bei mir Funktioniert keine Version auch die neue 1.0.41 hat keine Funktion auf die SolarFlow.
Für die Strommessung nutze ich einen Shelly 3em pro.

Habe extra einen sensor.power_actual angelegt. Der zeigt Positiven und Negativen Stromverbrauch in Watt an.

Aber die SolarFlow regeln nicht bei Strombedarf. 
Alle Sensoren von Zendure werden korrekt in Home Assistant angezeigt und Aktualisiert.

Was kann man da noch ausprobieren??

FireSon, 2025-05-01:
Was noch fehlt sind die richte MQTT befehle fur die Hub1200, die unterscheiden sich ein wenig von die fur die Hyper2000 benutzt werden.  Haben deine gerate die ProductKey: A8yh63? Die kann mann in die logging finden. Wenn das stimmt kann ich fur die Hub1200 andere befehlen verwenden. Alles ist da schon auf vorbereitet!
Kannst du die Hub12000 in smart matching mode verwenden? Das ware super, weil wir dan die befehle ueberprufen konnen. Wenn nicht versuchen wir es einfach mit die dokumentation die ich habe...

FireSon, 2025-05-01:
@mangri01 Stimmt, ich habe die log datei gefunden! Die produktKey fur ein Hub12000 ist 73bkTV. 
@Linus86529 Wenn ich es richtig verstanden habe hat die Hub1200 auch in die Zendure App verschiendene 'programmen', wie Smart Matching und CT mode stimmt das? Kannst du die auch in die Zendure App starten, wobei dann auch richtig gesteuert wird? Wenn die debug logging von HA integration activ ist werden auch die Befehle die von die App geschickt werden auch in die logging rein geschrieben. Kannst du die logging vielleicht zu mir schicken? Vielleicht stehen die notwendige Befehle darin...

Linus86529, 2025-05-01:
Also, das Log in Einstellungen --> System --> Protokolle zeigt nur zwei Einträge.
Das File "home-assistant.log" des "Studio Code Server"-Editors läuft schön voll...

Im MQTT-Explorer und im "home-assistant.log" sehe ich für die Devices folgendes:

[Hyper 2000]: 'deviceId': '2sHuahY6'
[Hub 1200]:  'deviceId': 'P8yNMq5B'

Zur Info: Ab 19:31 habe ich in der Zendure-App für den Hub1200 im "Energieplan" den "Smart-CT-Modus" aktiviert. Dadurch gibt der Hub 1200 auch Leistung ins Hausnetz ab. Ohne den "Smar-CT-Modus" machen der Hub 1200 und der Hyper 2000 nichts. 

Energy Mode:
[image]

Hier ist das Debug-Log:
[home-assistant_zendure_ha_2025-05-01T17-37-34.889Z.log](https://github.com/user-attachments/files/20002652/home-assistant_zendure_ha_2025-05-01T17-37-34.889Z.log)

Hub 1200 Page:
[image]

Hyper 2000 Page:
[image]

Wenn ich Du noch was brauchen kannst, melde Dich bitte. Ich bin heute bestimmt noch bis 23:00 am Rechner

Linus86529, 2025-05-01:
Sorry, irgendwie sind die Informationen für die beiden Devices verschwunden.
Im MQTT-Explorer und im "home-assistant.log" sehe ich für die Devices folgendes:
Hub 1200: 'deviceId': 'P8yNMq5B'
Hyper 2000: 'deviceId': '2sHuahY6'

Linus86529, 2025-05-01:
Der Produktkey stimmt. Auch bei Reinhard Brandstaedter sind die Produktkeys zu finden:
[https://github.com/reinhard-brandstaedter/solarflow-bt-manager](url)

mangri01, 2025-05-01:
Ok, dann wird es bald ein Update für die HA Zendure Integration geben,mit den neuen ProductKey damit auch die SolarFlow 1200 Hub geregelt wird.

Linus86529, 2025-05-01:
Entschuldige bitte. Deine vorherige Frage habe ich gerade noch gesehen:

@Linus86529 Wenn ich es richtig verstanden habe hat die Hub1200 auch in die Zendure App verschiendene 'programmen', wie Smart Matching und CT mode stimmt das? Kannst du die auch in die Zendure App starten, wobei dann auch richtig gesteuert wird? Wenn die debug logging von HA integration activ ist werden auch die Befehle die von die App geschickt werden auch in die logging rein geschrieben. Kannst du die logging vielleicht zu mir schicken? Vielleicht stehen die notwendige Befehle darin...

Ja, der Hub hat verschiedene Programme, siehe screen shot:
[image]

Das Log habe ich bereits weiter oben angefügt.

Linus86529, 2025-05-01:
Danke für die wahnsinns Arbeit die Ihr alle in dieses Projekt investiert. Ich würde mich freuen wenn ich weiter helfen kann.

mangri01, 2025-05-01:
> Danke für die wahnsinns Arbeit die Ihr alle in dieses Projekt investiert. Ich würde mich freuen wenn ich weiter helfen kann.

ich kann dir meine Log senden, die ich auch dem Firesun gesendet habe. Dann bitte deine Mail Adresse

Linus86529, 2025-05-01:
Wenn du möchtest, dann lasse ich das Debug-Log morgen wieder laufen, wenn die Sonne scheint und die Akkus nicht mehr voll sind und die beiden Devices etwas tun müssen. Das macht das Log bestimmt interessanter.

FireSon, 2025-05-01:
@mangri01 die datei enthalt leider keine weitere information die uns jetzt weiter helft.

Linus86529, 2025-05-01:
Es gibt in der Zendure-App bei den Energieplänen auch den "Intelligenter Abgleichmodus". Damit der richtig genutzt werden kann, muss man ein passendes "Untersrützungsgerät" haben. Da ich leider keinen "Shelly Plus Plug S", "Shelly Plug S", oder einen Zendure "Satellite Plug EU" habe, denke ich werde ich nur einen Wert für die Grundlast einstellen können.

Meinst Du das?

Hier zwei Screen shots des "Intelligenter Abgleichmodus": 

[image]

[image]

mangri01, 2025-05-01:
Ich habe 9 Shelly Plug S eingebaut. Könnte diese Messung auch machen und den Inteligenten Modus mit eine SolarFlow 1200 laufen lassen. Muss mir aber jemand sagen, was ich alles im HA einschalten soll um es zu loggen.

harrymayr, 2025-05-01:
Ich hatte mal die Erfahrung gemacht, dass die Manuellen Einträge für Input und Output-Limit den Smart-Mode behindern. Ich habe mir eine Automation gemacht, die diese Werte automatisch auf 0 setzt, wenn ich im Zendure-Manager auf "Smarte Regelung" gehe:

`alias: Smarte Regelung`
`description: ""`
`triggers:`
`  - trigger: state`
`    entity_id:`
`      - select.zendure_manager_operation`
`    to: smart`
`conditions: []`
`actions:`
`  - action: number.set_value`
`    metadata: {}`
`    data:`
`      value: "0"`
`    target:`
`      entity_id:`
`        - number.hyper_2000_output_limit`
`        - number.hyper_2000_input_limit`
`mode: single`

Linus86529, 2025-05-01:
@harrymayr: Die beiden Werte stehen bei meinem Hub 1200 beide auf 0. Danke für den Hinweis.

harrymayr, 2025-05-01:
Aber oben im Screenshot nicht beim Hyper2000.

Linus86529, 2025-05-01:
Stimmt. Die beiden habe ich jetzt auch auf 0 gezogen ;-)

FireSon, 2025-05-01:
> Ich habe 9 Shelly Plug S eingebaut. Könnte diese Messung auch machen und den Inteligenten Modus mit eine SolarFlow 1200 laufen lassen. Muss mir aber jemand sagen, was ich alles im HA einschalten soll um es zu loggen.

@mangri01 Das ware super! Du musst die debug logging einschalten:

[image]

Und die mqtt logging:

[image]

Local mqtt ist nicht notwendig. Wenn alles lauft einfach die mode uber die Zendure App einschalten. Wir suchen die function/invoke befehle 😄

harrymayr, 2025-05-01:
> Stimmt. Die beiden habe ich jetzt auch auf 0 gezogen ;-)

Dann nochmal im Zendure-Manager auf Manuell und wieder Zurück auf Smarte Leistungsregelung

harrymayr, 2025-05-01:
Und für das debugging wäre es gut auch den Anfang mitzubekommen. Also Debugging erlauben und dann HA neu starten. Nach 10 Minuten oder so, das debugging beenden und das Log einstellen.
Die deviceID und die email die dann in den Logs steht, muss man nicht unbedingt öffentlich machen und sollten unkenntlich gemacht werden.

mangri01, 2025-05-01:
Habe es so emacht.
SolarFlow einen Shelly Plug Hizugefügt, Kühlschrank Side by Side lüft immer.
Zendure App Inteligenten Modus mit dem Shelly Plug Aktiviert.
In der HA Zendure die MQTT log eingeschaltet, und das Debug Protokoll Aktiviert
Danach im HA Zendure den Betriebsmodus Smarte Leistungsregelung Aktiviert, so läuft es jetzt.

Wie lange soll ich nes laufen lassen??

[image]
[image]
[image]

Linus86529, 2025-05-01:
@harrymayr: Ok, dann gehe ich auch so vor. Dann lassen sich die Ergebnisse besser auswerten.

mangri01, 2025-05-01:
Kann ich es wieder ausmachen und wie bekomme ich glich nochmal die Log datei

harrymayr, 2025-05-01:
> Kann ich es wieder ausmachen und wie bekomme ich die Log datei

Wenn ich debugging ausmache, lädt mein Browser eine Datei herunter. Am Rechner landet die dann im Download Folder (Windows)

harrymayr, 2025-05-01:
Auf dem Handy landet es bei mir (Android - Samsung) im internen Speicher unter Download.

mangri01, 2025-05-01:
Anbei meine Log Datei

[log.docx](https://github.com/user-attachments/files/20003697/log.docx)

FireSon, 2025-05-01:
@mangri01 In die datei is leider keinen invoke befehle. Die ZendureManager sollte in OFF mode sein, um die Zendure App die controlle uber die gerate zu geben. In Manual mode werden die gerate immer noch gesteuert... Kannst du das nochmal probieren?

mangri01, 2025-05-01:
> [@mangri01](https://github.com/mangri01) In die datei is leider keinen invoke befehle. Die ZendureManager sollte in OFF mode sein, um die Zendure App die controlle uber die gerate zu geben. In Manual mode werden die gerate immer noch gesteuert... Kannst du das nochmal probieren?

Alos soll ich im HA Zendure keinen Betriebsmodus auswählen. Auf aus stehen lassen
MQTTLOG wieder Aktivieren
Debug Log wieder Aktivieren

Nur in der Zendure App den Energiemodus Intelligenter Modus mit einem Shelly Plug S.

harrymayr, 2025-05-01:
@FireSon Es scheint, die Solarflow werden nicht als ZendureDevice.devices erkannt. Die totalCapacity ist 0.

`2025-05-01 21:01:49.525 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 217 stateManagerState.DISCHARGING capacity: 0 max: 3600`

Und damit wird auch totalPower nicht aktualisiert.

[image]

mangri01, 2025-05-01:
Welchen Cluster soll ich einrichten. Jeder von den vier SolarFlow 1200 hat einen Akku.

[image]

Also soll ich im HA Zendure keinen Betriebsmodus auswählen. Auf aus stehen lassen
MQTTLOG wieder Aktivieren
Debug Log wieder Aktivieren

Nur in der Zendure App den Energiemodus Intelligenter Modus mit einem Shelly Plug S.

harrymayr, 2025-05-01:
Ich schätze, das kannst du für jeden Solar Flow einstellen.
Gesetzeskonform wäre für den ersten:
Cluster max. 800W

und für alle anderen:
Part of Solar Flow 1 AB 2000 cluster

mangri01, 2025-05-01:
> Ich würde mit der Cluster-Einstellung erst nochmal HA Zendure Smarte Leistungsregelung einstellen und Debug Log. Ich könnte mir vorstellen, das funktioniert dann schon. Vom MQTT Log habe ich nichts in dem Log gesehen. Das habe ich selbst noch nicht getestet wo das landet.
> 
> ... und in der Zendure App alles aus.

Ich kenne mich jetzt nicht mehr aus
Es war doch die rede, dass ich einen Shelly Plug S testen soll. Aber wenn ich in der Zendure APP alles aus mache habe ich den Intelligenten Modus nicht der den mit dem Shelly das System regelt.

Linus86529, 2025-05-01:
@harrymayr: Ich habe mein Debug Log mal nach "function/invoke" durchsuchen lassen und erhalte 107 Treffer. Wenn Du mal einen Blick ins Log werfen willst:

[clean____home-assistant_zendure_ha_2025-05-01T19-25-03.845Z.log](https://github.com/user-attachments/files/20004155/clean____home-assistant_zendure_ha_2025-05-01T19-25-03.845Z.log)

harrymayr, 2025-05-01:
> Ich kenne mich jetzt nicht mehr aus Es war doch die rede, dass ich einen Shelly Plug S testen soll. Aber wenn ich in der Zendure APP alles aus mache habe ich den Intelligenten Modus nicht der den mit dem Shelly das System regelt.

Die Frage ist doch, warum es mit den 4 Solar Flow nicht funktioniert hat. Wenn die Cluster-Einstellung vorher nicht gemacht wurde, kann dies der Grund dafür sein. Das würde ich jetzt zuerst testen. Wenn es damit funktioniert, wäre das Testen mit dem Intelligenten Modus und dem Shelly nicht mehr nötig.

mangri01, 2025-05-01:
Ich habe alles nochmal eingestellt und den Betriebsmodus in HA Zendure Smarte Leistungsregelung Aktiviert. Es sieht so aus als würden alle vier SolarFlow 1200 geregelt werden.

[image]

[image]

Log2

[log2.docx](https://github.com/user-attachments/files/20004254/log2.docx)

harrymayr, 2025-05-01:
Sieht doch gut aus :-)
Es wird immer Regelabweichungen geben, Schwankungen vom/ins Netz sehen bei mir z.B. so aus:

[image]

harrymayr, 2025-05-01:
> [@harrymayr](https://github.com/harrymayr): Ich habe mein Debug Log mal nach "function/invoke" durchsuchen lassen und erhalte 107 Treffer. Wenn Du mal einen Blick ins Log werfen willst:
> 

@Linus86529 Prinzipiell scheint es zu funktionieren:

[image]

Was mir allerdings fehlt, sind die Angaben vom p1 Sensor. Bei mir sieht das z.B. so aus:

[image]

Und vom Code her, kann das eigentlich nur Auftreten, wenn der p1 Sensor nicht aktualisiert wird, oder ungültig/unbekannt ist:

[image]

mangri01, 2025-05-01:
Bei meinem vierten SolarFlow wird in der HA Zendure bei Hyper Energy Programm nicht der Smart Leistungsregelung angezeigt

Bei den anderen schon

[image]
[image]

Linus86529, 2025-05-01:
@harrymayr:  Seltsam. lass mich das bitte nochmal machen. Ich melde mich dann wieder.

harrymayr, 2025-05-01:
> Bei meinem vierten SolarFlow wird in der HA Zendure bei Hyper Energy Programm nicht der Smart Leistungsregelung angezeigt
> 
> Bei den anderen schon
> 

Ggf. liegt es daran, dass es noch nicht aktualisiert ist. Ändert sich de

> Bei meinem vierten SolarFlow wird in der HA Zendure bei Hyper Energy Programm nicht der Smart Leistungsregelung angezeigt
> 
> Bei den anderen schon
> 

Anhand des Log sehe ich aber, dass er berücksichtigt wird. Die Batterie ist da aber schon auf 14%, da werden erst die anderen anteilig entleert.
Ggf. ist der Eintrag noch nicht aktualisiert. Arbeiten tut es aber wie es soll.

mangri01, 2025-05-01:
Jetzt regelt er nicht mehr nach unten und speißt schon seit ein paar minuten viel zu viel zum Netzbetrieber ein. 

[image]

harrymayr, 2025-05-01:
@mangri01 Da müsste ich nochmal ein Log haben, um zu sehen was passiert.

mangri01, 2025-05-01:
> [@mangri01](https://github.com/mangri01) Da müsste ich nochmal ein Log haben, um zu sehen was passiert. Mit MQTT logging aktiviert.

Habe es Deaktiviert und wieder Aktiviert. Regelt wieder, mal sehen wie lange

harrymayr, 2025-05-01:
@mangri01 Sind alle Input Limit und Output Limit Regler auf 0? Es sieht nämlich so aus, als würde beim Erreichen bestimmter Schwellen, die Regelung aus gehen.

mangri01, 2025-05-01:
Bei allen vier gleich eingestellt. Passt das so??

[image]

harrymayr, 2025-05-01:
Nach meiner Erfahrung müssen Ausgangs und Eingangslimit auf 0 für den Smart-Modus.

mangri01, 2025-05-01:
> Nach meiner Erfahrung müssen Ausgangs und Eingangslimit auf 0 für den Smart-Modus.

Woher soll er dann wissen was er einspeisen kann, wenn ich Ausgangslimit auf null stelle

harrymayr, 2025-05-01:
"Cluster max. 800W", der Rest wird automatisch gemacht.

mangri01, 2025-05-01:
Jetzt speist er wieder viel zu viel zum Netzbetreiner ein.

[image]

[log3.docx](https://github.com/user-attachments/files/20004812/log3.docx)

Linus86529, 2025-05-01:
@harrymayr: Ich habe vorsorglich das Konfiguraitons Popup nochmals aufgerufen, gecheckt und bestätigt.

[image]

Danach die gleichen Schritte gemacht wie vorher. 

Damit sich was rührt, habe ich den Toster eingeschaltet. Sofort hat der Hyper 2000 mit 400W dauerhaft eingfespeist, bis die Last wieder weg war. Der Hub1200 ist die ganze Zeit über bei 0W stehen geblieben.

96 x "Update P1" und "function/invoke" sind enthalten.
Log:

[clean___Linus86529___2._Anlauf___home-assistant_zendure_ha_2025-05-01T20-40-53.146Z.log](https://github.com/user-attachments/files/20004847/clean___Linus86529___2._Anlauf___home-assistant_zendure_ha_2025-05-01T20-40-53.146Z.log)

mangri01, 2025-05-01:
Ich klinke mich aus. Habe für heute keine Lust mehr.

Gerne Morgen wieder, Vormittag ab 9 Uhr könnte wir weitermachen wenn es bei euch geht

Linus86529, 2025-05-01:
Ich muss wieder auf den produktiven RasPi "umbauen". Ich bedanke mich für Eure Hilfe und verabschiede mich. Bis Morgen irgendwann...

harrymayr, 2025-05-01:
Alles klar, gute Nacht an euch.
Hier aber noch meine Analyse, den Grund für das Verhalten kann ich aber nicht geben.

@FireSon: Das Problem bei @Linus86529 ist, dass der HUB_1200 zwar auf das Invoke-Komando reagiert, es aber nicht über packInputPower oder outputPackPower anzeigt. In dem ganzen Log sind nur diese 10 Einträge drin.

[image]

Damit wird nur die Power ermittelt, auf die der Hyper 2000 eingestellt ist. Der bekommt dann aber immer wieder den gleichen Wert zugeteilt und die ganze "Last" trägt der HUB_1200 weil da wohl auch 4 Batterien dranhängen und damit die Kapazität größer ist. Erst wenn diese leer sind, wird der Hyper zugeschaltet werden.

Allerdings waren im vorherigen Log (clean____home-assistant_zendure_ha_2025-05-01T19-25-03.845Z.log) die fehlenden Einträge noch drin. Nochmal alles neu starten könnte da ggf. helfen.

[142 more replies]

closed 2025-05-26 by FireSon