Zendure/Zendure-HA #189: Smart CT Mode Smarter machen?

htpcent, 2025-05-03:
Hallo,

gäbe es eine Möglichkeit den Smart Modus so zu nutzen das er nur bei PV Überschuss die Batterien lädt, bei Netzbezug aber nichts ins Hausnetz einspeist? Eventuell sogar Batteriestands abhängig?

Zum Beispiel:

Eine Einstellmöglichkeit nur Smartes Laden: 
-> Es wird bei überschüssiger Energie nur in den Speicher geladen. Kein Überschuss vorhanden (nur Netzbezug) und es wird nichts vom Speicher ins Hausnetz eingespeist.

Eine Einstellmöglichkeit nur Smartes Laden Batteriestands Abhängig:  
-> Es wird zum Beispiel ein Wert von 70% eingestellt. Bis zu den 70% wird nur Überschuss abhängig geladen. Zwischen 70 und der eingestellten Batterladeobergrenze  funktioniert der normale Smart CT Modus mit Be- und Entladen?

lg
Max

Linus86529, 2025-05-03:
@htpcent:
Grüß Dich. Ist die Situation nicht etwas schmal betrachtet?

Mal Angenommen:
 - in einem Haus/Wohnung wird gerade ordentlich Strom verbraucht, ca. 2600Watt konstant, über 1 Stunde.
   Danach würde der Verbrauch in den nächsten Stunden auf die Grundlast zurück gehen, mal angenommen es wären dauerhaft 300 Watt.

- die vorhandenen PV-Panele liefern gerade konstant 1200 Wp an einen Hub 2000, oder einen anderen Zen.
- die 4 Akkus vom Typ AB2000 haben gerade einen SOC irgendwo unterhalb von 70% sagen wir mal z.B. 40%.
 
Bis die Akkus von 40% SOC bis auf 70% SOC geladen, sind dauert es bei den 4 Akkus ~ 2 Stunden.

Da 30 % der Akkukapazität geladen werden soll, sind das bei 4 Akkus AB2000: 7680 Wh x 0,3 = 2304 Wh.
Bei einer Ladeleistung von max 1200W, sind die 4 Akkus nach 1,92 Stunden auf 70% SOC geladen.
2304 Wh / 1200 W = 1,92 h.

In der Zeit zieht das Haus 2600 W für 1 Stunde aus dem Netz. Das sind 2600 Wh.
Danach wird nur noch die Grundlast von 300 W aus dem Netz bezogen. Für ~ 1 Stunde sind das 300 Wh.

In der Zeit die für das Nachladen der 30% benötigt wird, bezieht man in den knapp 2 Std. ca. 2,9 kWh. 

Ich würde in meinem Fall lieber die maximal mögliche Einspeisehöhe von 800 Watt ins Haus fließen lassen. Die restlichen 400 Watt könnten dann für eine Stunde die Akkus weiter laden. Danach ist der Bezug nur noch 300 Watt. Ab dann stünden 900 Watt zum Laden der Akkus zur verfügung.

Das würde meiner Vorstellung einer Grundlastdeckung näher kommen, als nur den Akku zu laden, bis X % erreicht sind. Gerade dann wenn die Sonne voll auf die Panels ballert, belibt reichlich Watt übrig um die Akkus zu laden.

;-)
   

 
-

htpcent, 2025-05-03:
@Linus86529 

Hy,

danke für deine Sichtweise. In diesem Fall und einer Grundlastabdeckung ist deine Rechnung so auch richtig und logisch. Aber natürlich gibt es auch andere Anwendungsbereiche.

Mit geht es eher darum, das ich ab späten Nachmittag die Priorität auf das Akkuladen legen könnte damit ich volle Akkus für die Nacht und eventuellen Stromausfällen hätte. Dabei wäre  die Funktion wichtig das  der Akku nur mehr Überschuss lädt aber nichts mehr ins Stromnetz liefert.

Angenommen es wird am Abend noch gekocht.  Es ist noch ein Überschuss von 400W für 2 Stunden vorhanden. Die Akkus sind auf 80%. Dann wäre mir lieber das ich 0,8kwh noch in den Akku laden könnte und eben nicht geschätzte 2kwh für das Kochen aus dem Akku entlade und nur noch mit 50% Akku in die Nacht starte.

Es geht nicht immer nur um reine Effizienz und Kostenersparnis. Es gibt auch die zweite Seite der Backup bzw. Notversorgung.

lg
Max

Linus86529, 2025-05-03:
@htpcent 

Je mehr wir den Software Jungs an Ideen und Wünsche liefern, um so mehr davon werden wir vielleicht mal bekommen und anwenden können.

Unser Backofen taktet nur wenn die Betriebstemperatur erreicht ist, und er sie dort halten muss. In der Einschaltphase, im Garraum herscht Raumtemperatur, legt der Backofen in der Aufheizphase mit 3200 Watt los, bis er die eingestellte Temperatur von z.B.: 290°C für Pizzen erreicht hat.
Es stimmt natürlich, dass es keine ganze Stunde dauert. ;-) 

Ich freue mich am Abend natürlich auch über volle Akkus. Am Morgen wenn die Sonne wieder die PV-Module kitzelt, habe ich i.d.R. noch ca. 55% im Akku. Das liegt im Sommer auch mit an der niedrigeren Grundlast in der Nacht. Ich betreibe die Akkus im Sommerhalbjahr zwischen 20% und 90% SOC. Im Winter ist das leider nicht so ergiebig. Aber wem sag ich das.

Weiterhin viel Spaß mit deiner Anlage.

jschroeter, 2025-05-14:
Bzgl. den von euch beschriebenen komplexeren Szenarien frage ich mich, ob es nicht sinnvoller ist zusätzlich zu dieser Integration Lösungen zu nutzen die genau dafür gemacht wurden, wie z.B. https://github.com/Akkudoktor-EOS/EOS - ich habs selbst noch nicht probiert, aber klingt vielversprechend. Ich glaube es macht wenig Sinn viel mehr Logik in Zendure-HA einzubauen.

Mit taktenden Lasten habe ich auch zu kämpfen, v.a. beim Induktionskochfeld. Was schade ist, denn da wird natürlich viel Energie verbraucht. Das macht auch die original Zendure-Smart-CT-Regelung (via Shelly) auch nicht besonders gut. Hatte die Hoffnung, dass es mit Zendure-HA etwas besser ist, aber soweit ich es bisher beobachten konnte nicht wirklich. Ich habe schon überlegt wie man dies verbessern könnte, aber einfach ist es nicht. Z.B.
- Durchschnittswert berechnen und einspeisen, sodass wenigstens ein Teil der Energie aus dem Akku kommt (und ein Teil eben im Netz landet)
- Taktung und Latenz erkennen und möglichst genau in der Takung einspeisen - wenn sich das überhaupt in so kurzen Intervallen regeln lässt.

Linus86529, 2025-05-15:
Hi @jschroeter, nicht dein Ernst? "Energy System Simulation and Optimization" dort wird simuliert und sonst nichts.

MacdiverOne, 2025-05-15:
Na ja, Andreas hat schon den Fokus auf Prediction. Wenn das mal "allgemeintauglich" und nicht nur für Nerds durchentwickelt ist, kann man mit dieser Sammlung schon einiges ableiten: wetterbasiert (past+predict / Vor-Ort-Daten:Aktuell+Verlauf (Hauswerte (Auskühlung/Aufheizung/Speichervermögen/...)) optimierte Nutzung von Ressourcen (PV/Speicher/...) + BEV laden jetzt/postponed/voll/xxx% +++ u.v.m.

Aber: Aktuell - noch viel zu viel Techspeach/Gefrickel/"da nehmen wir mal an..."!

Für meinen Geschmack muss das noch einige Zeit gären, bis es "allgemeinverträglich" wird und man damit z.B. mit einem Blick anzeigen lassen kann, dass man bis 15:30h die Wäsche in der Waschmaschine zu haben hat, weil die (bei geschlossener Tür...) danach dem Lastgang von wasauchimmer folgend gestartet wird bzw. die Umwälzpumpen für den Brauchwasserspeicher automatisiert rechtzeitig und energieeffizient angeworfen werden, weil die Kinder nach dem Sport immer vor dem Abendessen (regelmäßiger Wasserverbrauch...) am Dienstag, Donnerstag und Samstag noch in die Dusche springen.

Aktuell würde das bei Normalos (also tatsächlich uns!) wohl so ablaufen wie in der Fernsehwerbung (nach dem Zahnarztbesuch vor der Tür und auf dem Laufband) ;-)

FireSon, 2025-05-16:
Dreams power inovation, they help us to look at all options to solve problems 😄 

I think that the more context you have about your personal situation, the better you could optimize your energy costs, and the use of your battery.

- If you know the 'expected wheather':
-> you could estimate the expected amount of solar energy quite accuratly, if you have let say a history of a few months. 
-> you could estimate the necessary amount of energy for a heatpump, for heating, cooling and/or hot water
- If you have a dynamic energy price, you know when to buy or sell energy
- If you have an EV, you know if it can be used for charging
- If you have a Shelly plug for: dishwasher, washing machine, oven, etc. you know when they are started and know how much they wil consume while active.

My dream is to combine all these 'inputs' and use them to define a 'hourly strategy' for a day. Which could be re-evaluated if something unexpected happens. This strategy could help to:

- Sell solar power first, before you charge your batteries. (since there is enough) and the price is still high.
- Dump the remaining kWh from batteries in the EV, if much sun is expected
- Be more or less 'aggresive' with charging since there is little or much sun expected that day.
- Buy energy as cheap as possible to charge the batteries if no sun is expected.
- Decide how 'aggresive' we want to keep zero on the meter, since the next hour the energy price is higher.
- Use the 'baseline' power usage to improve short energy peaks due to coffeemachine, toaster etc.
- Others...

It all depends on you own situation, and how much energy storage you have. But with a generic approach I think it could work. In my 'dream' nobody in my houshold has to worry about when and how they use energy, it is automatically optimized and taken care of..

Well let me think what your idea's are, and maybe we could build our dreams as well😉

xeka0, 2025-05-20:
[image]

Whose post???

closed 2025-07-31 by FireSon