Zendure/Zendure-HA #192: Nulleinspeisung bzw Geregelte Einspeisung ans EW

swissky, 2025-05-03:
Hey, 

danke zuerst für die Tolle integration. Macht mir uns einigen sicher das Leben etwas einfacher ;) 
Aktuell nutze ich 3 Hyper, je einen auf einer Phase und frage mich, ob über "Limit Output" eine Nulleinspeisung kontrolliert werden könnte. Dann könnte ich statt dem Shelly einfach ein Tasmota Device an meinem Stromzähler (Smart-Meter) nutzen um exakte Werte zu erhalten.

Danke und Liebe Grüsse
Kevin

FireSon, 2025-05-03:
Danke! Das war die Grund das ich damit angefangen habe 😄 De details dazu stehen in die read me unter "Smart Matching Sensor Configuration". Grundsatzlich kannst du jede sensor verwenden die die richtige werte liefert!

closed 2025-05-03 by FireSon

swissky, 2025-05-03:
Danke für deine rasche Antwort. Wenn ich es aber richtig sehe, gebe ich nur EINEN P1 Sensor bei der Erstkonfiguration von deinem Addon an und nicht 3 je Phase. Mein Ziel wäre es schon, dass ich pro Phase einen Hyper nutzen kann. Oder übersehe ich etwas?

Danke

FireSon, 2025-05-03:
Wenn du die P1 sensor benutzt verteilt die Integration die Strom uber die 3 hypers, abhangig von die nutzbare kapazitat. Weil du die hypers auf sperate Phasen hast, kannst du die maximale erlaubte Leistung verwenden. 
Um das alles richtig ein zu ordnen, gibt es einen Device Cluster, wobei die Gerate niemals mehr als die erlaubt Strom liefern. Wenn mehrere gerate auf die gleiche Phase sind, wird die maximale kapazitat ueber die gerate in die geleiche Phase verteilt.

swissky, 2025-05-03:
Ist mir ganz ehrlich noch nicht ganz klar. Wird das dann alles über "Limit Output" gesteuert? Weil dann könnte ich damit selbst pro Hyper eine Automatisierung einrichten. Mein Ziel ist exakte Abgabe (+/- 20W) pro Phase und nicht einfach über eine Phase hinweg. Mit einem Sensor sind ja "nur" alle Phasen zusammengerechnet.

Danke

FireSon, 2025-05-03:
Ahh sorry, das habe ich falsch verstanden. Bei uns gibt es gezetzlich einen maximale leistung pro phase (ohne specielle Massnamen), ich dachte das du das meinte. 
Aus welcher grund mogtest du pro phase regeln?

swissky, 2025-05-03:
Gibt es bei uns auch ;) Da ich mein System aber gerne als "grosse Anlage" nutzen möchte und auch darf, jedoch nicht die ganze Hardware "wegschmeissen" möchte, wäre die Lösung pro Phase ideal.

FireSon, 2025-05-03:
Warum mogtest du dann pro Phase nul auf die Meter regeln? Bei uns ist es egal auf welche Phase man zuruck liefert, man bezahlt nur die 'delta'...

swissky, 2025-05-03:
Damit ich die Last besser verteilen kann. Auch wenn später weitere Geräte dazukommen.

FireSon, 2025-05-03:
OK, ich verstehe die grund immer noch nicht, aber das ist auch kein Problem. Die Integration kann das leider nicht. Er verteilt nur die Strom uber die verfugbare hypers, ohne rucksicht auf die Last pro Phase.

swissky, 2025-05-03:
über welchen Sensor bzw über welche Steuerung wird denn dein "Smart Matching Sensor Configuration" gesteuert? Das klappt ja, aber wie? Ich könnte ja dann genau das gleiche verwenden, einfach über eine eigene Automatisierung. Auf meine Frage, ob es "Limit Output" ist, habe ich ja noch keine Antwort bekommen

FireSon, 2025-05-03:
Er benutzt nicht die LimitOutput, aber identische mqtt Befehle als die Hyper "Smart Matching Mode". Das sind keine 'sensoren' weil alles berechnet wird auf basis van die verfugbare kapazitat der Hypers, und die input von die P1 meter. Du kanst das anschauen in die zenduremanager.py