Zendure/Zendure-HA #193: Missing entities on Hyper 2000

STUBBYs1, 2025-05-03:
Missing entities on Hyper 2000

Is it possible to add more entities for Hyper 2000, since they are currently not displayed there?

battery soh

battery current direction

battery rated capacity

Thanks

FireSon, 2025-05-04:
@STUBBYs1 All the properties which the Zendure devices report, are added to HA. The ones you are mentioning not, as far as I know. It is possible to add 'derived' entities, when I know how that should be done. 
So how would the entities you mentioned be calculated?

STUBBYs1, 2025-05-04:
Hello FireSon,
What I know about the calculation is this:

1. SOH = C curr C rat, where C curr is the maximum remaining capacity and C rat is the rated capacity of the battery.

2. battery rated capacity = capacity (mAh) = amperage (mA) multiplied by time (hours). 

Example: If you have a 2000 mAh battery that is discharged at a constant current of 500 mA and takes 4 hours to reach cut-off voltage, the capacity is 500 mA * 4 hours = 2000 mAh.

3 I am concerned with the following points:

    Monitoring the current state of charge
    Display of the remaining runtime
    Monitoring the total runtime and the number of charging cycles
    Monitoring the voltage and current
    Monitoring of temperature and ageing
     To extend the service life of the battery

4. not sure, but I think I once read that the Zendure Hub displays this ???

5. now I can't find the BatCur value either

Unfortunately I don't have any more KNOWLEDGE.

Thank you for your work

z-master42, 2025-05-04:
Curiously, the state of health of each battery is transmitted on the zendure side of the Hub 1200 and Hub 2000. However, this is not the case with the Hyper 2000. On the other hand, there is no information about the battery currents on the hubs.

closed 2025-05-25 by FireSon