Zendure/Zendure-HA #196: Hyper 2000 smart mode doesn't correctly switch to battery charging and output to home

FrColin, 2025-05-05:
Hi,
I discover a situation today battery is  not charging and hyper is in by-pass mode when soclevel is at minimum
outPower  < solarInputPower , soclevel 10 % but the surplus didn't go to charge the battery 

putting Zendure manager to stop, and back to smart correct the issue .

some log to illustrate
`2025-05-05 12:27:06.902 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 160 power: 0 operation: 2`
`2025-05-05 12:27:06.902 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Start discharging with p1: 160`
`2025-05-05 12:27:06.903 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 160 stateManagerState.DISCHARGING capacity: 4 max: 1200`
`2025-05-05 12:27:06.903 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => 160 capacity 4 program: True`
`2025-05-05 12:27:06.903 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper 2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 1, "outPower": 160}, "msgType": 1, "autoModel": 8}], "deviceKey": "59YfaLt8", "function": "deviceAutomation", "messageId": 5940, "timestamp": 1746440826}`
`2025-05-05 12:27:06.933 INFO (paho-mqtt-client-56519:47a0a9da-81c9-4636-a638-e2bb590efc5f) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper 2000/function/invoke => {'arguments': [{'autoModelProgram': 2, 'autoModelValue': {'chargingType': 0, 'chargingPower': 0, 'freq': 1, 'outPower': 160}, 'msgType': 1, 'autoModel': 8}], 'deviceKey': '59YfaLt8', 'function': 'deviceAutomation', 'messageId': 5940, 'timestamp': 1746440826}`
`2025-05-05 12:27:07.571 INFO (paho-mqtt-client-56519:47a0a9da-81c9-4636-a638-e2bb590efc5f) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/function/invoke/reply => {'messageId': 5940, 'timestamp': 1746440826, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}`
`2025-05-05 12:27:07.571 INFO (paho-mqtt-client-56519:47a0a9da-81c9-4636-a638-e2bb590efc5f) [custom_components.zendure_ha.zenduredevice] Receive: Hyper 2000 => ready!`
`2025-05-05 12:27:08.574 INFO (paho-mqtt-client-56519:47a0a9da-81c9-4636-a638-e2bb590efc5f) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 8905754, 'product': 'solarFlowPro', 'timestamp': 1746440827, 'properties': {'solarInputPower': 285, 'outputPackPower': 285, 'solarPower1': 194}}`
`2025-05-05 12:27:09.612 INFO (paho-mqtt-client-56519:47a0a9da-81c9-4636-a638-e2bb590efc5f) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 8905756, 'product': 'solarFlowPro', 'timestamp': 1746440828, 'properties': {'solarInputPower': 287, 'outputPackPower': 287, 'solarPower1': 195, 'solarPower2': 92}}`
`2025-05-05 12:27:11.662 INFO (paho-mqtt-client-56519:47a0a9da-81c9-4636-a638-e2bb590efc5f) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 8905760, 'product': 'solarFlowPro', 'timestamp': 1746440830, 'properties': {'solarInputPower': 288, 'outputPackPower': 288, 'solarPower1': 196}}`
`2025-05-05 12:27:13.717 INFO (paho-mqtt-client-56519:47a0a9da-81c9-4636-a638-e2bb590efc5f) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 8905764, 'product': 'solarFlowPro', 'timestamp': 1746440832, 'properties': {'solarInputPower': 290, 'outputPackPower': 290, 'solarPower1': 197, 'solarPower2': 93}}
`

FireSon, 2025-05-05:
@FrColin If I look at your log, it seems perfectly ok, There is a demand for 160w but since yout batteries are empty it does not do anything. Please let me know if I missed anything. The whole solarpanel direct charge thing is something to be worked on. So closing this for now.

closed 2025-05-05 by FireSon

FrColin, 2025-05-06:
@FireSon , it is not correct because the entire Solar power solarinputpower 290w is directed to the home outputpackpower 290w but it should be splited in 160w to home and 130w to charge battery.
This is what is done After a restart.