Zendure/Zendure-HA #210: Battery temperature values always have “,95” in the decimal place | The same with the Hyper 2000 device temperature.

Linus86529, 2025-05-07:
I have noticed that the decimal places for all battery temperatures never change. The decimal places are always at x,95 °C.

Who can confirm my observation?

If this is the case, please transfer all temperatures to HA without decimal places. 

So instead of 17,95°C the rounded value 18 °C

Temperature history file "*.cvs" attached hier:

[Linus86529___history-data___battery_temperature_decimal_places_always_at_x.95___.csv](https://github.com/user-attachments/files/20076520/Linus86529___history-data___battery_temperature_decimal_places_always_at_x.95___.csv)

MacdiverOne, 2025-05-07:
Thats based on the °F conversion to °C - based on the decimals of the input it will be .95 forever and should be rounded to 0 decimal places as you suggested.

algernon2607, 2025-05-07:
You can do that easily in HA .That's what I've done .
It is better to keep decimals for those who want °F

MacdiverOne, 2025-05-07:
Zendure always sends temperature values in °F ending solely on decimal value .1 for whatsoever reason, so it makes no sense to retain this or strip at HA, since it (the .1 decimal) makes no sense at all.

FireSon, 2025-05-11:
Is fixed in the last pre-release 7

closed 2025-05-11 by FireSon