Zendure/Zendure-HA #215: v1.0.41-pre5 | Because of issue #214, Z-HA installed from scratch. Hereafter Hub1200 does not work again | code difference in hub1200.py

Linus86529, 2025-05-09:
Curiously, the pre5 becomes pre4 again after installation. However, after that the Hub1200 no longer works. It no longer feeds AC power into the house.

After a code comparison of the pre5 with the pre4 I used until issue #214 shows a difference in hub1200.py. Which I thought would be included in future versions. Maybe I should wait until the pre6 is available in HACS.

Here is the code comparison fragment of Pre4 with Pre5:
[image]

harrymayr, 2025-05-09:
> Maybe I should wait until the pre6 is available in HACS.

Yes, the change was included 1 or 2 days ago, but pre5 ist 4 days old. There is not yet a release with this changes.

Linus86529, 2025-05-09:
Sorry for this stupid post. I later found the code snippet in the latest version (v1.0.41-pre 6). Next time, I'll check GitHub first. Please close #215.

closed 2025-05-09 by FireSon