Zendure/Zendure-HA #219: 1.0.41-pre6

GeneralRurra, 2025-05-10:
Wollte es gerade Testen mache ich was falsch.
I was just about to test it - am I doing something wrong?

oder muss ich noch was einstellen ?
or do I have to set something else?

[image]

[image]

FireSon, 2025-05-10:
@GeneralRurra Thanks for the detailed feedback! The local Mqtt is quite a difficult and complex thing to get working properly 😄 

But first you have to get the mqtt part working! As long as the clients keep disconnecting there will be no values in HA!
There are three things you can check: 

1) Bluetooth! Do you have a working bluetooth connection and is it able to 'see' your devices? If not it will not work! If you turn on debuglogging during startup you will see logging if the integration finds a bluetooth device which is one of your Zendure devices.

2) Go to Settings/People/Users. 
There should be users with the same name as the deviceId of all your devices, and one zendureMqtt user. These are necessary in order to connect to the mosquitto broker. If you are familiar with [Mqtt Explorer](https://mqtt-explorer.com/) you can add an additional user here and try to connect to mosquitto. You can restart the addon, or HA one more time to make sure all is working properly with the addon.

3) As long as it works without Mosquitto MQTT AddOn, there is either no bluetooth connection or the commands did not work. Then you can try to use the MqttReset switch, once you turn it 'on' if will automatically to off. Then you can look in your logging for a 'station' command sent to your device. This will change the mqtt server on your device. These commands are also sent on startup, but I had to use then also once my self to get the device look for the local mqtt

Please let me know if it helps!

GeneralRurra, 2025-05-11:
also laut log findet er das gerät 

so according to the log it finds the device

```
2025-05-11 07:14:36.745 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-05-11 07:14:36.745 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Found Zendure Bluetooth device: <BluetoothServiceInfoBleak name=ZenE1_92 address=94:C9:60:C7:7C:49 rssi=-66 manufacturer_data={17733: b'1LYMFEM270992.'} service_data={} service_uuids=['00002233a002-0000-1000-8000-00805f9b34fb'] source=E8:6B:EA:EA:CB:72 connectable=False time=191800.808533118 tx_power=None>
```

Komplette Log 
Complete log

```
2025-05-11 07:10:03.390 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ble_monitor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-05-11 07:10:03.391 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration zendure_ha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-05-11 07:10:06.683 DEBUG (MainThread) [custom_components.zendure_ha] Open API connection
2025-05-11 07:10:06.683 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
2025-05-11 07:10:06.964 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/v2 => Mqtt: mq.zen-iot.com
2025-05-11 07:10:06.964 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
2025-05-11 07:10:07.207 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding device: ck5R1F59 Hyper 2000
2025-05-11 07:10:07.207 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Data: {'id': 60532, 'deviceKey': 'ck5R1F59', 'snNumber': 'EE1LYMFEM270992', 'name': 'Hyper 2000', 'productId': 68, 'productKey': 'gDa3tb', 'onlineFlag': '00601', 'productName': 'Hyper 2000', 'wifiStatus': True, 'blueState': False, 'fourGStatus': False, 'isShareFlag': '00602', 'input': False, 'output': False, 'electricity': 13, 'hasPack': True, 'restState': 0, 'upsMode': False, 'upgradeStatusDes': '', 'productType': 17, 'deviceType': 0, 'upgradeStatus': {}, 'bindId': 0, 'bindStatus': 0, 'batteryCode': '', 'packList': [], 'inputPower': 0, 'outputPower': 0, 'slowChargePower': 0, 'temperature': 0, 'temperatureUnit': 0, 'remainOutTime': 0, 'bindType': 0, 'seriesMode': 0, 'parallelMode': 0, 'networkType': 0, 'standard': 'E', 'isSwitch': False, 'url': '', 'remainOil': 0, 'genMode': 0, 'shortCode': 'E1', 'phaseCheck': 0, 'pass': 0, 'clusterList': [], 'clusterNotice': False, 'nodeStatus': 0, 'clusterId': None, 'phase': None, 'reverseState': 0, 'meterId': 0, 'hemsId': 0}
2025-05-11 07:10:07.207 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding device: 34987a47048c ShellyPro3EM 
2025-05-11 07:10:07.207 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Data: {'id': 60534, 'deviceKey': '34987a47048c', 'snNumber': '34987a47048c', 'name': 'Zähler', 'productId': 55, 'productKey': 'Aw09fy', 'onlineFlag': '00601', 'productName': 'ShellyPro3EM ', 'wifiStatus': True, 'blueState': False, 'fourGStatus': False, 'isShareFlag': '00602', 'input': False, 'output': False, 'electricity': None, 'hasPack': False, 'restState': 0, 'upsMode': False, 'upgradeStatusDes': '', 'productType': 12, 'deviceType': 1, 'upgradeStatus': {}, 'bindId': 0, 'bindStatus': 0, 'batteryCode': '', 'packList': [], 'inputPower': 0, 'outputPower': 0, 'slowChargePower': 0, 'temperature': 0, 'temperatureUnit': 0, 'remainOutTime': 0, 'bindType': 0, 'seriesMode': 0, 'parallelMode': 0, 'networkType': 0, 'standard': '4', 'isSwitch': False, 'url': 'https://control.shelly.cloud/images/device_images/SPEM-003CEBEU.png', 'remainOil': 0, 'genMode': 0, 'shortCode': '', 'phaseCheck': 0, 'pass': 0, 'clusterList': [], 'clusterNotice': False, 'nodeStatus': 0, 'clusterId': None, 'phase': None, 'reverseState': 0, 'meterId': 0, 'hemsId': 0}
2025-05-11 07:10:07.207 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Device ShellyPro3EM  is not supported!
2025-05-11 07:10:07.214 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Found: 1 devices
2025-05-11 07:10:07.214 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding sensors Zendure Manager
2025-05-11 07:10:07.214 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 0
2025-05-11 07:10:07.214 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.power_actual to _update_smart_energyp1
2025-05-11 07:10:07.217 INFO (paho-mqtt-client-zendureMqtt) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper 2000/properties/read => {'properties': ['getAll']}
2025-05-11 07:10:07.217 INFO (paho-mqtt-client-zendureMqtt) [custom_components.zendure_ha.binary_sensor] Update binairy_sensor: Hyper 2000-MqttOnline => True
2025-05-11 07:10:07.420 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Zendure Manager initialized
2025-05-11 07:10:07.420 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-05-11 07:10:07.421 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching Zendure Manager data in 0.000 seconds (success: True)
[... cut ...]
2025-05-11 07:17:19.820 INFO (paho-mqtt-client-ck5R1F59) [custom_components.zendure_ha.zenduredevice] Zendure cloud => Hyper 2000 => b'{"messageId":123,"timestamp":1746940566,"params":{"token":"abcdefgh","result":0}}'
2025-05-11 07:17:19.937 INFO (paho-mqtt-client-ck5R1F59) [custom_components.zendure_ha.zenduredevice] Device Hyper 2000 disconnected from MQTT broker with return code 7
```
Die Benutzer werden auch erstellt 
The users are also created

[image]

Und das meiner Liste von den Bluetooth geräte 

[image]

[image]

aber leider da er nicht verbindet bleiben die daten leer 
but unfortunately as it does not connect the data remains empty

[image]

harrymayr, 2025-05-11:
Bei mir steht an dieser Stelle "True"

[image]

CWILDMOUNTAIN, 2025-05-16:
still having the same issue on the 1.0.41-pre8 version with my Hub2000+ACE .. 
repeatedly it logs

disconnected from MQTT broker with return code 7
Mqtt Client has been connected, return code: 0
Zendure cloud => Hyper 2000 => b'{"messageId":123,"timestamp":1746940566,"params":{"token":"abcdefgh","result":0}}'

Also the cloud access freezes in that state, so there is no data published anywhere and the WiFi LED of my Hub is flashing from time to time.

As [GeneralRurra](https://github.com/GeneralRurra) wrote, Bluetooth connections is there all the time.

CWILDMOUNTAIN, 2025-05-19:
I did not try if there is data when BLE, with a reset, I get back cloud connection. I assume disconnection from the cloud is not working with my setup for some reason, so I keep cloud access for now.

But something seems to be broken in my case, too.
(https://github.com/FireSon/Zendure-HA/issues/269#issuecomment-2892147865)

Zanooon97, 2025-05-20:
Hei can someone check this please, i assume that my cloud connection ist wrong after hitting the local mqtt setting. And I cant go back to cloud connection. How can I reset my solarflow 800? is ther no way for a factory reset?

harrymayr, 2025-05-20:
Did you try to remove your device in the App (under settings) and add it again using BT?

Zanooon97, 2025-05-20:
wenn ich sie über blutooth verbinde dann sehe ich alle daten richtig, sobal ich die netzwerkverbindung einrichte geht pvwatt auf null und watthausoutput bleibt auf der leistung gefrohren. 
ich würde nur gerne verstehen was die pre8 version alles verändert auf mein gerät, vllt kann ich dann besser eine lösung suchen

MarkP12345, 2025-05-20:
I had the same problem with a SF 800 Pro. 

This was the analyse of Zendure:
We have discovered that your device is stuck in an infinite reboot loop because it has been triggered by a registration response command. Did you send an MQTT retained message to the topic iot/R3mn8U/79djb8m7/register/replay? If so, please stop this operation.
According to the design logic of the Zen SDK framework:

When the device receives a message under the topic iot/R3mn8U/79djb8m7/register/replay, it writes the registration information and triggers a reboot.

If the message is set as Retained, the device will receive the retained message again upon reconnecting to MQTT after rebooting, triggering the registration process and rebooting again.

This causes the device to enter an infinite reboot loop.

Fireson is aware of this problem and will solve it in next update, he said. I think your only solution is to contact Zendure. They have fixed it for me at the end.

MarkP12345, 2025-05-20:
I have no support tickets. I have a direct contact. I think it's best to copy my text and to add it to your service ticket and to indicate that you used the fireson integration (they know of it and are cooperating with fireson to make it the official zendure integration).

Zanooon97, 2025-05-20:
@MarkP12345 okei lets wait so hehe 
thank zou have a nice day

Zanooon97, 2025-05-21:
okei, update:
gestern habe ich meine geräte entfernt, account gelöscht, meine geräte mit dem zendure cloud disconnector zu dem global mqtt server hinzugefügt, neuen account erstellt und die geräte hinzugefügt. anscheinend sehe ich jetzt wieder die aktuelle leistung in meiner zendure app doch die geräte werden nichtvon hems gesteuert, und es erscheinen auch keine daten in WattH. einer ist auf 72watt output gestellt und der eine auf 111watt. ich verstehe nicht wie die werte zustando kamen. 
wie bringt man den Hems wieder zum laufen?

[image]
[image]

closed 2025-05-25 by FireSon