Zendure/Zendure-HA #220: v1.0.41-pre6 | 1st Installation with all option failed, zero batteries detected | 2nd Installation with one option less works, but...

Linus86529, 2025-05-10:
a) 1st Installation: 

  All required fields were filled in correctly including the options: “Use local Mosquitto MQTT add-on” and “Log MQTT communication”
  - Only 3 devices were recognized: “Zendure Manager”, “Hub 1200” and “Hyper 2000”
  - None of the 5 batteries were recognized.

  - Further steps aborted, as it makes no sense without batteries

b) 2nd installation:

  Everything as in the 1st installation but this time without the option: “Use local Mosquitto MQTT AddOn”
  - The devices were recognized: “Zendure Manager”, “Hub 1200” and “Hyper 2000”
  - All 5 batteries were recognized.

  “Hub 1200”, the following ‘Entities not available’:
    - sensor.hub_1200_aggr_charge_totalk_wh
    - sensor.hub_1200_aggr_discharge_totalk_wh
    - sensor.hub_1200_aggr_solar_totalk_wh
    - sensor.hub_1200_energy_power

    “AB1000”, one entity is: ‘unknown’
      - sensor.ao4hlc9ldn00052_batcur # Calculation is apparently missing

  “Hyper 2000”, following ‘Entities not available’:
    - sensor.hyper_2000_aggr_charge_totalk_wh
    - sensor.hyper_2000_aggr_discharge_totalk_wh
    - sensor.hyper_2000_aggr_solar_totalk_wh

  Please verify if the version is actually v.0.0.0:
   - sensor.hyper_2000_masterhaer_version

[image]

GeneralRurra, 2025-05-10:
well then I'm not the only one who has already opened an issue :)

STUBBYs1, 2025-05-10:
I'm also not seeing any values ​​displayed, and all the batteries are gone. Furthermore, only the Zendure Manager is displayed; the other two displays are missing.

It would also be nice if it were possible to get an instruction manual so that people could read how things work and what they can achieve with all the settings.
Many thanks to everyone for their hard work on this project.

harrymayr, 2025-05-10:
> * Further steps aborted, as it makes no sense without batteries

How long did you wait? It take some minutes on my site, until the batteries came up.

> “Hyper 2000”, following ‘Entities not available’:
> - sensor.hyper_2000_aggr_charge_totalk_wh
> - sensor.hyper_2000_aggr_discharge_totalk_wh
> - sensor.hyper_2000_aggr_solar_totalk_wh

Yes, there are 2-3 things to change (allready post commits)

> I'm also not seeing any values ​​displayed, and all the batteries are gone. Furthermore, only the Zendure Manager is displayed; the other two displays are missing.

@STUBBYs1 And you removed your configuration first, and add a new one (see notes for pre6)

STUBBYs1, 2025-05-10:
@harrymayr Yes, I did everything as we were told.

Only the hyper was recognized, but all values are missing there.

[image]

STUBBYs1, 2025-05-10:
@harrymayr

[image]

Linus86529, 2025-05-10:
@harrymayr: Hi, I waited about 10 to 15 minutes. During that time, I documented my observations and took screenshots. All while enjoying a delicious cup of coffee.

harrymayr, 2025-05-10:
Die you use "Use local Mosquitto MQTT add-on”? Perhaps this is the problem, if you have no Blue-Tooth connection, or did not provide the WLAN credentials. @Linus86529 posted that his 2nd installation without "Use local Mosquitto MQTT add-on” worked

STUBBYs1, 2025-05-10:
@harrymayr I have entered all the required data and have all the Bluetooth connection and WLAN login information entered and running.

Linus86529, 2025-05-10:
@STUBBYs1: How long did you wait for the batteries to show up?

Linus86529, 2025-05-10:
@harrymayr: I'm going to try it again in 10 minutes.

STUBBYs1, 2025-05-10:
Linus86529 Approx. 30 minutes

FireSon, 2025-05-10:
In the #219 issue, I added some additional info on how to get it working. It might be good to check the log of the Mosquitto Addon to see if the connection with mosquitto is working. @GeneralRurra added some info on that as well.

Linus86529, 2025-05-10:
@FireSon:
Since more than one hour I'm stareing at the Mosquitto Log, waiting for changes. But... none. The last change was at 
I'm watching the data of "core-mosquitto.local.hass.io" in the MQTT Explorer. What do you expect me to see there?

Referring to 1.) BT:
Is it mandatory to use a smartphone or tablet to get it working with BT? btw I can see all the Zen devices on my laptop via BT, but the connection can't be established.

Do I need to disconnect the devices' Wi-Fi connection first to be able to connect them to BT only?

I prefer to do the HA installation and settings on my laptop. I usually only use the tablet for quick checks or to quickly use the Zendure app.

Referring to 2.) People/Users:
There are 3 new users. One zendureMQTT and one user per device (Hub1200 and Hyper2000)
But none for the bytteries.

Referring to 3.) MQTT Add-on:
In my case it works immediately without Mosquitto MQTT AddOn and all batteries are recognized and displayed within seconds.

STUBBYs1, 2025-05-10:
Hello, I've now completely deleted v1.0.41-pre6 and installed the previous version 1.0.41-pre4.
Everything is displayed there: ALL Hyper and ALL BATTERIES.

[image]

FireSon, 2025-05-10:
@Linus86529  To be clear your HA should have bluetooth! I do not know where or how you installed HA, but it must have a bluetooth adapter, and be able to 'see' your Zendure devices. Without this it will not work! Each 90 secondes HA will scan for bluetooth devices until they are all found. In the logfile you should see something like "ZenE1" for the hyper2000.

Linus86529, 2025-05-10:
@FireSon: 
I just made a curious observation. Before I performed the reinstallation of pre6 from scratch. The versatile thermostat integration asked me for updating the latest version available. I thought it's a good idea to do this in advance, before I continue with the Z-HA integration.

So, after the mandatory restart by versatile thermostat I was surprised by the Z-HA integration which showed me now 8 devices instead of only 3. Why did this happen?
 
Before I decided to reinstall the Z-HA, I used "Neu konfigurieren" to switch off the option "Lokalen Mosquitto MQTT AddOn verwenden" to get back all the devices in none local MQTT mode. But nothing happend after the deselection of the option.

Now I know this option needs a restart when changed.

STUBBYs1, 2025-05-10:
After several attempts to install the new update
v1.0.41-pre6
it finally worked.
Unfortunately, no values ​​are displayed for me (see photos).

[image]
[image]
[image]

Linus86529, 2025-05-10:
@FireSon: Pi4 and Pi5 can get only one of the two Zen devices. I'll need a ESP BT Gateway.

STUBBYs1, 2025-05-10:
@Linus86529  Buy an ESP BT gateway???
Why? All devices are now displayed on my Pi4, but no status values ​​are visible.

Raudi1, 2025-05-20:
Does it work with Shelly relays? I have a bunch of Shelly devices in my garage where the Hyper2000 is and the one I configured as a HA BT relay can see the ZenE1_34 device according to the HA BT advertisement-monitor. The BT adapter connected to the server can't see the Zendure. If I enable the local MQTT, I can't see any values.

FireSon, 2025-05-25:
Please checkout the  [wiki](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt) for more information on the bleutooth connection

closed 2025-05-25 by FireSon