Zendure/Zendure-HA #225: Error during Login

bananbabird, 2025-05-10:
Dear all,

I am failing during login, although I checked login and pwd several times

[image]

There are also errors in the system log

[image]

-----------------
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.zendure_ha.api
Quelle: custom_components/zendure_ha/api.py:85
Integration: zendure_ha (Dokumentation, Probleme)
Erstmals aufgetreten: 22:17:16 (1 Vorkommnis)
Zuletzt protokolliert: 22:17:16

Unable to connect to Zendure !

------------------------
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.zendure_ha.config_flow
Quelle: custom_components/zendure_ha/config_flow.py:76
Integration: zendure_ha (Dokumentation, Probleme)
Erstmals aufgetreten: 22:17:16 (1 Vorkommnis)
Zuletzt protokolliert: 22:17:16

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/config_flow.py", line 76, in async_step_user
    info = await validate_input(self.hass, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/config_flow.py", line 50, in validate_input
    raise ZendureConnectionError
custom_components.zendure_ha.config_flow.ZendureConnectionError: Zendure Integration

This is my setup:

[image]

Zendure HACS integration 1.0.40

Any advice what to do ?

Regards

FireSon, 2025-05-10:
Thanks for the extensive feedback! I think your setup is totally fine, the problem is that you are not able to login to the Zendure Cloud Api. Do you use exactly the same account for the Zendure App?

bananbabird, 2025-05-11:
I checked a gain, and in fact there was a problem with my password. Now it works :-). Thank you for support.
Maybe the code of the login section can be adapted so that no "Unexpeced exception" is shown in the error log during a failed login,

FireSon, 2025-05-11:
Yes that is a good suggestion! Great that it works 😄

closed 2025-05-11 by FireSon