Zendure/Zendure-HA #237: v1.0.41-pre7 issues

kutzera, 2025-05-12:
Derzeit sehe ich folgende Probleme bei der Pre Version 7:

- lokale MQTT Verbindung - keine Anzeige der Daten.
- ACE bisher nicht verfügbar
- Bypass - Modus wird falsch angezeigt. An statt aus und anders herum

MacdiverOne, 2025-05-12:
Bei mehreren Hyper2000 wird vom primären Clustersystem 100% eingespeist, von den Clusterteilnehmern jeweils wohl ihr berechneter Anteil. Insgesamt dann bei 3 Hyper2000 mehr als 150%

MacdiverOne, 2025-05-12:
Bei Änderung auf einen festen Einspeisewert passt es dann wieder, der wird ordentlich unter den Clustermembern verteilt. Beide Fälle sind reproduzierbar.

Christian23996, 2025-05-12:
When I select the local mqtt option, no incoming messages are displayed in the “MQTT Explorer”. 

But if I do not select the local mqtt option, then messages are displayed in the “MQTT Explorer”.

Wunni73, 2025-05-12:
Die Regelung geht nicht mehr , er Powert nur noch mit 800W rein.

kutzera, 2025-05-12:
Ich habe jetzt nochmal auf lokalen MQTT umgestellt. Jetzt bringt er mir im MQTT Explorer bei den Hubs folgende Meldung:

▼properties
read = {"properties": ["getAll"]}

mehr gibt's nicht.
unter den Geräten werden auch nur einzelnen statuswerte (wlan, mqttonline) angezeigt aber keine Verbrauchswerte.

harrymayr, 2025-05-12:
> > Bypass - Modus wird falsch angezeigt. An statt aus und anders herum
> 
> Bei welchem Gerat?

Da man einen ACE nicht mit dem Hyper koppeln kann, nehme ich an einen HUB xxxx. Für den Hyper kann ich einen falschen Bypass-Mode nicht bestätigen.

harrymayr, 2025-05-12:
> Bei mehreren Hyper2000 wird vom primären Clustersystem 100% eingespeist, von den Clusterteilnehmern jeweils wohl ihr berechneter Anteil. Insgesamt dann bei 3 Hyper2000 mehr als 150%

> Bei Änderung auf einen festen Einspeisewert passt es dann wieder, der wird ordentlich unter den Clustermembern verteilt. Beide Fälle sind reproduzierbar.

Wenn das mit der 1.0.41 pre8 immer noch so ist, könntest du dann dafür ein LOG bereit stellen? Bei der pre7 wurde unter bestimmten Umständen, wenn über die PV-Eingänge Leistung rein kam, die eingestellte Leistung falsch berechnet.

Christian23996, 2025-05-12:
[image]

Hi Fireson,

I have 3 Hyper2000 (I only use the AC mode) and when I disable the local MQTT option, I receive incoming messages on my local MQTT server (mosquitto broker HA-Addon) and the Z-HA integration works fine. (See my screenshot above (MQTT Explorer HA-Addon).

However, when I enable the local MQTT option, I receive nothing and the Z-HA integration does not work and all the hypers lost the wifi connection and the zendure-cloud connection. 

Then I can only connect the hyper with Bluetooth to get it back into the Wifi /zendure-cloud.

There are also 5 new HA users created with local rights.
1. hyper
2. hyper
3. hyper
zenAppMqqt
zendureMqtt

MacdiverOne, 2025-05-13:
Pre-8 läuft bisher mit "Sonne satt" in alle 3 Hyper mit stimmiger Abgabe ins Haus (etwa 150 Wh ungewollte Netzeinspeisung auf ca. 8 kWh Hausabnahme (davon etwa 500 Wh aus dem Netz, Rest PV-/Akku-mix))

Besser als je zuvor von Zendure (Cluster). Jetzt noch den Offset (hint-hint)...

Klasse Arbeit - superschnelle Reaktion! Ein dickes THX!!!

Christian23996, 2025-05-13:
@MacdiverOne / @Linus86529
Läuft die PRE8 bei Euch mit der lokalen mqtt option oder ohne? Kann aktuell leider nicht testen.
Aber die direkte lokale Steuerung der Hyper ohne die Zendure cloud wäre für mich der absolute Traum.

MacdiverOne, 2025-05-13:
Bei mir ohne, HA ist in ProxMox (und sowieso außerhalb der BT-Reichweite) mit EMQX als MQTT-Broker (auch ProxMox - läuft super).

Sobald sich da eine Lösung abzeichnet ist bei mir die Zendure-Cloud auch Geschichte.

Ich bin da guter Dinge, so fit und aufgeschlossen, wie die Truppe um FireSon ist.

Kommt Zeit, kommt WLAN-MQTT bzw. Pi4/Pi5 BT-Bridges (ich wäre da mit einem beherzten Griff in die Grabbelkiste dabei!)

FireSon, 2025-05-13:
Bei mir ist HA auch in ProxMox aber hat meinem Rechner einen usb adpter mit Bluetooth 😄  
Hat EMQX auch die Möglichkeit, HA-Benutzer zu authentifizieren? Das war eigenlich die wichtigste Grund Mosquitto zu verwenden...

MacdiverOne, 2025-05-13:
EMQX kann mit und ohne Authentifizierung arbeiten. Ich habe eine eigene Instanz als LXC laufen, auf die der MQTT-Client von HA verweist.

USB-BT: wäre möglich, aber der ProxMox Host ist zu weit entfernt, um sicher alle drei Hyper zu erreichen. 😶

kutzera, 2025-05-13:
> Bei mir ohne, HA ist in ProxMox (und sowieso außerhalb der BT-Reichweite) mit EMQX als MQTT-Broker (auch ProxMox - läuft super).
>
> Sobald sich da eine Lösung abzeichnet ist bei mir die Zendure-Cloud [...]

Wieso braucht denn die Pi4 eine Bluetooth Bridge? Das sollte auch mit der Bluetooth Schnittstelle direkt in der Pi4 gehen...

MacdiverOne, 2025-05-13:
->> Pi4/5 >>als<< BT-Bridge für die ProxMox-HA Instanz (also um BT näher an die jeweiligen Hyper zu bringen)

Christian23996, 2025-05-13:
Für mich kurz zum Verständnis, für den lokalen MQTT Betrieb brauche ich zwingend BT? 
Ich dachte BT wäre nur als Backup da, um die Hyper wieder zurück ins WLAN zu bekommen?
Wenn die Hyper dann über das lokale MQTT laufen, wie werden diese dann in HA eingebunden? Werden die dann über die MQTT Integration als Geräte hinzugefügt?

Habe gerade mal geschaut, über meinen BT-Adapter kann ich meine 3 Hyper reichen.

[image]

MacdiverOne, 2025-05-13:
Ich habe das so für mich mitgenommen, dass die MQTT Kommunikation auf BT umgebogen wird und die Zendure Systeme sämtlich per BT erreichbar sein müssen. Das war so der erste allgemein als ex-Cloud von R. Brandstädter im Netz publizierte Ansatz (der ja auch etwas für sich hat, weil WLAN nicht so die Zendure-Kernkompetenz zu sein scheint...).

Linus86529, 2025-05-13:
@Christian23996 
Mein HA läuft auf einem RasPi5 mit 8 GB RAM und als Testsystem auf einem RasPi4B mit 4 GB RAM. Auf beiden sehe ich inzwischen die Zen-Geräte über Bluetooth, habe es aber bisher nicht geschafft, dass sich die Software über Bluetooth mit den Zen-Geräten verbindet.

Darum fahre ich noch ohne lokale Verbindung. Wenn ich mal viel Lust habe, dann mache ich mich an die lokale Verbindung heran.

FireSon, 2025-05-13:
Die einzige moglichkeit die MQTT server in die Zendure Gerate zu aendern ist mit Bluetooth. Die WLAN Credentials sind notwendige parameter fur das commando.

harrymayr, 2025-05-13:
Aber die MQTT Kommandos laufen dann weiter über WLAN, nur die Änderung des MQTT Servers ist über BT?

Linus86529, 2025-05-13:
@harrymayr: Hab die Beschreibung vom Brandstaedter auch so verstanden, dass BT für die Umstellung auf den lokalen Broker benötigt wird und über BT Telemetriedaten vom Zen Gerät gelesen werden. Er schreibt weiter das für seine Lösung auch das WLAN umgestellt werden muss. 

Zitat [Brandstaedter:](https://github.com/reinhard-brandstaedter/solarflow-bt-manager?tab=readme-ov-file#offline-mode)
"Das Trennen der Verbindung erfolgt durch Neuinitialisierung der Netzwerkverbindung (WLAN) des Hubs und Anweisung an den Hub, eine Verbindung zu einem anderen MQTT-Broker herzustellen. (eventuell müssen Sie zuerst das WLAN auf ein anderes Netzwerk umstellen, bevor Sie dem Zielnetzwerk beitreten können) Hinweis: Beim AIO2400 müssen Sie die Netzwerkverbindung manuell zurücksetzen, indem Sie die Verbindungstaste ca. 3 Sekunden lang drücken, bevor Sie das AIO mit Ihrem lokalen Broker verbinden. "

FireSon, 2025-05-14:
@harrymayr 

> Aber die MQTT Kommandos laufen dann weiter über WLAN, nur die Änderung des MQTT Servers ist über BT?

That is correct.

@Linus86529 Yes he reversed engineered the protocol. It is quite easy if you have Wireshark 😉 and know what you must do. 
I know that as of the SF800, all devices also have the possibility to communicate via HTTP, which could make it much easier to manipulate. Unfortunately, they did not plan it for the Hyper or Hub, as far as I know..

Christian23996, 2025-05-14:
Hallo,

ich habe mich kürzlich an den Zendure Support gewandt, um detailliertere Informationen zur Funktionsweise von drei Hyper als Clustersystem zu erhalten. Nachdem ich angedroht hatte, die Hyper zurückzusenden, da die auf der Zendure-Website versprochene Funktion nicht annähernd umsetzbar ist, teilte man mir Folgendes mit:

> Lieber Kunde,
> 
> Es tut mir leid, dass Sie warten mussten! Wir sind für Sie da.
>  
>  Wir planen, Hyper2000 in Zukunft mit HEMS kompatibel zu machen, die Clusterfunktion vollständig zu ersetzen und die Hyperdispersion in verschiedenen Phasen für eine einheitliche Steuerung zu unterstützen, so dass Ihr Problem in Zukunft gelöst werden kann.
>  
>  Wir danken Ihnen für Ihr Verständnis und Ihre Mitarbeit. Wenn Sie weitere Fragen zum Produkt haben, zögern Sie bitte nicht, uns zu kontaktieren.
>  
> Beste Wünsche 
>  Zendure Team_

Könnte die umfangreiche Arbeit an dieser Integration dadurch gefährdet sein?

FireSon, 2025-05-16:
@Christian23996 Hoffentlich nicht 😉 Die HEMS Gerate haben auch einen webserver, die mit HTTP abgefragt werden kann. So offentlich die Hypers dann auch...

Zanooon97, 2025-05-20:
Hallo zusammen, nachdem ich alles gelesen habe denke ich ich hab mein problem verstanden, meine mqtt zugangsdaten wurden auf den geräten geändert durch das eingeben der wifi ssid und pw. 
wie bekomme ich das wieder auf original zurück, taste 3 sekunden drücken geht nicht, stromlos machen auch nicht? Bitte helft mir ich bin ohne verbindung seit drei tagen 😅

FireSon hat das so erklärt meiner meinung, dass die wifi daten die mqtt auf den geräten ändern
[image]

harrymayr, 2025-05-20:
> Bitte helft mir ich bin ohne verbindung seit drei tagen 😅

Perhaps this [Tool](https://github.com/nograx/zendure-cloud-disconnector) can help

Zanooon97, 2025-05-20:
Ich habe es versucht aber nach einem neustart gibt es keine verànderung. das hat mir schon hoffnung gegeben 

[image]

FireSon, 2025-05-25:
@Zanooon97 Version 43 shows the actual connections of the devices. Please checkout the  [wiki](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt) for more information!

closed 2025-05-25 by FireSon