Zendure/Zendure-HA #238: v1.0.41-pre6 | Whose AB1000 is this ? | Since one hour my Hub1200 has 6 x AB1000 connected | I don't know that SN

Linus86529, 2025-05-12:
Since one hour I got 6 x AB1000 connected to my Hub1200. One AB1000 partly show data. The other has no valid Data at all. The 2nd one in the device list is a clone of one of my real AB1000. This is since the HA restart after an update of another HA-integration. This happened some days ago too. 

[image]

FireSon, 2025-05-12:
OK, can you investigate the differences between the names of the batteries? The integration looks for the serial number of the battery, if it is not there it creates a new one. The assumption is that serial numbers do not change...

Linus86529, 2025-05-13:
@FireSon 
The battery SN I've never seen before is the one mentioned earlier above. 'Seriennummer AO4HLEJFC00201'
The 2nd battery is on of my battery pack 'Seriennummer AO4HLC9LDR01201'. This battery has been a kind of a clone of the the real and existing one 'AO4HLC9LDR01201'. The clone has had no valid entities and absolutely no values.

Linus86529, 2025-05-13:
@FireSon Surprisingly today I got another ghost battery 'Seriennummer AO4HLC9LDR01052'.
What reason or situation or mode change would forces a kind of 're-detect batteries'?

[image]

Linus86529, 2025-05-13:
@FireSon Sorry, I forgot to mention all my real batteries SN. Therefore I provide you the hole list of my devices. The battery with the dash in the column 'Bereich' is the new ghost battery.

[image]

Linus86529, 2025-05-13:
@FireSon Hi, I'm just preparing the start of PRE8. And as usually I check the differences of the new and the previous releases to get an idea what has been changed.

As far as I know is the Hub2000 the big brother of the Hub1200. They can handle the same type of batteries, up to 4 batteries in one battery pack, and both can handle the same type mixture of AB1000 and AB2000.

Right? 

Why are the devices/hub1200.py and the devices/hub2000.py not totally equal?

[image]

Linus86529, 2025-05-13:
@FireSon Sure, the Hub2000 can handle the max Power of 1800, and the Hub1200 can't. I forgot to mention it.

Linus86529, 2025-05-14:
Hello @FireSon, the ghost battery is back. She shows up with the same SN as last time. There is no entry in the home-assistant.log for its SN. She popped up one hour ago.

The only thing that helps in this case to get rid of the ghost battery is to reinstall the entire Z-HA version. I'm going to do this in the next minutes.  

[image]

MacdiverOne, 2025-05-14:
BTW: I lost one of my Hypers today to the Part1 of 2 of a Firmware update which successfully completed but the Hyper never came back from it.  It's now a dead brick without any kind of life signs (not even a single LED shines) but with pending RMA request. So I'm currently switching to 2 x Hyper and 1 x Hub1200 - a bit similar to your config.

Linus86529, 2025-05-14:
Hi @MacdiverOne, well a ticket is for sure the right start. As you said, the data has been sent by Zendure to us without a reason. WTH are they doing at Zendure.

I'm sorry that one of your Hypers got the slip. It's good that you have a backup. A Hub 1200 is more of a spare wheel than a full-fledged replacement in that setup. You should be prepared for the Hub's very sluggish response times. For me, it takes about 140 seconds to go from 50 watts to 500 watts. It's a bit faster on the downshift.

FireSon, 2025-05-16:
Well, since the 'ghost' battery, does never report any properties, I think it might be better to create the batteries when a property is reported. In the code that is a sligtly other place but easy to do. Let's try that first, it it does not work, we will find a way to get rid of them 😄