Zendure/Zendure-HA #239: Hyper2000 entities gone

DeckardCain2014, 2025-05-12:
Hi,

since three hours all my Hyper2000 entities are gone. I tried re-loading the integration and restarting HomeAssistant. In the Zendure app everything is working.

HA Version: 2025.5.1
Zendure-HA Version: 1.0.40

[image]

Linus86529, 2025-05-12:
@DeckardCain2014: Moin, Du hast bestimmt die Version genommen, die per Default zum Download bereitsteht. Die hat noch einige Shortfalls.

Wenn Du es mit einer aktuelleren Version versuchen möchtest, installiere die Z-HA Integration in einer der weiterentwickelteren [PRE-Versionen.](https://github.com/FireSon/Zendure-HA/releases) Ich verwende die v1.0.41-pre6.

In HACS kannst Du die PRE-Versionen über diese Schritte finden:

[image]

[image]

[image]

DeckardCain2014, 2025-05-12:
Jetzt habe ich zwei Manager und immer noch keine Entity Daten 😁

DeckardCain2014, 2025-05-12:
Did the update, re-added it, still nothing

DeckardCain2014, 2025-05-13:
```
2025-05-13 07:45:54.575 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:45:54.687 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 7
2025-05-13 07:45:55.722 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:45:55.831 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 0
2025-05-13 07:45:55.839 DEBUG (MainThread) [custom_components.zendure_ha] Open API connection
2025-05-13 07:45:55.839 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
2025-05-13 07:45:56.639 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/eu => Mqtt: mqtteu.zen-iot.com
2025-05-13 07:45:56.640 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
2025-05-13 07:45:56.729 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding device: FGs9xU5b Hyper 2000
2025-05-13 07:45:56.729 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Data: {'id': 170728, 'deviceKey': 'FGs9xU5b', 'snNumber': 'EE1LYMGJN010012', 'name': 'Hyper2000', 'productId': 68, 'productKey': 'gDa3tb', 'onlineFlag': '00601', 'productName': 'Hyper 2000', 'wifiStatus': True, 'blueState': False, 'fourGStatus': False, 'isShareFlag': '00601', 'input': False, 'output': False, 'electricity': 5, 'hasPack': True, 'restState': 0, 'upsMode': False, 'upgradeStatusDes': '', 'productType': 17, 'deviceType': 0, 'upgradeStatus': {}, 'bindId': 0, 'bindStatus': 0, 'batteryCode': '', 'packList': [], 'inputPower': 0, 'outputPower': 0, 'slowChargePower': 0, 'temperature': 0, 'temperatureUnit': 0, 'remainOutTime': 0, 'bindType': 0, 'seriesMode': 0, 'parallelMode': 0, 'networkType': 0, 'standard': 'E', 'isSwitch': False, 'url': '', 'remainOil': 0, 'genMode': 0, 'shortCode': 'E1', 'phaseCheck': 0, 'pass': 2, 'clusterList': [], 'clusterNotice': False, 'nodeStatus': 0, 'clusterId': None, 'phase': None, 'reverseState': 0, 'meterId': 0, 'hemsId': 0, 'totalPower': 0, 'keyFlag': 0}
2025-05-13 07:45:56.989 DEBUG (MainThread) [custom_components.zendure_ha.sensor] Restored state for sensor.hyper2000_aggr_charge_totalk_wh: 0.0
2025-05-13 07:45:56.990 DEBUG (MainThread) [custom_components.zendure_ha.sensor] Restored state for sensor.hyper2000_aggr_discharge_totalk_wh: 0.0
2025-05-13 07:45:56.990 DEBUG (MainThread) [custom_components.zendure_ha.sensor] Restored state for sensor.hyper2000_aggr_solar_totalk_wh: 0.0
2025-05-13 07:45:57.013 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Found: 1 devices
2025-05-13 07:45:57.013 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding sensors Zendure Manager
2025-05-13 07:45:57.013 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 0
2025-05-13 07:45:57.013 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.power_actual to _update_smart_energyp1
2025-05-13 07:45:57.309 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Zendure Manager initialized
2025-05-13 07:45:57.309 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-05-13 07:45:57.340 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching Zendure Manager data in 0.031 seconds (success: True)
2025-05-13 07:45:57.361 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:45:57.386 INFO (paho-mqtt-client-zendureMqtt) [custom_components.zendure_ha.binary_sensor] Update binairy_sensor: Hyper2000-MqttOnline => True
2025-05-13 07:45:57.451 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 7
2025-05-13 07:45:58.510 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:45:58.610 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 7
2025-05-13 07:45:59.643 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:45:59.759 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 7
2025-05-13 07:46:00.806 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:46:00.906 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 7
2025-05-13 07:46:01.967 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:46:02.074 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 7
[... cut ...]
2025-05-13 07:46:09.977 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
2025-05-13 07:46:10.080 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 disconnected from MQTT broker with return code 7
2025-05-13 07:46:11.123 INFO (paho-mqtt-client-FGs9xU5b) [custom_components.zendure_ha.zenduredevice] Device Hyper2000 Mqtt Client has been connected, return code: 0
 ```

closed 2025-05-15 by DeckardCain2014