Zendure/Zendure-HA #241: hyper 2000 over shoot demand

FrColin, 2025-05-12:
Hi,
Since the update of 1-0-41-pre7
I got the P1 parameter going to négative value.
The output is over the demand by many watts 100, 300
Try reset or recharing didn't solve the problem.
The 1.0.41-pre6 was fine
Any change in the calculation between the two version ?

FrColin, 2025-05-12:
here is some log output
```
16:49:00.092 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 24 power: 0 operation: 2
2025-05-12 16:49:00.092 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Wait 10 sec for state change p1: 24
2025-05-12 16:49:00.244 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696833, 'product': 'solarFlowPro', 'timestamp': 1747061339, 'properties': {'solarInputPower': 290, 'outputPackPower': 290, 'solarPower1': 199, 'solarPower2': 91}}
2025-05-12 16:49:01.283 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696835, 'product': 'solarFlowPro', 'timestamp': 1747061340, 'properties': {'solarInputPower': 347, 'outputPackPower': 39, 'outputHomePower': 308, 'solarPower1': 238, 'solarPower2': 109}}
2025-05-12 16:49:02.323 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696837, 'product': 'solarFlowPro', 'timestamp': 1747061341, 'properties': {'solarInputPower': 377, 'outputPackPower': 48, 'outputHomePower': 329, 'solarPower1': 256, 'solarPower2': 121}}
2025-05-12 16:49:03.350 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696839, 'product': 'solarFlowPro', 'timestamp': 1747061342, 'properties': {'solarInputPower': 408, 'outputPackPower': 79, 'solarPower1': 277, 'solarPower2': 131}}
2025-05-12 16:49:04.382 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696841, 'product': 'solarFlowPro', 'timestamp': 1747061343, 'properties': {'solarInputPower': 424, 'outputPackPower': 95, 'solarPower1': 286, 'solarPower2': 138}}
2025-05-12 16:49:05.615 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696843, 'product': 'solarFlowPro', 'timestamp': 1747061344, 'properties': {'electricLevel': 57, 'solarInputPower': 454, 'outputPackPower': 125, 'solarPower1': 305, 'solarPower2': 149, 'solarPower1Cycle': 114, 'solarPower2Cycle': 52, 'outputHomePowerCycle': 21, 'outputPackPowerCycle': 145}}
2025-05-12 16:49:06.462 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696845, 'product': 'solarFlowPro', 'timestamp': 1747061345, 'properties': {'solarInputPower': 504, 'outputPackPower': 175, 'solarPower1': 336, 'solarPower2': 168}}
2025-05-12 16:49:07.073 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 30 power: 658 operation: 2
2025-05-12 16:49:07.073 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 688 stateManagerState.DISCHARGING capacity: 188 max: 1200
2025-05-12 16:49:07.073 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper2000 => 688 capacity 188 program: True
2025-05-12 16:49:07.074 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Invoke function Hyper2000 => {"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 0, "chargingPower": 0, "freq": 2, "outPower": 688}, "msgType": 1, "autoModel": 8}], "function": "deviceAutomation", "messageId": 700022, "deviceKey": "59YfaLt8", "timestamp": 1747061347}
2025-05-12 16:49:07.105 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper2000/function/invoke => {'arguments': [{'autoModelProgram': 2, 'autoModelValue': {'chargingType': 0, 'chargingPower': 0, 'freq': 2, 'outPower': 688}, 'msgType': 1, 'autoModel': 8}], 'function': 'deviceAutomation', 'messageId': 700022, 'deviceKey': '59YfaLt8', 'timestamp': 1747061347}
2025-05-12 16:49:07.741 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/function/invoke/reply => {'messageId': 700022, 'timestamp': 1747061347, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}
2025-05-12 16:49:07.741 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zenduredevice] Receive: Hyper2000 => ready!
2025-05-12 16:49:07.951 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696847, 'product': 'solarFlowPro', 'timestamp': 1747061347, 'properties': {'solarInputPower': 580, 'outputPackPower': 251, 'solarPower1': 400, 'solarPower2': 180}}
2025-05-12 16:49:08.991 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696849, 'product': 'solarFlowPro', 'timestamp': 1747061348, 'pro [... cut ...]
2025-05-12 16:49:30.658 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696891, 'product': 'solarFlowPro', 'timestamp': 1747061370, 'properties': {'packInputPower': 367, 'outputHomePower': 1107}}
2025-05-12 16:49:31.680 INFO (paho-mqtt-client-56519:1c1fc034-fbef-4cd7-afa0-3ec68835b119) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper2000/properties/report => {'messageId': 696893, 'product': 'solarFlowPro', 'timestamp': 1747061371, 'properties': {'packInputPower': 367}}
2025-05-12
```

FrColin, 2025-05-12:
reverting to 1.0.41-pre6 is OK
so the bad thing is [https://github.com/FireSon/Zendure-HA/commit/88ab2a491e28d44f9b32b3b87acad4b4eb455643](url)
taking the outputHomePower instead of packInputPower make the output power increase to max

Linus86529, 2025-05-12:
Ich habe das heute Morgen beobachtet. Der Grund ist das [hier.](https://github.com/FireSon/Zendure-HA/issues/180#issuecomment-2872666893)

FireSon, 2025-05-12:
Sorry for the inconveniace, fixed in pre8

closed 2025-05-12 by FireSon