Zendure/Zendure-HA #249: Zendure battery calculation | remain-out time is always calculated “incorrectly” | a request

Linus86529, 2025-05-13:
@FireSon 

Does it bother anyone else besides me that the remain-out time is always calculated “incorrectly”? Why wrong? It's right!

Yes, but ...

Zendure always calculates the “Remain Out Time” starting from the current minSOC down to a SOCbat of 0 %. Who lets their batteries run down to SOC of 0 %?

Depending on the time of year, I set the minSOC to either 20 % in summer or 50 % in winter. In both cases, Zendure always calculates the remaining runtime for me (incorrectly).

If for some reason I need to know how long my battey pack could deliver power at a certain amount of power output up to a SOCbat of X %, I have to reach for the calculator. And that's annoying!

How would it be, if the Z-HA integration did this “better” than Zendure and instead of calculating the remaining battery life down to a SOCbat of 0 %, did this up to the adjustable minSOC?

[image]

Linus86529, 2025-05-13:
@FireSon Instead of "...starting from the current minSOC..." it schould be "...starting from the current SOCbat..." I'm sorry for the typo.

Linus86529, 2025-05-13:
@Linus86529 If we take values on the screen shot for example. 

Remain Out Time = 17 Std. 15 Min.
SOCbat = 73 %
Discharge Power =  176 W
minSOC = 20 %
Akku Capacity = 4 * 960 Wh = 3840 Wh   # 4 * AB1000

Real Remain Out Rime = 3840 Wh / 100 % * (73 - 20) % / 176 W = 11,563 Std. 

The real Remain Out Time is 11 Std. 33 Min. 

A tiny difference isn't it?

harrymayr, 2025-05-14:
A template sensor can do the trick ;-)
MinSoc = 14%

[image]

[image]

Linus86529, 2025-05-15:
@harrymayr Das könnte ich machen. Danke dafür.

Und wenn der Offset Einsteller nicht kommt, die Diskussion darüber habe ich gelesen, dann bastle ich mir zwei Helfer. Einer mit dem Namen Nummern-Helfer für den Offset-Bereich -50..+50 und einen "my_power_actual_plus_offset" und kopiere den folgenden JinjaCode rein:

{{ states('sensor.3em_31c_labc_w') | float(0) + states('input_number.nr_p1_offset') | float(0) }}

... und übergebe den Helfernamen "my_power_actual_plus_offset" im Konfigurationsdialog an die Zeile P1-Sensor. Geht auch.

harrymayr, 2025-05-16:
> A template sensor can do the trick ;-) MinSoc = 14%

Habe jetzt noch einen anderen Template-Sensor basierend auf der Leistung gemacht. Der zappelt nicht so stark rum. Ich schau mal, ob ich den auch für die Integration bereitstelle. Mein Hyper zeigt mir auch (nicht mehr?) die Dauer bis zur vollständigen Ladung an. Unten wäre auch ein Template Sensor dafür.
Man muss halt jeweils die Leistung seiner Batterien entsprechend angeben.

[image]

[image]

[image]

harrymayr, 2025-05-16:
Eine Änderung in Zendure-HA könnte [so aussehen](https://github.com/FireSon/Zendure-HA/compare/master...harrymayr:Zendure-HA:TimeUpdate)
Die Lade-Zeit habe ich aber noch nicht in der Integration getestet, ist grad so wenig Sonne ;-D ;-)

harrymayr, 2025-05-17:
Ladezeit wird noch nicht aktualisiert, da komme ich aber frühestens heute Abend dazu.

harrymayr, 2025-05-17:
gefixt:

[image]

Linus86529, 2025-05-17:
Moin @harrymayr perfekt! Danke für deine ersten Anregungen. Ich habe mich deswegen für ein paar Stündchen zurückgezogen und es etwas übertrieben. Aber wenn ich schon rangehe ...

Ich hab mir für jedes Zen Device ein Dashboard gebaut und die Template Helfer dort eingepflanzt.

[image]

[image]

Der Jinja2 Code fürs Template steht weiter unten. Falls jemand den brauchen kann...
Ich konnte mich nicht recht entscheiden ob die klassische Ausgabe 'Tag(e), hh Std., mm Min.' oder das Arival Date Format: 'HH:MM am dd.mm.JJJJ' besser passt. Darum sind beide Möglichkeiten enthalten und können am Ende entsprechend ausgewählt werden.

`{# Set default values and convert to numbers, handling potential errors #}
{# Use -1 for current_soc to differentiate invalid state from actual 0% #}
{% set remain_soc0_hours = states('sensor.hyper_2000_remain_out_time') | float(0) %}
{% set current_soc = states('sensor.hyper_2000_electric_level') | int(-1) %}
{% set min_soc = states('number.hyper_2000_min_soc') | int(0) %}

{% set time_to_minsoc_hours = 0 %}
{% set time_to_minsoc_formatted = 'N/A' %} {# Default value if calculation is not possible #}
{% set arrival_time_formatted = 'N/A' %}   {# Default value #}

{# --- Calculation Logic --- #}

{# Check if sensors have valid data and if minSOC is not already reached #}
{% if current_soc is number and current_soc >= 0 and remain_soc0_hours is number and remain_soc0_hours > 0 and current_soc > min_soc %}
  {# Calculate time to minSOC in hours #}
  {# Avoid division by zero. current_soc > min_soc implies current_soc > 0 if min_soc >= 0 #}
  {% if current_soc > 0 %}
    {% set time_to_minsoc_hours = remain_soc0_hours * (current_soc - min_soc) / current_soc %}
  {% else %}
    {# This case should theoretically not be reached due to the outer if, but as a safeguard #}
    {% set time_to_minsoc_hours = 0 %}
  {% endif %}

  {# Convert total calculated hours to total minutes, rounded #}
  {% set total_minutes = (time_to_minsoc_hours * 60) | round(0) %}

  {# --- Format time to minSOC as "nn Tage, HH Std., MM Min." or "HH Std., MM Min." --- #}
  {% set total_minutes_in_day = 24 * 60 %}
  {% set num_days = (total_minutes / total_minutes_in_day) | int %}
  {% set remaining_minutes_for_display = total_minutes % total_minutes_in_day %}
  {% set display_hours = (remaining_minutes_for_display / 60) | int %}
  {% set display_minutes = remaining_minutes_for_display % 60 %}

  {# Build the time string parts #}
  {% set time_parts = [] %}

  {% if num_days > 0 %}
    {% if num_days == 1 %}
      {% set time_parts = time_parts + ['1 Tag'] %}
    {% else %}
      {% set time_parts = time_parts + [num_days ~ ' Tage'] %}
    {% endif %}
  {% endif %}

  {% if display_hours > 0 %}
    {% set time_parts = time_parts + [display_hours ~ ' Std.'] %}
  {% endif %}

  {% if display_minutes > 0 or (num_days == 0 and display_hours == 0) %} {# Always show minutes if time is 00:xx #}
     {% set time_parts = time_parts + [display_minutes ~ ' Min.'] %}
  {% endif %}

  {# Join the parts with a comma and space, or just space if only two parts #}
  {% if time_parts | length > 1 %}
    {% set time_to_minsoc_formatted = time_parts | join(', ') %}
  {% elif time_parts | length == 1 %}
    {% set time_to_minsoc_formatted = time_parts | first %}
  {% else %}
    {# This case should ideally not happen if total_minutes >= 0, but as a safeguard #}
    {% set time_to_minsoc_formatted = '0 Min.' %}
  {% endif %}

  {# --- Calculate and Format arrival time --- #}
  {% set now_dt = now() %} {# Get current datetime object #}
  {# Add the calculated duration to the current time #}
  {% set arrival_dt = now_dt + timedelta(hours=time_to_minsoc_hours) %}

  {# Format the arrival datetime object as HH:MM am DD.MM.YYYY #}
  {% set arrival_time_formatted = arrival_dt.strftime('%H:%M am %d.%m.%Y') %}

{% elif current_soc is number and current_soc >= 0 and current_soc <= min_soc %}
  {# Case where minSOC is already reached or exceeded, and current_soc is valid (>= 0) #}
  {% set time_to_minsoc_formatted = '0 Min.' %} {# Display 0 Min. when already reached #}
  {# Arrival is now, format with current date #}
  {% set arrival_time_formatted = now().strftime('%H:%M am %d.%m.%Y') %}

{% else %}
  {# Handle cases with invalid or missing sensor data #}
  {% set time_to_minsoc_formatted = 'Invalid Data' %}
  {% set arrival_time_formatted = 'Invalid Data' %}

{% endif %}

{# --- Output --- #}
{{ time_to_minsoc_formatted }}  {# " 3 Tage, 9 Std., 30 Min." #}
{# or:
  {{ arrival_time_formatted }}     "10:31 am 18.05.2025"
#}
`

Linus86529, 2025-05-17:
@harrymayr Das mit der [Glättung](https://github.com/FireSon/Zendure-HA/issues/249#issuecomment-2887653071) ist sehr gut. Damit reduzieren sich auch die Schriebvorgängein die Influx-DB. Ist dein Python [Code](https://github.com/FireSon/Zendure-HA/issues/249#issuecomment-2887822924) nur zu Testenzwecken gedacht oder kommt der in einer nächsten pre-n ?.

Wenn es recht ist, dann würde ich den gerne mal bei mir reinklopfen ;-)

harrymayr, 2025-05-17:
Das ist erst mal quick and dirty. Werde es noch überarbeiten und einen PR starten. Dann liegt an FireSon ;-)

harrymayr, 2025-05-17:
@Linus86529 Hab jetzt noch was geändert und auch die Übersetzung (bisher nur DE) eingetragen. Ich habe auch einen neuen Sensor für die Ladezeit eingefügt. Das Überschreiben des ursprünglichen Sensors hat zwar funktioniert, aber da der Hyper anscheinend keine Zeit liefert, wurden immer mal wieder 999 vom Hyper überschrieben. Das hat mir nicht gefallen. :-)

Linus86529, 2025-05-17:
@harrymayr Ich bin begeistert! Ich hatte mir für meine Template-Version überlegt, die Anzeige von 999 Stunden einfach durch drei Striche oder etwas Ähnliches zu ersetzen, vielleicht durch „unbekannt” oder etwas Ähnliches, da die Anzeige von 999 Stunden ziemlich nichtssagend ist.

Bleib dran! Ich finde es um einiges besser als alles davor.

harrymayr, 2025-05-17:
Ja, das mit den 999 gefällt mir auch noch nicht. Da bin ich an überlegen, die Zeit mit der letzten Entlade/Ladeleistung ungleich 0 (oder eine Mindestwert von 50 für die Leistung) zu berechnen

Linus86529, 2025-05-17:
Oder das. Alles passender als 999. Ich hab mich schon gewundert warum die nicht die Chinesische Glückszahl im Trippel genommen haben. --Gesendet mit der GMX Mail App

Linus86529, 2025-05-27:
@harrymayr 
Heute hab ich mal was verrücktes probiert ;-)

Ich habe in der laufenden Z-HA-Instanz (v1.0.43) einfach das automatisch angebotene Update auf v1.0.44 gestartet, ohne zuvor, wie bei allen Vorgängern, die alte Installation zu stoppen und zu löschen/entfernen.

Ich bin begeistert! Es hat funktioniert. Ich habe keine Geister-Akkus dazubekommen oder sonstige Ungereimtheiten gesehen. Vor dem HA-Neustart habe ich nur kurz die interessanten Python-Dateien gecheckt und die Konstanten für mich angepasst – und los ging's. Das war ein fool proof Update.

Der einzige Schönheitsfehler war beim Hub1200, ein binärer Sensor der nicht mehr verfügbar ist. Dafür gibt es jetzt den Select „select.hub_1200_auto_recover”. Den Binärsensor habe ich auf der Geräteseite einfach gelöscht und das Problemchen war behoben.

In der Device-Ansicht für den „sensor.hyper_2000_remain_out_time” sehe ich noch 999 statt etwas anderem.
Aber was solls...

[image]

harrymayr, 2025-05-27:
> In der Device-Ansicht für den „sensor.hyper_2000_remain_out_time” sehe ich noch 999 statt etwas anderem.
> Aber was solls...

Du könntest in der Zendurebase.py ganz unten die beiden
`return 999`
in
`return None`
ändern

FireSon, 2025-05-28:
Well I think the remaining time should now be correct 😄 The reason I kept 999 is that 'unknown' is normally used for an error, so I expect many questions about it. Unfortunately, Home Assistant does not have an option for 'forever'...

closed 2025-05-28 by FireSon