Zendure/Zendure-HA #250: v1.0.41-pre8 | Hyper2000 consuming AC_Power during the night in intervals| Started last night at 01:00(L)

Linus86529, 2025-05-14:
@FireSon Good morning,  this is the first time seeing this behavior on an HA plot. Who has an idea why this happens. Please check the attached HA plot from last night. The Hyper 2000 Grid Input Power is drawn in green. Thanks. 

[image]

harrymayr, 2025-05-14:
Moin @Linus86529, ohne Log wird das beliebig kompliziert. Aber wenn du die Daten von outputHomePower, gridInputPower, packInputPower, outputPackPower von Hyper und Hub und deinem P1 Sensor liefert, und zwar als Plot und CSV (über die 3 Punkte oben rechts), kann ich heute Abend mal versuchen rauszufinden was passiert ist.

Linus86529, 2025-05-14:
Moin @harrymayr, darum hab ich eben eines Aufgezeichet. Da ich mir nicht sicher war, ob der Hyper nach dem HA Neustart das Verhalten noch zeigt, habe ich nur das Debug Log eingeschaltet und den HA Neustart nicht gemacht. 

[clean___Linus86529___v1.0.41-pre8___GridInputPower_at_a_low_level____home-assistant_zendure_ha_2025-05-14T10-20-42.319Z.log](https://github.com/user-attachments/files/20205528/clean___Linus86529___v1.0.41-pre8___GridInputPower_at_a_low_level____home-assistant_zendure_ha_2025-05-14T10-20-42.319Z.log)

harrymayr, 2025-05-14:
> Moin [@harrymayr](https://github.com/harrymayr), darum hab ich eben eines Aufgezeichet. Da ich mir nicht sicher war, ob der Hyper nach dem HA Neustart das Verhalten noch zeigt, habe ich nur das [...]

Hallo @Linus86529: Ich habe mir jetzt mal das Log angeschaut. Das was hier vom Netz in den Hyper fliest, ist die Differenz zwischen dem, was der Hyper meint in den Akku zu laden (outputPackPower) und das was zu dem Zeitpunkt über die Solareingänge (solarInputPower) reinkommt. Immer wenn nicht genug PV-Leistung zur aktuellen Ladeleistung vorhanden ist, nimmt er die aus dem Netz.

Wenn ich jetzt mal davon ausgehe, dass du Nachts keine Leistung an deinen Solareingängen des Hypers hast, beschreibt das Log nicht den Verhalten von heute Nacht.

[image]

> Aber wenn du die Daten von outputHomePower, gridInputPower, packInputPower, outputPackPower von Hyper und Hub und deinem P1 Sensor lieferst, und zwar als Plot und CSV (über die 3 Punkte oben rechts), kann ich heute Abend mal versuchen rauszufinden was passiert ist.

Also kann ich vielleicht noch mit den Angaben von heute Nacht (+ die SolarInputPower) was rausfinden.

Schönen Abend,

Harry (der sich mal dringend einen Parser für die LOG-Dateien schreiben muss...)

Linus86529, 2025-05-15:
Moin @harrymayr, ich hab deine Nachricht erst um 02:00(L) gesehen. Ich war da nicht mehr wach genug um das CSV File zu ziehen und ...

Wie diese Art von Fehlern nun mal sind, tauchen die nicht mehr auf wenn Du denen eine Fangschaltung aufgestellt hat. Darum habe ich die Daten aus der Histiory-Kiste gekramt.

CSV-Datei:

[Linus86529___2025-05-14___Hyper2000_zieht_AC-Power_und_sieht_PV-Power_in_der_Nacht___history.csv](https://github.com/user-attachments/files/20226181/Linus86529___2025-05-14___Hyper2000_zieht_AC-Power_und_sieht_PV-Power_in_der_Nacht___history.csv)

Ich bin ab 13:00(L) zu einem Termin in Ingolstadt. Bin erst wieder ab ca. 18:00(L) zurück.

Hatte vorhin noch eine weitere eigenartige Beobachtung gemacht. Der Hyper blieb für mehr als 2 Minuten bei 600Watt 'kleben' obwohl der P1-Wert zu der Zeit Minuswerte um -300W angezeigt hat.

Die CSV-Datei zu der Beobachtung:

[Linus86529___Hyper_hängt_über_2_Minuten_auf_600W_fest_obwohl_P1-Werte_im_hohe.csv](https://github.com/user-attachments/files/20226246/Linus86529___Hyper_hangt_uber_2_Minuten_auf_600W_fest_obwohl_P1-Werte_im_hohe.csv)

[image]

harrymayr, 2025-05-15:
Wenn du sch PV Power in der Nacht schreibst, ist es vielleicht doch der selbe Fehler, der dann aber bei Zendure liegt. Auch [hier](https://github.com/FireSon/Zendure-HA/issues/252#issuecomment-2881764103) wurde von SolarIput während der Nacht berichtet. Ich schau es mir heute Abend mal an.

harrymayr, 2025-05-15:
PV-Power war jetzt nicht in deinem CSV, aber wenn ich mir mal die Zeit um 3:50 rausnehme und mir gridInputPower und outputPackPower des Hypers anschaue, so sieht das ähnlich aus wie oben, nur das so gut wie keine Solar-Power ansteht.
Meine Vermutung: Irgendetwas (falsche Erkennung am SolarInputPower) triggert den Hyper eine Ladung zu starten, und da keine PV-Leistung da ist, zieht er die Leistung aus dem Netz.
Nach meinem Kenntnisstand können wir die Ladung durch PV nicht steuern (ggf. nur durch den Bypass), ich sehe also nicht, was man dagegen machen kann. Vielleicht kann man den Wert VoltWakeup (steht der bei dir auch auf 0?) verändern, so dass es mehr Leistung braucht, um eine Ladung über PV zu starten.

[image]

closed 2025-07-31 by FireSon