Zendure/Zendure-HA #252: sensors pack_input_power and pack_output_power not updated on a regular basis

CWILDMOUNTAIN, 2025-05-14:
I‘m not sure whether this is a HA issue or from the integration (updated to HA Core 2025.5 and 1.0.41 pre-8 the same day) but I‘m facing the issue, that power pack_input_power and pack_output_power are not updated (for example when battery is fully charged or no solar power is there). 

This results in delayed values of kWh template sensors and further there are no updated values in energy dashboard or delayed values respectively (see screenshot). 

Is it possible to update those sensor at least once per hour? I‘m working the hub‘s sensors.

(Setup: Hub 2000+ACE+2 AB2000, cloud-based)

[image]

FireSon, 2025-05-14:
That is strange, normally each 90 seconds, all properties are queried from the devices. Since I do not own these devices I am not 100% sure that the report all properties when the command is sent. Can you check the logfile with mqttlog on, if you see any activity if the battery is fully charged?

CWILDMOUNTAIN, 2025-05-14:
Yes will do tomorrow.
I could do a log for now as there is no Solar input anymore and it’s the same issue  (input_power, output_power)

CWILDMOUNTAIN, 2025-05-14:
I observe the following (after about 15min logging)
a) there is activity (update request every 90s)
b) it‘s only updating the ACE 
c) strangly the command is custom_components.zendure_ha.zendurermanager] Zendure local => Zendure Manager => /xxxxxxx/xxxxxxx/time-sync

it seems like a local request, but I use the cloud access (well, at least I thought)

CWILDMOUNTAIN, 2025-05-14:
.. and after the sun had set even more is missing: all entities of the batteries are "not available" anymore (attached pic)
But ACE is doing freak things (attached vid) and although it's midnight showing solarinput from time to time .. maybe all symptoms are part of the same issue (wrong values lead to no values??) 

[image]
https://github.com/user-attachments/assets/c0c40fff-ae57-4a5f-89fc-c8fa50c07240

MacdiverOne, 2025-05-15:
That looks like postponed (for whatsoever reason) MQTT data coming in as bulk…

FireSon, 2025-05-16:
@CWILDMOUNTAIN At least it looks cool 😄 Are you sure you are not using local MQTT? The logging indicates that:
custom_components.zendure_ha.zendurermanager] Zendure local => Zendure Manager => /8bM93H/xxxxxxx/time-sync

CWILDMOUNTAIN, 2025-05-16:
Well .. I'm pretty sure not to, otherwise I should see data populating in MQTT-Explorer, right?

FireSon, 2025-05-16:
Yes that is true! It just seems strange that only only one device would have these 'problems'. If you look at the Zendure App during these 'blackouts' is the information there correct and up to date?

CWILDMOUNTAIN, 2025-05-16:
yes, all data coming in nicely and live in the App, no 'blackouts'

FireSon, 2025-05-16:
That is really strange, since the App uses the same MQTT broker as the integration. I will try to think of something...

CWILDMOUNTAIN, 2025-05-18:
so I did a little bit of testing, resetting the hub and recunfiguring .. I‘m 100% sure there‘s no local mqtt (this I just can‘t get running, it‘s already issued in here https://github.com/FireSon/Zendure-HA/issues/237#issuecomment-2877536005). 

so When I do no use the Integration (App only) there 'freaking' values from the ACE, but as soon as I configure the intergration it’s back (also in the App!) 

Further I noticed, that there is an entity in the ACE 'Hyper Energy Program' and this is jumping between Smart Matching and None back and forth. 
Well.. needless to say the ACE has no Hyper Energy Program at all. 
So Integration of the ACE still seems to be tricky.

CWILDMOUNTAIN, 2025-05-20:
Is it possible somthing like this https://github.com/FireSon/Zendure-HA/issues/219#issuecomment-2895342299 in the pre8 screwed my hub as well? 

I mean it’s not exactely a boot loop, but in a way it "reboots" power distribution all the time ..

CWILDMOUNTAIN, 2025-05-28:
I cannot get the localMQTT running .. did not find the problem.

well .. with 1.0.44 values are coming in properly, that's good.

my main issue is still not working (https://github.com/FireSon/Zendure-HA/issues/269). One of my Hub is still crashing when output exceeds around 850W .. it's really annoying.

Also I cannot get the smart matching mode running. Both of my 2 Hubs 2000 are not reacting properly to the smart matching commands oder manual mode .. I'm not sure but is it possible my issue has something to to with it, maybe wrong limits written to the hub?

CWILDMOUNTAIN, 2025-05-28:
Oh yes, there is .. that's my workaround right now .. not to happy about it though.
all devices are on the newest fw from the beginning on.