Zendure/Zendure-HA #254: In version 1.0.41-pre8:  No local MQTT connection other broker

sankukace, 2025-05-15:
In version 1.0.41-pre8:

No local MQTT connection when I check "local" because my broker is on a different IP.

In a previous pre-release, there was a prompt for the address — but not anymore. So no connection via Bluetooth; only cloud works.

How can I set the IP address, username, and password for the MQTT broker?

Thanks again.

FireSon, 2025-05-16:
@sankukace what MQTT broker do you use? 
The reason for this, and why it is now called "Use Mosquitto MQTT AddOn" is the authentication of the MQTT connection. For this it creates the users you have in your screenshot. Mosquitto uses these users to do the authentication.

sankukace, 2025-05-19:
Hi,

I don’t want to install the MQTT broker on my Home Assistant because I prefer to separate Z2M, MQTT, etc.

I don’t think I’m the only one in this situation.

Thanks

bilbon0bud, 2025-05-21:
hello

i use a diffrent mqtt broker to, it could be cool if we can configure IP + Port + username + password.
maybe you could add an advance configuration for advanced users.

Best regards

otbe, 2025-05-21:
Main reason why I want to try this setup is that my hyper2000 sometimes just stops being reachable via Wifi. This seems to be a more common issue and thats why I want to give the local BT connection a go :)

harrymayr, 2025-05-21:
> thats why I want to give the local BT connection a go :)

BT is only used to change the MQTT server and credentials, the rest of the communication and control is still WiFi.