Zendure/Zendure-HA #257: For version 1.0.41-pre8: No battery values

STUBBYs1, 2025-05-16:
Hello,
Unfortunately, version 1.0.41-pre8:
doesn't provide any battery values ​​for me, and this applies to all

[image]

Maggi0r, 2025-05-16:
Hmm you are on MQTT or Cloud. By my System i have Values.

Greets Maggi

STUBBYs1, 2025-05-16:
@Maggi0r  

[image]

harrymayr, 2025-05-18:
My experience: changing configuration or update the version always need a HA restart to work properly

STUBBYs1, 2025-05-18:
@harrymayr  Home Assistant was restarted during each installation, which is logical.

harrymayr, 2025-05-18:
Than I only have the idea, that local MQTT does not work, because of BT incompatibility, insufficient signal, ...
Local MQTT requires a BT connection to the Zendure device. Not sure yet, if pre4 had already Local MQTT.

If you provide a log, I can try to figure out, what's the problem.

STUBBYs1, 2025-05-18:
@harrymayr 

[home-assistant_zendure_ha_2025-05-18T08-23-18.410Z.log](https://github.com/user-attachments/files/20272254/home-assistant_zendure_ha_2025-05-18T08-23-18.410Z.log)
[home-assistant_zendure_ha_2025-05-18T09-29-22.453Z.log](https://github.com/user-attachments/files/20272255/home-assistant_zendure_ha_2025-05-18T09-29-22.453Z.log)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Hello harrymayr,

Here I am sending you my data for the update from 1.041 pre4 to 1.0.41-pre8.

After the update and HA restart, some data is not visible, but was visible before the update.

Before the update, there were 16 devices + 419 entities. After the update, there are now only 4 devices + 176 entities.

STUBBYs1, 2025-05-18:
@harrymayr 
All settings will reset to default when HA or just the dashboard is reloaded.

[image]
Thanks

harrymayr, 2025-05-18:
For me it looks like, that even with localMQTT enabled, the pre4 has a fallback to serverMQTT, while the pre8 tries to reach lokalMQTT without success. So I think, none of the version work with localMQTT, but perhaps @FireSon can check this and align with his expectations wrt BT.

I propose for now: don't use localMQTT

FireSon, 2025-05-24:
@STUBBYs1 In the new version 41, there is a new entity which reflects the current connection status of the devices. Can you check them? Another important thing for the communication to work is the correct authentication of the Zendure devices, since mosquitto does not allow anonymous users. In one of the first pre-releases the 'device-users' were added with an incorrect password. It might help to remove them all, the integration will add them again with the correct password.
On the log page of mosqitto you can check if a connection is made.

FireSon, 2025-05-25:
@STUBBYs1 In version 43 this should be resolved, please let me know if you still have issues!

closed 2025-05-25 by FireSon