Zendure/Zendure-HA #259: Featur Request Integrate SolarFlow 800 Pro

H0d0rKF, 2025-05-17:
Currently, the SolarFlow 800 Pro cannot be integrated using this option. It isn't recognized, not even via the main account (I've read about the problems with secondary accounts for HEMS devices). I'm not a programmer, but I can provide some assistance with the SF800 Pro itself.

H0d0rKF, 2025-05-17:
Think this courd be closed. Tryed with the newest pre-version and it works.

kaischmerer, 2025-05-19:
@H0d0rKF Did you try with the 2nd account or with the main accound? I only got Solarflow 800 Pro integrated with the main account.

copadewe, 2025-05-21:
How did you get it working? The 800Pro did Not have "Gerätefreigabe"? Only in hems you can Share the whole HEMS, but HEMS is working terrible so i Not Use it.

H0d0rKF, 2025-05-21:
@kaischmerer @copadewe It only works with the main account so far and only with the newest pre-release.

copadewe, 2025-05-25:
> [@kaischmerer](https://github.com/kaischmerer) [@copadewe](https://github.com/copadewe) It only works with the main account so far and only with the newest pre-release.

Hi, thx, but there is no "item share" with the new ones , how it works than?
Where to finde the pre-release - sorry github is not my favorite

closed 2025-05-25 by FireSon