Zendure/Zendure-HA #260: Server Side Values possible?

gintonic26, 2025-05-17:
First of all: THANK you for you work!

I have a Hyper2000, and inside the Zendure App we have values for daily solar engery and aggregated values for daily charge and discharge. 

Is there any possibility to pull these values?  The integration calculates the aggrChargeDaykWh & aggrDischargeDaykWh but no daily solar power. 

Would be nice for the energy dashboard and the values server sided may be a bit more precise if there is a connection problem with homeassistant. 

Thank you.

harrymayr, 2025-05-17:
please use the 1.0.41 pre8. This calculates all the values.

FireSon, 2025-05-25:
@gintonic26 I think @harrymayr is correct, in the last version 43 there are three aggr sensors, one for charging/discharging and solar.