Zendure/Zendure-HA #262: Does I have control or not? with this integration

surfer1264, 2025-05-17:
Subject: Question about Control Elements in the HA Integration

Hi,

I installed version 1.0.40.
The installation went smoothly, and both the Hub1200 and two batteries are displayed with their relevant values — that’s great!
Thanks a lot for making this integration so easy.

**My question:**
You've implemented control elements like `MinimalSoc `and `OutputLimit`.
Are these real, functional control elements?

The reason I’m asking is this:
When I change values such as `MinimalSoc `or `OutputLimit `in the official Zendure app, the changes are immediately reflected in the Home Assistant (HA) integration — without any delay.

However, when I use the sliders in HA to change `MinimalSoc `(e.g., from 15% to 20%), the change is not reflected in the official app. Nothing happens.
The same behavior occurs with `OutputLimit`.

In other words, changes made in HA are not applied to the SF1200. When I leave and return to the configuration page in HA, the old values (as shown in the official app) are back.
This suggests to me that the control elements in HA do not actually affect the functioning of the SF1200.

**Is this the expected behavior?**
Given this, I’m also unsure what the purpose of the ZendureManager is supposed to be.

Thanks a lot for helping clarify this!

**For reference:**
The HA integration is using my main Zendure account.
In the Zendure app, I’m logged in with a secondary account.

harrymayr, 2025-05-17:
Please use Version 1.0.41 pre8 und check again. V40 has several things which does not behave as expected.

surfer1264, 2025-05-17:
Dose Not Work, the wrong device Solarflow 2.0 was installed. All Entities are unavailable.

Wrong Device

I have a SF1200

Linus86529, 2025-05-17:
Moin @surfer1264, nice [Projekt](https://github.com/surfer1264/Zendure-Stuff). Leider kann ich es für mein HW-Setup nicht einsetzen. Ich kann einen Hub1200 und einen Hyper2000 damit nicht zusammen betreiben.

Das funktioniert jedoch mittlerweile recht ordentlich mit der Zendure-HA-Integration im Z-HA-Cluster.

Richtig ist, dass dein SF1200 (Hub 1200) als „Solarflow 2.0” bei dir auftaucht. Nicht gut ist, dass deine Entitäten unavailable sind. Hast du sichergestellt, dass die Entitäten, die du für dein Projekt verwendest, sich nicht mit denen, die durch die Z-HA-Integration erzeugt werden, in die Quere kommen?

Wenn du die pre8 nochmals installierst, entferne bitte in HACS die Z-HA-Integration über das Drei-Punkte-Menü. Das hilft bei mir immer.

Das mit dem Local MQTT würde ich als Einsteiger erstmal nicht aktivieren. Dazu müssen deine Zen-Devices im BT sichtbar sein.

surfer1264, 2025-05-17:
Danke
Ok Neuinstallation klappt.
Ich sehe SF1200 und 2 Akkus als eigenständige Geräte, alle mit Daten. Fein

Dann

Ging ich auf konfigurieren um meinen ShellySensor 3empro hinzuzufügen. 

Danach waren bei beiden Akkus alle Entities ungültig und nicht mehr verfügbar.

Komplett gelöscht, im Hacs neu geladen,
Neue Integration
Den Shelly Sensor gleich in der Startkonfig hinterlegt....

Funktioniert.

Man sieht aber nicht mehr welchen Sensor von Shelly man bei Konfiguration hinterlegt hat.    

Integration scheint sich nicht zu beißen mit meiner bisherigen MQTT Integration der Hubdaten von Zendure. Ich beobachte.

Linus86529, 2025-05-17:
@surfer1264 Das ist erfreulich.

Da wir derzeit mit der pre8 arbeiten, müssen wir noch mit dem einen oder anderen Fehlverhalten leben.

Ich wundere mich nicht mehr darüber, dass mein 3EM-Pro-Sensor nicht mit dem Namen angezeigt wird, den ich ihm irgendwann einmal gegeben habe. Solange es funktioniert, ist das erst einmal okay.

Vielleicht kann @harrymayr herausfinden, ob wir irgendwo sehen können, wo der Name unseres P1-Sensors steht.

Hast du in der Zendure-App alle Energiepläne deaktiviert?

In den ersten Tagen mit der Z-HA wusste ich nicht, was alles in der Zendure-App deaktiviert sein soll, damit sich die beiden nicht gegenseitig beeinflussen oder im schlimmsten Fall behindern.

Die Readme auf GitHub ist leider nicht auf dem aktuellsten Stand.

In der App hatte ich alle drei Phasen des Shelly 3EM Pro im Smart-CT-Modus an den Hub1200 gebunden, so wie man das eben macht. Gleichzeitig habe ich den Betriebsmodus „Smarte Leistungsregelung” im Zendure Manager aktiviert. Zu meinem Erstaunen hat das gut funktioniert. Sogar der Hyper hat sich entsprechend verhalten und zusammen mit dem Hub anteilig eingespeist.

Nach einiger Zeit habe ich dann alle Energiepläne in der Zendure-App deaktiviert, damit nur eine Kontrollinstanz die Geräte steuert.

Viel Spaß mit der Z-HA-Integration!

surfer1264, 2025-05-17:
Integration scheint zu laufen.
Den Zendure Manager habe ich nicht verstanden. Smart Matching ? Da passiert Nix.

ABER

Ich habe den Output Limit Sensor in mein Regelprojekt eingebunden. Also gegen den WR Sensor getauscht. Funktioniert. Siehe Bild die gelben Kringel.

Frage: ist die Abfrage lokal oder geht die API gegen die Cloud. 

[image]

Linus86529, 2025-05-17:
@surfer1264 Wenn Du im Konfigurationmenü "Lokalen Mosquitto MQTT AddOn verwenden" ausgewählt hast, dann soll es lokal laufen. Wenn nicht, dann ist es über die Cloud.

Ich habe es noch nicht hin bekommen auf lokalen MQTT zu switchen, da mein RasPi4 nicht in Reichweite beider Zen-Geräte steht. Somit kann ich zur Zeit den Cloud-Weg nehmen.

Linus86529, 2025-05-17:
Ich vergaß das dafür  BT benötigt wird. Und die Reichweite meinte ich.

Linus86529, 2025-05-17:
@surfer1264 Ich glaub der Soll-Wert Einsteller ist so etwas wie ein Offset von der Null-Linie.

surfer1264, 2025-05-17:
Danke, mein HA Green steht auch zu weit weg.

Fühlt sich die API stabil an? Verfügbarkeit?

Mein Ansinnen mit meinem Projekt war ... den Hub in der cloud zu lassen (nur im Basismode) aber über den Wechselrichter und Shelly zu regeln. Was perfekt funktioniert. 

Eure HA Integration macht Spaß, wenn outputlimit auch immer setzbar ist, also verfügbar. 

War cool das eben schnell in meiner NodeRed Regelung zu switchen. 5 Sekunden Aufwand.

Danke Euch, verfolge das weiter.

Linus86529, 2025-05-17:
@surfer1264 Danke.
Ich verwende den Cloud Mode. Die App ist voll nutzbar und beide Devices laufen in der Z-HA zusammen. Perfekt wird es mit der Zeit.

surfer1264, 2025-05-17:
> [@surfer1264](https://github.com/surfer1264) Was verbirgt sich hinter den zweiten gelben ovalen Kringel? Ein Offset Einsteller?

Der zweite Kringel ist der `OutputLimitSensor` Eurer Integration...ich habe den da nur mal sichtbar gemacht, um zu sehen ob wirklich was passiert

Der erste Kringel ist mein Shelly 3emPRO.

Ich arbeite aber tatsächlich mit einem Sollwert zw. -20 (Nullbezug) und +20 (Nulleinspeisung) je nach Überschussverfügbarkeit

surfer1264, 2025-05-17:
Zendure Manager macht nix bei mir, weder manuell noch automatisch

Energiemodi sind immer aus bei mir, da ich ja eine eigene Regelung betreibe, die deaktiviere ich natürlich.....der letzte Regelwert bleibt immer erhalten....es passiert nix....debuglog gibt für mich nix her.

harrymayr, 2025-05-17:
> .debuglog gibt für mich nix her

Vielleicht für mich ;-)
Aber verwende mal den Manuellen Modus und statt outputLimit die Manuelle Leistung im Zendure Manager.

surfer1264, 2025-05-17:
Er scheint die Werte im manuellen Mode gar nicht anzunehmen.....ich trage immer 100W ein.....es passiert nichts

[home-assistant_zendure_ha_2025-05-17T18-02-29.551Z 2025-05-17 18_02_51.log](https://github.com/user-attachments/files/20266242/home-assistant_zendure_ha_2025-05-17T18-02-29.551Z.2025-05-17.18_02_51.log)

[image]

surfer1264, 2025-05-17:
Kann auch nach Aktivierung des Zendure Managers bzw. Deaktivieren keine anderen Werte mehr schreiben, wie outputlimit oder SOcMinimum

[home-assistant_zendure_ha_2025-05-17T19-14-24.317Z.log](https://github.com/user-attachments/files/20266252/home-assistant_zendure_ha_2025-05-17T19-14-24.317Z.log)

harrymayr, 2025-05-17:
Wenn das alles ist, was von der custom_components.zendure_ha im Log steht, starte bitte mal HA neu. Da scheint mir keinerlei MQTT Kommunikation stattzufinden.

surfer1264, 2025-05-17:
Out of battery :-) Morgen mehr......
Irgendwo gabs den Schalter mqttlog...richtig?  Ich glaube der ist aus :-( Tut mir leid.

Ohne ZendureManager war die Interaktion schon ok......meine eigene Regelung hat den OutputLimiter geschrieben....über Stunden (im Nulleinspeisemodus)

Irgendwas hat sich dann immer mal wieder aufgehangen.. mindestens dann wenn ich in den ZendureManager aktiviert oder auch wieder deaktiviert habe....HA_Start dann schon mehrfach heute gemacht, seit ich mit der Beta heute unterwegs bin....morgen mehr dazu. (natürlich habe ich die eigene Regelung vorher offline genommen, der Hub selbst läuft ohne Ernergiemodus)

harrymayr, 2025-05-18:
> Vielleicht kann [@harrymayr](https://github.com/harrymayr) herausfinden, ob wir irgendwo sehen können, wo der Name unseres P1-Sensors steht.

Wenn ihr über Neu konfigurieren geht, seht ihr den eingestellten Sensor. Die Schaltfläche KONFIGURIEREN zeigt nur den Standard-Wert an und hat bei mir auch sonst keinen Effekt.

_If you click on "Reconfigure," you'll see the configured sensor. The "Configure" button only displays the default value and has no other effect for me._

[image]

surfer1264, 2025-05-18:
You can close this ticket here
For the problem with the Zendure Manager I started a new ticket
https://github.com/FireSon/Zendure-HA/issues/270

#270

closed 2025-05-18 by surfer1264