Zendure/Zendure-HA #263: V 1.0.41-pre8 - bridged bypass ?

olivr2s, 2025-05-17:
Hello,
With the return of the sun, I've just realized on the solar production graphs that with v1.0.41-pre8, when the batteries reach 100%, the bypass is done but the power seems limited and oscillates between 1200W and 1300W, whereas the solar power is much higher.
Before this version was implemented, I was able to bypass up to 1500W.

Wunni73, 2025-05-17:
This is set in Zendur App, the maximum feed-in of your country

olivr2s, 2025-05-17:
In the end, I'm the one who didn't take a good look at my Zendure's solar input power graph and analyze it.😅

As my Zendure is equipped with 2 batteries, the maximum charging power is 1600W, and I remembered having had up to 1580W of solar power...., but this was in cases where the batteries were not full and with sufficient consumption to be compensated for by the Hyper (example: battery charging at 400W and 1200W output from the Hyper).

So effectively, when the batteries are full, the solar power is limited to 1200W at the bypass output.
In short, if I want to take advantage of the maximum power of my panels (1600W), my batteries must not be full, and I need to adapt my consumption.

harrymayr, 2025-05-27:
@FireSon this can be closed?