Zendure/Zendure-HA #265: Bypass Modus vertauscht

masterblaster15, 2025-05-17:
Hallo, erstmal vielen danke für die Integration. Es hat alles super geklappt. Ich habe die Version 1.0.41-pre8 da habe ich festgestellt,
das bei den Steuerelementen des HUB2000 anscheinend der Bypass Modus vertauscht ist. Wenn ich in der Zendure App den Bypass auf 
Immer Aus setze springt der Bypass Modus bei der Integration in Home Assistant auf Immer An. Wenn ich in der Zendure App den Bypass auf 
Immer An setze springt der Bypass Modus bei der Integration in Home Assistant auf Immer Aus. Da scheint irgendwas nicht zu stimmen.

harrymayr, 2025-05-17:
Vielen Dank für die Info. Beim Hub 1200 war es vertauscht im Vergleich zum Hyper, daher die Annahme, dass es beim Hub 2000 auch so ist. Wird in der nächsten pre geändert.

masterblaster15, 2025-05-17:
Super Danke Macht weiter so

harrymayr, 2025-05-17:
Eine Frage an einen HUB2000 Besitzer: Ist der auch so lahm wie der HUB 1200. Von 0 auf 100W braucht der mehr als 30 Sekunden

[image]

masterblaster15, 2025-05-17:
ja ist er

kutzera, 2025-05-18:
> Eine Frage an einen HUB2000 Besitzer: Ist der auch so lahm wie der HUB 1200? Von 0 auf 100W braucht der mehr als 30 Sekunden
> 
> [image]

Über openDTU on Battery kann der HUB1200 innerhalb 10 sek von 0 auf 1000W gesteuert werden…

kutzera, 2025-05-18:
> Vielen Dank für die Info. Beim Hub 1200 war es vertauscht im Vergleich zum Hyper, daher die Annahme, dass es beim Hub 2000 auch so ist. Wird in der nächsten pre geändert.

Habt ihr bereits eine Vorstellung, wann die nächste Pre Version rauskommen soll?

harrymayr, 2025-05-18:
> > Vielen Dank für die Info. Beim Hub 1200 war es vertauscht im Vergleich zum Hyper, daher die Annahme, dass es beim Hub 2000 auch so ist. Wird in der nächsten pre geändert.
> 
> Habt ihr bereits eine Vorstellung, wann die nächste Pre Version rauskommen soll?

Das macht/entscheidet @FireSon

harrymayr, 2025-05-18:
> Über openDTU on Battery kann der HUB1200 innerhalb 10 sek von 0 auf 1000W gesteuert werden…

Wenn ich das richtig verstehe, steuert openDTU on Battery aber nicht den HUB, sondern den Inverter der dahinter hängt. Was aber dann bedeutet, das nicht der HUB sondern der Inverter die Bremse ist? Dem HUB wird von der Integration immer die benötigte Power kommandiert, die Reaktion darauf ist aber so lahm. Ist die Reaktion den auch so langsam, wenn der HUB auf Bypass steht? Das würde dann auch auf den Inverter hinweisen (könnte man mal testen, indem die Solar-Panels bei voller Sonne ab- und dann wieder aufgedeckt werden).
Wie auch immer, ich sehe eine Möglichkeit bei einem gemischten Cluster (Hyper + HUB) dafür zu sorgen, dass der Hyper die langsame Reaktion des HUB ausgleicht, bei Clustern (1 bis x Devices) nur mit HUB sehe ich nicht, wie wir eine langsame Reaktion beschleunigen können.

CWILDMOUNTAIN, 2025-05-18:
> Eine Frage an einen HUB2000 Besitzer: Ist der auch so lahm wie der HUB 1200? Von 0 auf 100W braucht der mehr als 30 Sekunden
> 
> [image]

I can confirm this issue having a Hub 2000, Hoymiles inverter + openDTU. 

And yes, Bypass switch turn on/off is just the Order way round.

kutzera, 2025-05-18:
> Wenn ich das richtig verstehe, steuert openDTU on Battery aber nicht den HUB, sondern den Inverter der dahinter hängt. Was aber dann bedeutet, das nicht der HUB sondern der Inverter die Bremse ist? [...]

Die Trägheit liegt am HUB. Wenn die openDTU steuert, dann wird die Leistung sofort hochgefahren. Das liegt daran, dass bei Regelung über den Wechselrichter der Hub die ganze Zeit auf maximale Abgabelast eingestellt ist. Der Wechselrichter kann in Sekunden die Volllast abgreifen und umwandeln. Wird über den Hub geregelt, dann fährt er die Last langsam hoch und runter.

harrymayr, 2025-05-18:
> Hub die ganze Zeit auf maximale Abgabelast eingestellt ist

Dann sehe ich trotzdem nicht, was Zendure-HA dran ändern kann ohne dass der Inverter gesteuert wird, wenn man eine Nulleinspeisung erreichen will. Die Steuerung des HUB (Elektronik) scheint so träge zu sein.

kutzera, 2025-05-18:
> > Hub die ganze Zeit auf maximale Abgabelast eingestellt ist
> 
> Dann sehe ich trotzdem nicht, was Zendure-HA dran ändern kann ohne dass der Inverter gesteuert wird, wenn man eine Nulleinspeisung erreichen will. Die Steuerung des HUB (Elektronik) scheint so träge zu sein. 
> 

Das sehe ich genauso. Ich weiß nicht, ob es ein Schutz sein soll oder die Regelung durch die Hardware so träge ist. Daher habe ich auch vor zwei Jahren einfach auf Maximum gestellt und lasse nur noch über die openDTU runrerregeln...

MacdiverOne, 2025-05-18:
_...Wie auch immer, ich sehe eine Möglichkeit bei einem gemischten Cluster (Hyper + HUB) dafür zu sorgen, dass der Hyper die langsame Reaktion des HUB ausgleicht..._

Bei mir mit 2 x Hyper2000 + 1 x Hub1200 funktioniert das ohne Probleme.

harrymayr, 2025-05-18:
> _...Wie auch immer, ich sehe eine Möglichkeit bei einem gemischten Cluster (Hyper + HUB) dafür zu sorgen, dass der Hyper die langsame Reaktion des HUB ausgleicht..._
> 
> Bei mir mit 2 x Hyper2000 + 1 x Hub1200 funktioniert das ohne Probleme.

Glaube ich, da es auch immer davon anhängt, wie das Verhältnis der Batterien an den devices ist. @Linus86529 z.B. hat 4*AB1000 an einem HUB und 1*AB2000 an einem Hyper. Da übernimmt der Hyper nur 1/3 der Leistung. Und wenn die 2/3 Leistung des HUB langsam hochfahren sieht man das schon.

MacdiverOne, 2025-05-18:
OK, bei mir ist das jetzt eher, sagen wir mal "üppig", mit 2 x (4 x AB2000S (Hyper2000)) und 1 x (2 x AB1000 (Hub1200))

harrymayr, 2025-05-18:
:-D  Ja, mit einem 1/8 der abgegebenen Leistung siehst du vom HUB Verhalten nicht mehr viel

FireSon, 2025-05-24:
@masterblaster15 Is the bypass mode correct in the new version 41?

masterblaster15, 2025-05-24:
Nein, noch nicht in der Original App steht er bei mir auf Immer aus und im Addon jetzt auf Immer an.

Sepp2812, 2025-05-24:
> Nein, nicht noch in der Original App steht er bei mir auf immer aus und im Addon auf immer auf Immer an.

so ist es bei mir auch  in der v 41

harrymayr, 2025-05-27:
@masterblaster15 , @Sepp2812 behebt die 1.0.44 jetzt das Problem?

masterblaster15, 2025-05-27:
Jetzt passt es. Vielen Dank

Sepp2812, 2025-05-27:
ja behebt das Problem

closed 2025-05-27 by FireSon