Zendure/Zendure-HA #268: English

algernon2607, 2025-05-18:
Hello,
There are a lot of interesting things covered in this section, but the language used is German (or Dutch or...).
For everyone's sake, is it possible to stick to English?

Thanks in advance, everyone.

MacdiverOne, 2025-05-18:
There are a lot of translation tools available, in my opinion there has been no need for years to patronize people who prefer to communicate in their native language or do not speak English (well) or even refuse to do so

harrymayr, 2025-05-18:
I aggree with the translation tools, but I don't think that @algernon2607 wanted to patronize

MacdiverOne, 2025-05-18:
I didn't imply that either, but in my opinion it shouldn't be expected or prescribed by the project itself.

closed 2025-05-18 by algernon2607