Zendure/Zendure-HA #269: energy distribution- very slow reaction and crashing at 800W

CWILDMOUNTAIN, 2025-05-18:
since the 1.0.41-pre8 I sensed a very very slow reaction in Energy Distribution from battery.
Setup: Hub 2000 + ACE connected to Hoymiles Inverter controlled by openDTU. 

So I was used to get an increase of power distribution from zero to 100 within seconds, now it increases power in 2W per second steps, so it takes ages until power needed is there. 

Further I see a strange behavior: 
My Hub is said to push 1200W (maximum possible) to the inverter, but everytime the power exceeds 800W it crahes down to like 100W and and the slow distribution process is starting again. It only happens when Power comes from the battery, Solarpower (e.g. In Bypass) is going through nicely up to 1200W. 

I suppose there is some kind of 'wrong' limit set to the Hub, when taking power from battery (maybe it‘s in a way stuck on the regional boundary '800W by law' in Germany). 

Any ideas?

harrymayr, 2025-05-18:
> Any ideas?

Not yet for the slow increase.
Wrt to the limit: There is an outputlimit of 800W (but long before 1.0.41-pre8). But the intendend behavior is, that it keep the output at 800W. Need to think if we make it changable also for 1 device (if you have more, you can select clusters with more than 800W output).
Is it possible to provide a log (w/ MQTT logging on), so we can see, why it is that slow?

CWILDMOUNTAIN, 2025-05-18:
Ah interesting. As the cluster settings are not given for the hubs, I did not choose a setting.

I have 2 Hub 2000 indeed. I was not aware, I would have to choose Cluster. 
One of my Hubs is providing 1000W, the other one shows the behavior stopping at 800W

What do you recommend which settings should I use if I want both hubs to provide full power each? Both are on the same inverter. 
Set a cluster for each one or put both in one big Cluster?

CWILDMOUNTAIN, 2025-05-18:
so to explain: Hub Ost is the one which makes trouble (crashing somewhere between 800-900W output). Both Hubs (Ost and West) are in combination with an ACE and 2 AB2000.
Ah and as stated before, in pre8 Bypass 'always on' is in fact 'always off' for the Hub 2000.

CWILDMOUNTAIN, 2025-05-18:
.. it looks like the ACEs are "spamming" quite a lot. Maybe ACE's output pack overwrites outputhome of the Hub??
Also I cannot adjust the AC Power for charging the ACE .. 

Problems startet with adding the ACEs as own devices .. when there were only the hubs (version 1.0.36) everything worked perfectely (input and output power) 
.. hope you guys find a solution! Great work so far though.

harrymayr, 2025-05-19:
First question: does sensor.aktueller_hausverbrauch deliver a single value for every phase? Or will the sensor triggered 3 times in milliseconds? I see in the complete log that the P1 sensor is triggered with the same value but with "packages" of 3 inbetween 1-4msec.
The next package of 3 come than 5-7 sec. later as expected.

[image]

CWILDMOUNTAIN, 2025-05-19:
it delivers the sum of all 3 phases .. it’s a shelly 3EM pro, no trigger values just come in as they come.

well, I don’t need to use the power sensor option.. I‘d rather control my inverter than the hubs. 

the main issue happens, no matter if the zendure manager is on or off. 

I find it still strange, that all I tried to reset the Hub failed .. and even more strange, that the second Hub (exactely same setup) is working properly. 

also the Hub is reacting really slow compared to the other. As I said, it seems like at anytime power is deployed from battery a new "hard output limit“ is set and power is crawling sloooowly until it exceeds 800W and then starts over again. 

any ideas, why and where could be a limit I can‘t change anymore (neither in the App nor with the Integration)?

harrymayr, 2025-05-19:
I looked now ~2h through the logs. I figured out, that the Output Drop on Hub Ost happen, if the power out of the battery exceed 400W. But I have no idea why. Hub West do have enough solar power, that the power out of the battery does not exceed 400W in the log.
However, I don't understand were the power come from in Hub West.
Because outputPackPower means, that it charges the battery.
So solarInputPower - outputPackPower or solarInputPower + packInputPower is always ~ 800W and more than 200W difference to the shown outputHomePower.

[image]

A little bit later, the solar power goes down and Hub West start to use some power out of the batteries, but the sum of that is also ~800W

[image]

CWILDMOUNTAIN, 2025-05-19:
> However, I don't understand were the power come from in Hub West. Because outputPackPower means, that it charges the battery.

This should be explained by the Solar-Panel connected to the ACE - as far as I know, it can‘t bypass and is always feeding the battery or - if battery is discharged - pushes the power to the Hub. 

> A little bit later, the solar power goes down and Hub West start to use some power out of the batteries, but the sum of that is also ~800W

Well-  maybe not shown in these logs, but I can confirm the Hub West DOES discharging up to 1200W without Solarpower and on the other hand the Hub Ost only can provide a discharge power up to ~800W. 

So the "hard limit" always is the 800W to home as long as power from battery is provided- in Bypass >1000W is pushed through, so I‘m very confident it’s not the cables or the inverter.