Zendure/Zendure-HA #270: Zendure Manager does not work (SF1200)

surfer1264, 2025-05-18:
Installation works fine (last 1.041-pre8) 
I can set output limit
I can set SocMin

BUT

Zendure Manager seems not to have any impact at the moiment

I started with the "Manuelle Leistungsregelung" and 100W but nothing happen
The oberservation is: The value of 100 could not be passed. The history keeps empty
see this picture here

[image]

I also tried the Smartmatching, seems not to work (with SF1200)

providing some logs here
[home-assistant_zendure_ha_2025-05-18T08-02-10.121Z.log](https://github.com/user-attachments/files/20271813/home-assistant_zendure_ha_2025-05-18T08-02-10.121Z.log)

harrymayr, 2025-05-18:
Need to check deeper in the log, but a first question: What kind of battery is the one with serial-number starting BO4NH...?

surfer1264, 2025-05-18:
AB1000S, I was also wondering when I saw AB????

harrymayr, 2025-05-18:
We did not have a AB1000S for now, so we don't know the start of the serial-no.

Wrt your problem: Out of the log, it seems, that the Bypass is set to Auto. Could you please set this to either On or Off?

surfer1264, 2025-05-18:
No  change
First 
Bypass Off set the Bypass "Immer AN"
Bypass On set the Bypass "Immer Aus"
#265 

There is no change in behavior of "Manuelle Leistungssregelung"
The passed value will not be written...no history

new log here, I played a bit around

[home-assistant_zendure_ha_2025-05-18T08-59-19.944Z.log](https://github.com/user-attachments/files/20271998/home-assistant_zendure_ha_2025-05-18T08-59-19.944Z.log)

harrymayr, 2025-05-18:
I see some output if Bypass is set to 2 (Immer AN) @ 10:58:05 but no set of an output limit. Bypass Mode 2 was only kept for 13 seconds.

BTW "Manuelle Leistung" is only an input value, no real sensor. Changes to this value will be seen in "Batterie Ausgangsleistung"

[image]

surfer1264, 2025-05-18:
So, ok
maybe it works
maybe the problem is sitting in front of the screen :-) 

I have actually no idea what the Zendure Manager is supposed to do .... I red the description many times....but....I dont get it totally.

So "Manuelle Leistungsregelung" is a limiter for the batterie output? Right? ok I see......That wasnt my understanding.

Could be helpful to have a bit more detailed description of the philosophie of the Zendure Manager.

But at the end, after looking at my logs...you would say: It works????

MacdiverOne, 2025-05-18:
Merkwürdig. Irgendetwas in deiner Konfiguration muss noch "weird" sein. Das kriege ich nur hin, wenn ich mit lokalem MQTT-experimentiere - das zerschießt mir regelmäßig meinen Aufbau so, dass ich neu laden muss (weil ich keine Daten mehr bekomme).

Ich nutze sowohl den manuellen Modus mit verschiedenen Leistungen, als auch die automatische Leistungsregelung über längere Zeiträume ohne Probleme - auch in Kombination mit Node-Red-Steuerungen (z. B. Wetter-Forecast bestimmt und setzt min-Soc, die Kalibrierungssteuerung von dir auf Cluster adaptiert, ... sowohl über io-Broker-HA-Interface als auch aus HA heraus) - in den letzten Tagen mit viel Sonne/Wolken-Mix und recht dynamischem Lastgang durchgängig mit der Pre-8 erstaunlich stabil mit allen Daten für alle Devices.

MacdiverOne, 2025-05-18:
[image]

harrymayr, 2025-05-18:
> But at the end, after looking at my logs...you would say: It works????

Can't say to 100%

Depending what you want, I would propose the following:

1: Let Zendure-HA control in- and output to have nearly zero feed-in: Use "Smarte Leistungs-Regelung" in Zendure Manager. Your p1 sensor should give +/- value of your in- and ouput from the grid. Which sensor you've selected, you see in "Neu konfigurieren"

2: Control the output on your self defined control (Node Red, ...): Use "Manuelle Leistungs-Regelung" and change the value of "Manuelle Leistung" by your control.

surfer1264, 2025-05-18:
ok thanks a lot 
I did a test with Smart Matching, pvoding the logs here

[home-assistant_zendure_ha_2025-05-18T09-54-33.912Z.log](https://github.com/user-attachments/files/20272299/home-assistant_zendure_ha_2025-05-18T09-54-33.912Z.log)

and the relevant settings

[image]

1)
Its a bit to less sun today.......would be more helpful to see what happend, when using SmartMatching

2)  
Its interesting...you recommend to use 
number.zendure_manager_manual_power ("Manuelle Leistung")  instead of "number.solarherwegh_output_limit (the direct output value of the hub. 

I played around with last one yesterday, integrating in my own control environment (NodeRed) and this worked also fine.

WOuld like to here some feedback here...and than I am going to close this tcket and observing my environment

Thanks a lot .......Love your investment

harrymayr, 2025-05-18:
With "Manuelle Leistung" you have **one** value to control charge/discharge depending on the sign. Maybe not that interessting for the HUB because this can't be charge from the grid, but indeed for the Hyper ;-)

surfer1264, 2025-05-18:
Thanks would love to see some usecase/example based documentation 

I am a bit confuse with "**one value to control charge/discharge depending on the sign.**" (in my SF1200 context)

Manuelle Leistung : Minus 100 W .....what does this mean  ...the value could be negative or positive?

I dont get it 

But nevertheless...Love your implementation

harrymayr, 2025-05-18:
- SF1200 = Solarflow **HUB** 1200. No possibility to load by AC (from the grid). Manuelle Leistung with negativ value may not have an effect (can't test). Positive value give power to the inverter (Batterie Ausgangsleistung = Manuelle Leistung)
- Hyper = Hyper 2000. Possibility to load from the grid: Manuelle Leistung with negativ value charge battery  (Batterie Eingangsleistung = abs(Manuelle Leistung)). Positive value give power to the grid (Batterie Ausgangsleistung = Manuelle Leistung)

MacdiverOne, 2025-05-18:
Will test the maximum AC charging with 2 x Hyper2000 right now...

surfer1264, 2025-05-18:
I am sorry I dont get it with Zendure Manager

I started two new tests.
enough power available and power request from my flat...but there is no power management existing...... (using a Shelly 3 EM Pro

[home-assistant_zendure_ha_2025-05-18T13-20-33.180Z.log](https://github.com/user-attachments/files/20272972/home-assistant_zendure_ha_2025-05-18T13-20-33.180Z.log)

[home-assistant_zendure_ha_2025-05-18T13-03-56.207Z.log](https://github.com/user-attachments/files/20272973/home-assistant_zendure_ha_2025-05-18T13-03-56.207Z.log)

Checked my Power Sensor again, does not work

[image]

[image]

MacdiverOne, 2025-05-18:
Ich habe mir mal die Logs cursorisch angeschaut. Außer ein wenig Gemecker bezüglich APIs habe ich jetzt spontan nichts auffälliges gesehen. Last/Load Daten sind da und ändern sich. Triggerschwellen sollten eigentlich auch erreicht werden.

Der Manager dümpelt einige Zeit mit Status 0 rum, das ändert sich jedoch mit der Zeit.

Hast du Einträge im Logbuch? Normalerweise sit da auch Bewegung zu In/Out/Status.

Ich hatte mal den Fall, dass ich wirklich das komplette HA System (als auch das OS) neu starten musste, um Konfigurationsänderungen zu laufen zu bekommen. Eventuell ist das ja so etwas, dass sich irgendwelche Prozesse verhakt haben. Wenig hilfreich für dich aktuell, aber als "Silberstreif am Horizont": Das "Teil" läuft, also kriegen wird das auch bei dir zum fliegen... @harrymayr: Ideen?

harrymayr, 2025-05-18:
Auf was steht der Zendure Manager? Ich sehe nur Abfragen der Werte, so als wäre der Zendure Manager auf AUS und es werden einfach nur die aktuellen Zustände geloggt, Aber keinerlei Änderungen des Betriebsmodus, noch irgendwelche Anfragen um den Setpoint zu setzen.

Beim Ändern von z.B. Smarte LeistungsRegelung auf Aus muss im Log sowas stehen:
`[custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 2`
`[custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => 0 capacity 468 program: False`

DiePinkEule, 2025-05-18:
Ich würde mich gerne an das Thema dran hängen, da ähnlich wie bei surfer1264 mein hub2000 keinerlei output mehr liefert sobald ich im Zendure Manager auf Manuelle- oder Smarte-Leistungsregelung schalte. 

Die Intergration des Hub 2000 hat super funktioniert und liefert zügig Werte. 

Da die Aktivitäten im Debug Log nicht unbedingt nach Fehlern aussehen scheitert es bei mir ggf. an Grundlegendem Verständnis des Zendure Managers:

- Setzt der Zendure Manager alle notwendigen Einstellungen für das Hub oder muss ich in der Zendure App noch etwas vorgeben um die HA Integration nutzen zu können?
- Ich möchte für die Smarte Leistungsregelung die Werte meines Tasmota SmartMeter verwenden und habe daher als P1 Sensor für Smart Matching den entsprechenden „Helfer“ Sensor angegeben.

Danke euch..

harrymayr, 2025-05-18:
> * Setzt der Zendure Manager alle notwendigen Einstellungen für das Hub oder muss ich in der Zendure App noch etwas vorgeben um die HA Integration nutzen zu können?
> * Ich möchte für die Smarte [...]

Sinnvoll ist es, in der App alle Energiepläne auszuschalten. Dann macht der localMQTT gerne mal Probleme, den am besten erst mal nicht verwenden. MQTT logging aber anmachen. Was bei mir auch (zumindest am Anfang) mit dem Smarte Leistungsregelung Probleme gemacht hat, ist wenn die Ausgangs- und/oder Eingangslimits nicht auf 0 stand. Ich habe mir daher eine Automation gebastelt, die beim Auswählen von "Smarte Leistungsregelung" die beiden Werte auf 0 setzt.

Und im Zweifel lieber einmal mehr HA starten, speziell nach Ändern der Konfiguration oder Update. Dabei macht es Sinn, erst das Logging zu aktivieren und dann HA neu zu starten, dann wird der komplette Verbindungsaufbau mit geloggt.

Und wenn es dann immer noch nicht funktioniert, das Log hier bereit stellen, dann können FireSon oder ich schauen, warum es nicht funktioniert.

surfer1264, 2025-05-19:
HA Neustart
Eingangs und Ausgangslimit auf 0 gesetzt (sieht man auch in der App)
Manueller Modus
Manuelle Leistung 100W
Da passiert nix.

[image]

[image]

[image]

[image]

[image]

[home-assistant_zendure_ha_2025-05-19T15-32-56.957Z.log](https://github.com/user-attachments/files/20299921/home-assistant_zendure_ha_2025-05-19T15-32-56.957Z.log)

surfer1264, 2025-05-19:
Zweiter Test:

HA Neustart
ZendureHub im Basismodus ohne Energiemodus
DebugLog AN
Smarte Leistungsregelung AN
Eingangs und Ausgangslimit auf 0 gesetzt (sieht man auch in der App)
Lasse ich mal laufen
Nach ein paar Minuten Bypass ausgeschaltet  (Sieht man dann auch in der App)
Akkustand 29% aktuell 
Panelpower 72W .....gehen komplett in den Akku
Smartmeter im PLUS Bereich
Habe dann die Ausgangsleistung auf 50W gesetzt....damit überhaupt der Wechselrichter startet (der APSzstem EZ1 skaliert zw. 30W aufwärts )
Aber da wird bislang nichts ausgeregelt
17:58 schalte ich die Regelung wieder aus und deaktiviere das DebugLog

[home-assistant_zendure_ha_2025-05-19T15-58-34.942Z.log](https://github.com/user-attachments/files/20300366/home-assistant_zendure_ha_2025-05-19T15-58-34.942Z.log)

[image]

[image]

DiePinkEule, 2025-05-19:
> > * Setzt der Zendure Manager alle notwendigen Einstellungen für das Hub oder muss ich in der Zendure App noch etwas vorgeben um die HA Integration nutzen zu können?
> > * Ich möchte für die Smarte Leistungsregelung die Werte meines Tasmota SmartMeter verwenden und habe daher als P1 Sensor für Smart Matching den entsprechenden „Helfer“ Sensor angegeben.
> 
> Sinnvoll ist es, in der App alle Energiepläne auszuschalten. Dann macht der localMQTT gerne mal Probleme, den am besten erst mal nicht verwenden. MQTT logging aber anmachen. Was bei mir auch (zumindest am Anfang) mit der Smarten Leistungsregelung Probleme gemacht hat, ist wenn die Ausgangs- und/oder Eingangslimits nicht auf 0 stand. Ich habe mir daher eine Automation gebastelt, die beim Auswählen von "Smarte Leistungsregelung" die beiden Werte auf 0 setzt.
> 
> Und im Zweifel lieber einmal mehr HA starten, speziell nach Ändern der Konfiguration oder Update. Dabei macht es Sinn, erst das Logging zu aktivieren und dann HA neu zu starten, dann wird der komplette Verbindungsaufbau mit geloggt.
> 
> Und wenn es dann immer noch nicht funktioniert, das Log hier bereit stellen, dann können FireSon oder ich schauen, warum es nicht funktioniert.

Hi, ich habe heute Abend mal verschiedene Einstellungen ausprobiert und das Debug Log laufen lassen..

Eingangs und Ausgangslimit auf 0 gesetzt 
Manueller Modus
Manuelle Leistung 200W
Hub2000 stellt seine Funktion ein.. ;-(
Ich habe hier positive als auch negative Werte ausprobiert, da in der Beschreibung zum Zendure Manager steht, das negative Werte zum entladen eingegeben werden sollen.
Der eingegebene Wert für die Manuelle Leistungsregelung bleibt übrigens nicht erhalten wenn ich die Maske schließe und erneut Aufrufe. 

Umschalten auf den Smarte Leistungsregelung erzielte keinerlei Auswirkung auf das Hub2000.
Neustart des HA durchgeführt..
Keine Veränderung, abgesehen davon das die Integration nun alle Werte in die Mitte geschoben hat und mich auf eine unbekannte Clusterkonfig aufmerksam macht.

Wie oben beschrieben wird auch hier meine Eingabe zum Sensor sensor.power_actual scheinbar nicht übernommen, da jedesmal wieder der Wert sensor.power_actual aufgeführt wird und nicht den von mir eingetragene sensor.smart_meter_aktuelle_wirkleistung.

Anbei das Log: 

[home-assistant_zendure_ha_2025-05-19T15-58-34.942Z.log](https://github.com/user-attachments/files/20310970/home-assistant_zendure_ha_2025-05-19T15-58-34.942Z.log)

harrymayr, 2025-05-19:
> Wie oben beschrieben wird auch hier meine Eingabe zum Sensor sensor.power_actual scheinbar nicht übernommen, da jedesmal wieder der Wert sensor.power_actual aufgeführt wird und nicht den von mir eingetragene sensor.smart_meter_aktuelle_wirkleistung.

Änderungen/Überprüfung nicht mit Konfigurieren, sondern über die 3 Punkte und "Neu Konfigurieren". Danach HA neu starten.
Und bitte vor dem Neustart das Logging anschalten, dann kann man den kompletten Verbindungsaufbau verfolgen.
Welche Version ist installiert? Es sollte die 1.0.41 pre8 sein.

[image]

harrymayr, 2025-05-19:
Wieso sind das die gleichen Logs???

[image]

[image]

harrymayr, 2025-05-19:
> Da passiert nix.

Wieso heißt das Teil eigentlich SolarHerwegh? Da müsste eigentlich Hub 1200 stehen. Jetzt wird mir einiges klar auch bei anderen Problemen.
Bitte den Namen wieder zurück auf Default stellen, sonst funktioniert das nicht.

Im Log würde ich so etwas erwarten:
`[custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /73bkTV/Hub 1200/properties/report`

Ich sehe aber nur 

`[custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /73bkTV/SolarHerwegh/properties/report`

Das könnte schon das ganze Problem sein, weil nach dem Namen gesucht wird.
Ich bräuchte dann auch das Log von Anfang an, also Logging aktivieren und dann HA neu starten.

harrymayr, 2025-05-19:
> Da passiert nix.

@surfer1264 Es könnte sein, dass es am Bypass-Mode liegt. Bitte mal auf "Immer Aus" stellen und nochmal testen.

Und ich bräuchte ein Log, bei dem auch die Verbindung mit geloggt wird. Also Log aktivieren, und HA neu starten, dan die Tests machen und wieder beenden.

surfer1264, 2025-05-20:
> Wieso sind das die gleichen Logs???
> 
> [image]
> 
> [image]

WEIL @DiePinkEule mein Log dahin kopiert hat, sollte sicher nicht so sein

DiePinkEule, 2025-05-20:
> > Wieso sind das die gleichen Logs???
> > [image]
> > [image]
> 
> WEIL [@DiePinkEule](https://github.com/DiePinkEule) mein Log dahin kopiert hat, sollte sicher nicht so sein

Ja, Entschuldigt... da war der Tag dann doch zu lang.. 
Hatte zuvor Surfer1264's Log nach Analogen Meldungen durchsucht und am Ende fälschlicherweise in meinen Kommentar übernommen..
Reiche ich heute Abend nach..

surfer1264, 2025-05-20:
So lets start a new test:

ZendureHub with no Energiemode
Ausgang maximal aufgedreht (600W)
Debugprotokoll activated
HA Restart
Wait
Shelly 3M im Plus (bezieht Energie)
Integration gestartet, alle Daten sind da
Bypass Aus
20:20Uhr Smarte Leistungsregelung gestartet
Stelle sicher, dass WR APSystem EZ1 aufgedreht ist (600W)
Hub liefert nun volles Rohr ...knapp 600W
Shelly 3EM dreht ins Minus (also pustet die Energie ins Netz)
Es passiert nix.
20:25: Ich starte meine eigene Regelmaschine und Stoppe den Zendure Manager

[image]

[image]

Anbei die Logs

[home-assistant_zendure_ha_2025-05-20T18-26-34.270Z.log](https://github.com/user-attachments/files/20353452/home-assistant_zendure_ha_2025-05-20T18-26-34.270Z.log)

DiePinkEule, 2025-05-20:
> Änderungen/Überprüfung nicht mit Konfigurieren, sondern über die 3 Punkte und "Neu Konfigurieren". Danach HA neu starten. Und bitte vor dem Neustart das Logging anschalten, dann kann man den [...]

Ok, Danke. Ich habe heute nun nochmal meine Werte über "Neu Konfigurieren" eingetragen, bis auf meinen Benutzernamen sind die Felder aber auch wieder zurückgesetzt wenn man diese erneut aufruft.

Zum heutigen Log:
22:40Uhr Energieplan in der App deaktiviert.
22:44Uhr Debug Aktiviert und Integration über 3 Punkte Neu Konfiguriert
22:45Uhr  HA Neustart
22:46Uhr Betriebsmodus: Manuelle Leistungsregeung aktiviert: 100W
22:49Uhr Betriebsmodus: Smarte Leistungsregelung
22:51Uhr Betriebsmodus: Aus

[home-assistant_zendure_ha_2025-05-20T20-51-20.479Z.log](https://github.com/user-attachments/files/20355395/home-assistant_zendure_ha_2025-05-20T20-51-20.479Z.log)

Alle Einstellungen haben keine Auswirkungen auf das Hub2000 gehabt.
(Der Durchlauf erfolgte zuvor schon mal mit längeren Pausen zwischen den Schritten; Ergebnis bleibt das Gleiche.)

harrymayr, 2025-05-20:
> Alle Einstellungen haben keine Auswirkungen auf das Hub2000 gehabt.

Das liegt daran, dass anscheinend noch das alte Release verwendet wird. Bitte die 1.0.41-pre8 installieren.

harrymayr, 2025-05-20:
> Alle Einstellungen haben keine Auswirkungen auf das Hub2000 gehabt.

Das liegt daran, dass noch das alte Release verwendet wird, bitte die 1.0.41-pr8 (wie auch oben geschrieben) verwenden.

harrymayr, 2025-05-21:
Laut Log wird der P1 sensor nicht getriggert. An der Länge das Namens kanns nicht liegen, ich habe für meinen Template-Sensor der gleichen Namen verwendet und da funktioniert es.
Bin langsam etwas ratlos.
Man könnte noch versuchen, einen Template-Sensor zu erzeugen und ihn wie von der Integration vorgeschlagen power_actual nennen. Dann über die drei Punkte die Integrationseinträge löschen und neu hinzufügen. Dabei den Template-Sensor verwenden.
Danach HA neu starten. Bin da aber nicht sehr hoffnungsfroh, dass sich da was ändert.

[image]

Zu den Tests wie oben beschrieben noch prüfen, ob ein Eintrag der Manuellen Leistung irgend einen Effekt hat. Dazu Manuelle Leistungsregelung im Zendure Manager auswählen und einen Wert, z.B. 50W angeben. Der sollte dann nach einer Weile als Batterie Ausgangsleistung im HUB zu sehen sein.

Falls das alles nichts bringt, bin **ich** langsam am Ende meiner Ideen. Vielleicht fällt dann @FireSon noch was ein.

surfer1264, 2025-05-22:
Jetzt funktioniert auch "Manuelle Leistungsregelung"

[image]

Ich spreche statt meinen Wechselrichter nun Euren Sensor `number.zendure_manager_manual_power`  an und nutze mein eigenes Regelwerk, weil ich da etwas mehr Einfluss auf die Regelung habe (und haben möchte :-) )

[image]

SEEEEHR, SEEEEHHHRRRRRR COOOOOOL Leute !!!

surfer1264, 2025-05-22:
FINALE LÖSUNG, bei allen Problemen offensichtlich

AUFRÄUMEN, CLEANDESK machen.....alles alte Wegwerfen, Repos aus HACS löschen und ganz neu anfangen....irgendwas muss sich im HA ja eingegraben haben (ich hatte mit der 1.0.40 begonnen und dann die 1.0.41-pre8 nachgeladen, natürlich auch mit Neukonfig und Restart und Co)

harrymayr, 2025-05-22:
> What is a SolarPower1Cycle? What kind of information?

Not that clear. The integration ask the Zendure Server for all availabe variables und get back a bunch of values. Not all of them are usefull for every device. If you check i.e. Hyper.py, then you see the sensors, which have some meaning for the Hyper. These defined sensors get a state- and device-class and also a translation. All others have the returned sensor name and no defined state or device class.

surfer1264, 2025-05-22:
Mit der Manuellen Leistungsregelung muss ich noch Testen

Hatte gerade den Eindruck, dass die eingefroren ist.....

Hintergrund.......Ich habe ein Grundlastpanel zugeschaltet......das bringt den Shelly 3EM in den Minusbereich....
Mein Regelung regelt dann runter auf MInimum 35W (um den WR ein APSYstemEZ1) am Leben zu erhalten)....

das passierte aber nicht ...... der Output blieb bei ca. 75W hängen.......egal was ich dann machte

Bin erstmal wieder zurück auf die Smarte Steuerung

surfer1264, 2025-05-22:
So läuft......auch der manuelle Modus
Habe Eure Z-HA-Integration in meine Regelung integriert und kann nahtlos umschalten 
zw. 
1.) meiner Regelung (nur WR ansteuern)
2.) oder => Zendure Manager
3.) oder Verwendung meiner Regelung aber Ansteuerung Eurer manuellen Leistungsregelung (anstatt meines Wechselrichters)

Läuft, Genial, Great Work, Danke an alle Mitwirkenden
Ich schließe das Ticket hier 

Für alle Interessierten meiner HA-Regelung (mit Node Red)

https://github.com/surfer1264/Zendure-Stuff/wiki/Regelung-zur-Nulleinspeisung-Nullbezug

closed 2025-05-22 by surfer1264

surfer1264, 2025-05-24:
Just to say 
Official 1.0.41 is working with my System without any Problems