Zendure/Zendure-HA #273: 1.0.41-pre 8 crashed my system

Zanooon97, 2025-05-18:
Hello, after installing 1.0.41-pre8 I entered all infos, acc. credentials and wifi. after the login i cant see anny updates in my zendure app. how can i get back to normal? i think the zendure is not talking anymore to the cloud mqtt!
please help as fast as you can

i have two solarflow 800
one shelly 3 empro
David

harrymayr, 2025-05-18:
Hi David, did you restart HA?

MacdiverOne, 2025-05-18:
Did you use different credentials as described in readme? If not: only the last logged in app gets data...

Zanooon97, 2025-05-18:
yes to restart and no second account. 
in the last stable version i used the main account without problems.
after updating to pre8 i stucked immediately.
the first time i have changed server from italy to global and i could reconnect the devices and functionallity!
after retrying the pr8 same thing but now i have no other server to retest.. 

but with pre8 i have only local acces to the solarflow? how can i downgrade and bring the mqtt settings from the solarflow to factory, I think the mqtt settings on the Solarflow are changed now.

Zanooon97, 2025-05-18:
Logger: homeassistant.util.logging
Quelle: util/logging.py:74
Erstmals aufgetreten: 20:40:08 (1 Vorkommnis)
Zuletzt protokolliert: 20:40:08

Module custom_components.zendure_ha.zendurermanager is logging too frequently. 200 messages since last count

MacdiverOne, 2025-05-18:
Use HACS >>> Zendure-HA >>> 3 Points Menu at right hand >>> Reload >>> select other Version >>> select desired version

Do not forget to restart and report what happened.

harrymayr, 2025-05-18:
And do not use localMQTT if you cannot reach your Zendure device with your HA device with bluetooth.

Zanooon97, 2025-05-18:
i have downgraded to 38, restarted, and email an pw. 
i think i messed the device up. why on server change from italy to global fixed the thing. 

do i need bloothoot connection for local mqtt? 
okey why after logging in with this integration the hole system henged up! when i connect the device with bluethoot to my phone zendure app works fine, after making the networksetting it crash at 0watts solar and like 100 watts hous output.

[image]
[image]
[image]

Zanooon97, 2025-05-18:
its 9 pm it shoul have 0 watts. look

[image]

harrymayr, 2025-05-18:
> do i need bloothoot connection for local mqtt

YES. And it tries to connect using BT all the time.

Zanooon97, 2025-05-18:
Now i have the rpi 10cm from the solarflows, configured all with local mqtt but no change,i will wait till tomorrow. 

thank you

harrymayr, 2025-05-18:
Also with restarting HA? If you can provide a debug logging, I can try to figure out, what happen.

[image]
Restart HA
Wait for 10-20 minutes and deactivate Logging. On a computer, the browser download a file. This file will be in the Download-folder of your computer. You can attach this file here.

Zanooon97, 2025-05-19:
its connecting and disconnecting the hole time

[home-assistant(1).log](https://github.com/user-attachments/files/20277238/home-assistant.1.log)

harrymayr, 2025-05-19:
If the localMQTT does not work, disable it an restart HA. It's not necessary to use it (just nice to have).
I can't analyse to log before this evening.

Zanooon97, 2025-05-19:
also without local its nor working... heheh

Zanooon97, 2025-05-19:
**English--**

I’m in a bit of a mess: even when I remove the integration and restart Home Assistant, Zendure stays frozen on my end—it just won’t respond anymore!

Today at lunchtime I connected the inverters to the Zendure app via Bluetooth and set them to a fixed wattage. Using Bluetooth only, I can see the data correctly in the app and change the settings. But as soon as I establish a network connection, nothing works (see screenshot).

Another odd observation:

Yesterday I registered both SolarFlow units under a new account and linked them to Home Assistant with the latest stable version—nothing worked. So today at noon I simply re‑registered both units to my real account (without deleting them first) using only Bluetooth, and suddenly the outputHomeWatt entity was visible again in Home Assistant! But only for as long as my smartphone was connected via Bluetooth. As soon as I switched over to the network connection, everything died again—haha.

Anyway, what I really want now is just to get my app working again so I can start fresh with this integration, because I think it’s great! It’s won me over!

**Deutsch--**

Ich sitzte ein bisschen in der scheise, auch wenn ich diese Integration lösche, HA neu starte. Bleibt zendure bei mir eingefrohren, es reagiert nicht mehr!
Ich habe heute zu mittag die Inverter per Bluethoot mir der Zendure app verbunden und sie auf einen Fixen watt eingestellt. Nur mit bluethoot sehe ich die Daten in der app Richtig und kann einstellungen tätigen. Sobal ich die netzztwerkverbindung aufbaue geht nichtsmehr ( siehe screen).

Noch eine komische beobachtung:

Ich hatte gestern die beiden solarflows in einem neuen account registriert, und in HA mit der letzten stable version verbunden, es ging nichts, also habe ich heute zu mittag beide einfach ohne vorher zu löschen in mein richtigen account wieder registriert nur mit bluethoot. siehe da in HA waren die outputHomeWatt wider zu sehen ??!
Nur solange ich mit meinem smartphone per bluethoot verbunden war. Sobald ich die netztwerkverbindung aufgebaut habe alles tot wieder hehe.
Naja jetzt will ich einfach wieder nur dass meine App funktioniert und ich neu starten kann mit dieser Integration denn ich finde sie klasse! Hat mich überzeugt!

[image]
[image]
[image]
[image]

https://github.com/user-attachments/assets/70605231-a6aa-4e5b-b1fd-a2de39bfdd89

Zanooon97, 2025-05-26:
Hei the Zendure support have deleted the retained message on global server, when I register my devices on Italy server it starts to reboot infinitely. Could I try to delete my Zendure Integration, then change to Italy server, re- configure the integration with local mqtt and hopefully I repare my problem alone without zendure support? 
@FireSon

FireSon, 2025-05-28:
Hi @Zanooon97 yes that should be possible, since the Integration will always remove these topics for all devices it finds. Yous houls make sure you do not select Local MQTT because in that case the message is removed from the local MQTT server

closed 2025-05-31 by FireSon