Zendure/Zendure-HA #274: log errors in 1.0.41-pre8

FrColin, 2025-05-18:
Cette erreur provient d'une intégration personnalisée

Enregistreur: py.warnings
Source: custom_components/zendure_ha/binary_sensor.py:12
intégration: Zendure Home Assistant Integration (documentation, problèmes)
S'est produit pour la première fois: 17 mai 2025 à 08:00:14 (1 occurrence)
Dernier enregistrement: 17 mai 2025 à 08:00:14

/usr/local/lib/python3.13/site-packages/stringcase.py:247: SyntaxWarning: invalid escape sequence '\W' return re.sub("\W+", "", string)

harrymayr, 2025-05-18:
Hmm, perhaps it's because of the french translation and some characters which give some problems. Could you please provide an image of the Hyper or Hub sensors, so that we can compare the intended translation with the real one? An image like this:

[image]
[image]

FireSon, 2025-05-25:
@FrColin Not sure how to solve this issue. Where did you see that stringcase import snakecase is deprecated?
We will be having a closer look at all the current entities, and remove the ones how are not relevant. That would also be a good moment to look at all the translantions.

FrColin, 2025-05-25:
I see on a comment
https://github.com/okunishinishi/python-stringcase/issues/22

    DeprecationWarning: invalid escape sequence \W : return re.sub("\W+", "", string)

Must be :

return re.sub(r"\W+", "", string)

harrymayr, 2025-05-27:
> I see on a comment [okunishinishi/python-stringcase#22](https://github.com/okunishinishi/python-stringcase/issues/22)
> 
> ```
> DeprecationWarning: invalid escape sequence \W : return re.sub("\W+", "", string)
> ```
> 
> Must be :
> 
> return re.sub(r"\W+", "", string)

Ok, not snakecase is deprecated, but this escape sequence. Difficult to do somethung against, beside figuring out, which french character causes this.
Let's see, if I can find a way. But could take several days.

harrymayr, 2025-05-27:
I setup my computer to French and also HA to french language, but could not see this error. So I propose to install 1.0.44 (if you are back at home) and check if the error still exists.
If so, you could try to replace the french translation in custom_components\zendure_ha\translations\fr.json step by step to the english version till the error os gone.
I know, not a nice task, but otherwise I don't see a way to figure out the problem.
On the other hand, if it does not prevent the integration from function, ...

closed 2025-07-31 by FireSon